nRF24/RF24 #579: [Question] Error compiling NodeMcu v1.0 with NRF24L01

DeDisaster, 2020-03-16:
Hello and greeting there ^_^. I alrd try to solve by following this [here](https://github.com/nRF24/RF24/issues/125). But for some reason when I use this library [here](https://github.com/PaulStoffregen/RadioHead) it work lol. However when I use RF24 library it become like these.

Arduino: 1.8.7 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Disabled, 4M (no SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

In file included from C:\Users\Haikal\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0/tools/sdk/libc/xtensa-lx106-elf/include/sys/stdio.h:6:0,

                 from C:\Users\Haikal\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0/tools/sdk/libc/xtensa-lx106-elf/include/stdio.h:63,

                 from C:\Users\Haikal\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266/Arduino.h:32,

                 from C:\Users\Haikal\Documents\Arduino\libraries\RF24\RF24_config.h:60,

                 from C:\Users\Haikal\Documents\Arduino\libraries\RF24\RF24.cpp:10:

C:\Users\Haikal\Documents\Arduino\libraries\RF24\RF24.cpp: In member function 'void RF24::printDetails()':

C:\Users\Haikal\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:76:81: error: 'const void*' is not a pointer-to-object type

     #define pgm_read_ptr(addr)              (*reinterpret_cast<const void*>(addr))

                                                                                 ^

C:\Users\Haikal\Documents\Arduino\libraries\RF24\RF24.cpp:587:13: note: in expansion of macro 'pgm_read_ptr'

     "\r\n"),pgm_read_ptr(&rf24_datarate_e_str_P[getDataRate()]));

             ^

C:\Users\Haikal\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:76:81: error: 'const void*' is not a pointer-to-object type

     #define pgm_read_ptr(addr)              (*reinterpret_cast<const void*>(addr))

                                                                                 ^

C:\Users\Haikal\Documents\Arduino\libraries\RF24\RF24.cpp:590:13: note: in expansion of macro 'pgm_read_ptr'

     "\r\n"),pgm_read_ptr(&rf24_model_e_str_P[isPVariant()]));

             ^

C:\Users\Haikal\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:76:81: error: 'const void*' is not a pointer-to-object type

     #define pgm_read_ptr(addr)              (*reinterpret_cast<const void*>(addr))

                                                                                 ^

C:\Users\Haikal\Documents\Arduino\libraries\RF24\RF24.cpp:593:13: note: in expansion of macro 'pgm_read_ptr'

     "\r\n"),pgm_read_ptr(&rf24_crclength_e_str_P[getCRCLength()]));

             ^

C:\Users\Haikal\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:76:81: error: 'const void*' is not a pointer-to-object type

     #define pgm_read_ptr(addr)              (*reinterpret_cast<const void*>(addr))

                                                                                 ^

C:\Users\Haikal\Documents\Arduino\libraries\RF24\RF24.cpp:596:15: note: in expansion of macro 'pgm_read_ptr'

     "\r\n"),  pgm_read_ptr(&rf24_pa_dbm_e_str_P[getPALevel()]));

               ^

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

TMRh20, 2020-03-16:
Your error indicates you are running the 2.5.0 version of esp8266 core. 

You can update it via Arduino IDE > Tools > Board > Boards Manager

DeDisaster, 2020-03-17:
First of all, thnx for ur reply....so yea...I update the esp8266 core
(2.5.0 version), and the result still same (not working)....and then I try
to use version 2.5.0-beta 3, and still no luck...but then i realise the
download was error, so below is the error:

java.lang.RuntimeException: java.lang.Exception: Error downloading
http://downloads.arduino.cc/packages/package_index.json
at
cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$1(ContributionManagerUI.java:150)
at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: Error downloading
http://downloads.arduino.cc/packages/package_index.json
at
cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:139)
at
cc.arduino.contributions.packages.ContributionInstaller.download(ContributionInstaller.java:333)
at
cc.arduino.contributions.packages.ContributionInstaller.downloadIndexAndSignature(ContributionInstaller.java:307)
at
cc.arduino.contributions.packages.ContributionInstaller.updateIndex(ContributionInstaller.java:285)
at
cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$1(ContributionManagerUI.java:146)
... 1 more
Caused by: java.nio.file.FileSystemException:
C:\Users\Haikal\AppData\Local\Arduino15\package_index.json.tmp: The process
cannot access the file because it is being used by another process.

at
sun.nio.fs.WindowsException.translateToIOException(WindowsException.java:86)
at
sun.nio.fs.WindowsException.rethrowAsIOException(WindowsException.java:97)
at
sun.nio.fs.WindowsException.rethrowAsIOException(WindowsException.java:102)
at
sun.nio.fs.WindowsFileSystemProvider.implDelete(WindowsFileSystemProvider.java:269)
at
sun.nio.fs.AbstractFileSystemProvider.delete(AbstractFileSystemProvider.java:103)
at java.nio.file.Files.delete(Files.java:1126)
at
cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:156)
at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:132)
at
cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:137)
... 5 more

Avamander, 2020-03-17:
Core download errors are not something we can solve. Check your internet connection.

DeDisaster, 2020-03-18:
Umm, so yea my prob was the downloading error from board manager which i cant update anything especially esp8266 community. After i delete all C:\Users\Haikal\AppData\Local\Arduino15 EXCEPT for preferences.txt, and it work...so for just checking, i try example simple LED blynk and it work. but when I #include RF24 library in coding, nothing happen wew :(

DeDisaster, 2020-03-18:
After long troubleshoot, finally it work as I wanted lol. Thnx to Avamander and TMRh20 for helping me. Anyway, the problem was....I really embarrassed to tell but I was not wiring yet NRF24L01 with nodemcu v2 zzz. Well, so yea...problem solved (^_^)

closed 2020-03-18 by DeDisaster