nRF24/RF24 #582: Bare AVR support [Question]

bitomaxsp, 2020-03-20:
Hi. I am working on my project. I use bare AVR mcu's. 
Is there support for bare AVR (non arduino)?
If not, is it planned?
I can probably come up with some reasonable solution.

TMRh20, 2020-03-20:
The support for non-Arduino AVRs is limited to a portability “template”.

No plans to expand support by me, but we will accept pull requests to do so.

Users mainly need to provide for GPIO (digitalWrite) functions, timer functions (delay, millis), and for SPI functionality. Also some progmem stuff, etc.

A good example that would apply here is the ATXMEGA D3, see https://github.com/nRF24/RF24/tree/master/utility/ATXMegaD3

There is also a general compatibility section in the docs. http://tmrh20.github.io/RF24/Portability.html

Avamander, 2020-03-20:
Supporting "bare AVR" would still require a HAL because even AVR MCUs vary a lot, megaAVR is quite different for example. One has to simply provide the required Arduino HAL functions if you want to use the library on an Arduino-unsupported platform.

bitomaxsp, 2020-03-24:
Thank you for all replies.

closed 2020-03-29 by TMRh20