nRF24/RF24 #583: [Question]Python 3.7 Wrapper Installation - RF24 library is not recognized

avaS21, 2020-03-24:
Hello,

I am trying to install the Python Wrapper for Python 3.7 based on: http://tmrh20.github.io/RF24/Python.html

After installing python3-dev & libboost-python-dev, I try to manually link the boost library as follows (similar to a previous post): 

**sudo ln -s /usr/lib/arm-linux-gnueabihf/libboost_python-py37.so.1.67.0 /usr/lib/arm-linux-gnueabihf/libboost_python3.so**

_Output: Failed to create symbolic link 'libboost_Python3.so': File exists._

I then changed into the pyRF24 directory (~rf24libs/RF24/pyRF24) and ran without errors:
**./setup.py build**

However, when I run the pingpair_dyn.py example from terminal or in Thonny IDE, I get the error:

_Output: ModuleNotFoundError: No module named 'RF24'_

I am running the code from a folder that includes the RF24 files (.cpp, .h, .o) and RF24_config.h

Some guidance as to where I'm going wrong would be very appreciated.

Thank you.

TMRh20, 2020-03-25:
Since you were so specific about the steps taken, I have to ask, did you run “sudo setup.py install” ?

avaS21, 2020-03-25:
Yes, I did run “sudo setup.py install.” Sorry I forgot to include it above.

Now taking a closer look at the output however, it seems it was installed for python 2.7 not 3.7:

_Copying RF24-1.3.4-py2.7-linux-armv7l.egg to /usr/local/lib/python2.7/dist-pack$
Adding RF24 1.3.4 to easy-install.pth file

Installed /usr/local/lib/python2.7/dist-packages/RF24-1.3.4-py2.7-linux-armv7l.$
Processing dependencies for RF24==1.3.4
Finished processing dependencies for RF24==1.3.4_

Do you think the problem is related to linking the boost library as described above? I am relatively new to Linux so please forgive me if it's a simple syntax error. 

Thank you!

TMRh20, 2020-03-25:
A couple things I guess, mainly you need to install using python3.

1. See https://github.com/nRF24/RF24/issues/514
You may want to remove the existing link:  sudo rm
/usr/lib/arm-linux-gnueabihf/libboost_python3.so
Then rerun the following command:  sudo ln -s $(ls
/usr/lib/arm-linux-gnueabihf/libboost_python-py3*.so | tail -1)
/usr/lib/arm-linux-gnueabihf/libboost_python3.so

2. Run: python3 setup.py build
3. Run: sudo python3 setup.py install
4. I also had to run: sudo pip3 install RPi.GPIO
5. Retry the example

The python documentation has not been updated for some time, so we will
have to look at that...

avaS21, 2020-03-25:
Thank you for the quick replies! These instructions worked to install the wrapper for Python 3.7, although I didn't need to do (4).

I am now able to run pingpair_dyn.py (see SPI Configuration below), select 0, which results in the output: _"Role: Pong Back, awaiting transmission"_ and then nothing happens.

On the Arduino side, I used the pingpair_dyn.ino sketch with updated pins and the following settings, which I matched in the RPi:

**const int role_pin;
radio.setChannel(0x4c);
radio.setPALevel(RF24_PA_LOW);
radio.setDataRate(RF24_1MBPS);**  

**Sanity check:** I am able to successfully run the gettingstarted C++ example with similar settings, which suggests I have set up the hardware correctly. My preference is to use the Python script because I will be adding code that is easier developed in Python than C++.

My code is attached for reference.

Thanks again for your assistance.
[Code.zip](https://github.com/nRF24/RF24/files/4382269/Code.zip)

**Arduino Output:**
_RF24/examples/pingpair_dyn/_
_ROLE: Ping out_
__

**RPi Output**
================ SPI Configuration ================
CSN Pin          = CE0 (PI Hardware Driven)
CE Pin           = Custom GPIO22
Clock Speed      = 8 Mhz
================ NRF Configuration ================
STATUS           = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1     = 0xf0f0f0f0d2 0xf0f0f0f0e1
RX_ADDR_P2-5     = 0xc3 0xc4 0xc5 0xc6
TX_ADDR          = 0xf0f0f0f0d2
RX_PW_P0-6       = 0x20 0x20 0x00 0x00 0x00 0x00
EN_AA            = 0x3f
EN_RXADDR        = 0x02
RF_CH            = 0x4c
RF_SETUP         = 0x03
CONFIG           = 0x0e
DYNPD/FEATURE    = 0x3f 0x04
Data Rate        = 1MBPS
Model            = nRF24L01+
CRC Length       = 16 bits
PA Power         = PA_LOW
 ************ Role Setup ***********
Choose a role: Enter 0 for receiver, 1 for transmitter (CTRL+C to exit) 0
Role: Pong Back, awaiting transmission

TMRh20, 2020-03-25:
Basically you would be down to standard troubleshooting at this point.

1. You can reboot the RPi or disconnect VCC from the radio to ensure default config when changing sketches.
2. See the Arduino examples to enable printDetails() and print out the Arduino config, compare to RPi.
3. Double check the pins used in the python constructor are the same as used in the c++ example.

Those are the main things that come to mind.
Its probably going to be a minor setting or something.

avaS21, 2020-03-26:
I was able to get the pingpair_dyn code to run from Arduino to RPi Python. 

A few notes that may help someone else trying to implement:

(1) The Arduino code gets stuck on radio.printDetails(). Perhaps I am missing a library?
(2) I renamed the variable "len" in the Python code because "len" is already a built-in Python function. 

Thanks again for your help!

TMRh20, 2020-03-26:
Right on, thanks for the info.

Btw the printDetails() function requires:
#include “printf.h”
and
printf_begin();

See the examples.

closed 2020-03-29 by TMRh20