nRF24/RF24 #584: Hi ! I am using a RaspberryPi 4 and I am having this issue: RF24 HARDWARE FAIL: Radio not responding, verify pin connections, wiring. It is strange because everything is connected as it is instructed and the nRF24L01+ module works in Arduino. Can you please help me figure out this problem ?

leonardo1997-sh, 2020-03-25:


Avamander, 2020-03-25:
Do you have SPI enabled in RPi's config? Are you sure about the wiring on the RPi? Which module do you have? (High powered ones don't work on RPi's 3v3 bus)

leonardo1997-sh, 2020-03-25:
Thank you for your reply. I am sure the SPI is enabled and that the wiring is correct as I have been trying this for days. How do I know if it is a high powered one? It is the normal module. It is written on it JHP 16.000. Does HP stand for high power? Thank you very much!

MiguelAgueda, 2020-04-12:
Are you still having issues with this? I will be building this project soon, except on a RPi 3B. What have you tried to get this working?

leonardo1997-sh, 2020-04-15:
Hi Miguel. I still have that problem. I followed the instructions as specified. Please let me know if it works when you try it. Thank you.

leonardo1997-sh, 2020-04-15:
> Are you still having issues with this? I will be building this project soon, except on a RPi 3B. What have you tried to get this working?

Hi Miguel. I still have that problem. I followed the instructions as specified. Please let me know if it works when you try it. Thank you.

MiguelAgueda, 2020-05-03:
All of my parts have finally arrived. I was disappointed to see the same error on my end. I have definitely wired the module correctly. 
I am not going to stop until this is solved, Ill let you know if I ever find a solution.

closed 2021-01-02 by 2bndy5