nRF24/RF24 #586: [Question]What Maximum range we can get from below settings?

marcus011, 2020-04-07:
DATA_RATE : 250 KBPS
POWER : MIN OR LOW
AUTO ACK: TRUE

Hi, 
I want to know what is the maximum range i can achieve with these settings?

Avamander, 2020-04-07:
> I want to know what is the maximum range i can achieve with these settings?

You have to test it in your conditions, your data and your hardware.

closed 2020-04-07 by Avamander

marcus011, 2020-04-08:
Due to this lockdown we can't , So if anyone done this before,it would be very helpful for us.

wmarkow, 2020-04-08:
With this settings:
DATA_RATE : 250 KBPS
POWER : MAX
AUTO ACK: FALSE

I had like max 50 meters in an open outside space. I was testing with a regular RF24 chip with the antena printed on the PCB (so it was not a PA/LNA version). I have also noticed that the location of the transmitter/receiver in the space has also meaning: when I rotated a bit the transmitter (or receiver), so the PCB antenna looks into a different angle then the reception of the signal was worse.

marcus011, 2020-04-08:
Thanks for the information @wmarkow