nRF24/RF24 #587: Cómo puedo acceder a los dispositivos IP de Gateway desde mi red lan

potier97, 2020-04-08:
Hola, estoy trabajando con los NRF24L01+ y estoy utilizando NRF24GATEWAY con una RASPBERRY PI 3 para crear mi red de sensores, y estoy creando un servidor como el ejemplo de  InteractiveServer_Mesh con una esp8266, pero no puedo acceder desde mi computadora al servidor para interacturar.

Generé estos permisos, pero no me genera el enrutamiento a http://10.10.2.4:10000

1. sudo sysctl -w net.ipv4.ip_forward = 1 
2. sudo iptables -t nat -A POSTROUTING -j MASQUERADE 

Mi RPI tiene una dirección IP estática 192.168.1.50

Gracias

TMRh20, 2020-04-08:
Hola, solo hablo inglés, así que espero que esto tenga sentido. Estoy usando el traductor de google.

Es posible que deba configurar el enrutamiento localmente o en su enrutador, algo como esto:

Local (linux): `sudo route add -net 10.10.2.0/24 gw 192.168.1.50`
Local (Windows): `route ADD 10.10.2.0 MASK 255.255.255.0 192.168.1.50`
Router might have something like: `Subnet IP 10.10.2.0 Subnet Mask 255.255.255.0 Gateway IP 192.168.1.50`

FYI: Principalmente he detenido el desarrollo de RF24Gateway / RF24Ethernet. Hay algunos problemas pendientes, y no es súper estable, pero funciona bastante bien en su mayor parte.

potier97, 2020-04-08:
Hola nuevamente, ya me funciona, gracias por tu tiempo, realmente me gusta esta biblioteca y lo que has logrado con el NRF24L01+, no se si en el futuro vas a actualizar la biblioteca, pero me gusta trabajar con estos dispositivos, gracias por su tiempo

closed 2020-04-08 by TMRh20