nRF24/RF24 #589: [Question] Distributing Python wrapper module with pip

dsilva-vd, 2020-04-21:
Hi,

Is it possible to distribute the Python wrapper library by itself (i.e. the output .egg file from /rf24libs/RF24/pyRF24/dist) so it can be used with pip?

I have a project and I want my colleagues to recreate the exact same environment when developing. I thought I could copy the .egg file into our code tree and add it to the requirements.txt file so they can install them all with ```sudo pip3 install -r requirements.txt```.

Is this possible? I know the .egg file can't be installed by pip so I could convert it to .whl
Thanks!

2bndy5, 2020-09-17:
Looking at the age of this issue and considering the many platforms this library supports, this request may not be as simple as it sounds.

Are you familiar with [CircuitPython](https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/installing-circuitpython-on-raspberry-pi)? It supports a wide variety of platforms including Raspberry Pi and Jetson, and it works seamlessly with CPython3's standard libraries. I wrote a [CircuitPython library for the nRF24L01+](https://github.com/2bndy5/CircuitPython_nRF24L01) which is [available via pip3](https://pypi.org/project/circuitpython-nrf24l01/), but the latest "stable" version is actually a year old. I'm [currently working on a PR](https://github.com/2bndy5/CircuitPython_nRF24L01/pull/14) to update and improve the CircuitPython library for a new release, but I hit a snag while trying to make it OTA compatible with this library. The [API](https://circuitpython-nrf24l01.readthedocs.io/en/stable/api.html) is a little different as well (mostly due to PEP conventions).

I don't mean to self-promote on this repo, but if the information helps you solve this issue...

2bndy5, 2020-09-21:
github is building versions of this C++ library in this repo's "actions", so maybe we can use an extended workflow to upload a pyRF24 build to pypi.org

dsilva-vd, 2020-09-24:
@2bndy5 THanks Brendan I had forgotten about this request and was busy in development of other tasks. I might be coming back to using the RF24 so your comments are timely. I found your github repo so I'll ask some followup questions in there.

closed 2020-09-24 by dsilva-vd