nRF24/RF24 #590: [Question] Compiling/Installing on Ubuntu on rPI4?  Anyone making it happen?

kitplummer, 2020-04-28:
I'm trying to use my rPI4 + Ubuntu Server as a development environment.  

Getting this:

```
g++ -fPIC -marm -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -mfloat-abi=hard -Ofast -Wall -pthread  -c RF24.cpp
g++: error: unrecognized command line option ‘-marm’
g++: error: unrecognized command line option ‘-mfpu=vfp’
g++: error: unrecognized command line option ‘-mfloat-abi=hard’
make: *** [Makefile:42: RF24.o] Error 1
make: Leaving directory '/home/ubuntu/rf24libs/RF24'
```

kitplummer, 2020-04-28:
Some progress, by cleaning up the Makefile.inc and not cross-compile.

```
ubuntu@ubuntu:~/Raspberry-Pi-NRF24$ sudo ./scannerkit 
================ SPI Configuration ================
CSN Pin  	 = CE0 (PI Hardware Driven)
CE Pin  	 = Custom GPIO17
Clock Speed	 = 8 Mhz
================ NRF Configuration ================
STATUS		 = 0x0f RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=1
RX_ADDR_P0-1	 = 0xf7f7f7f7f7 0xe3e3e3e3e3
RX_ADDR_P2-5	 = 0xe3 0xe6 0xe7 0xe7
TX_ADDR		 = 0xf7f7f7f7f7
RX_PW_P0-6	 = 0x00 0x00 0x00 0x00 0x00 0x00
EN_AA		 = 0x00
EN_RXADDR	 = 0x03
RF_CH		 = 0x6e
RF_SETUP	 = 0x1f
CONFIG		 = 0x0f
DYNPD/FEATURE	 = 0x00 0x00
Data Rate	 = 2MBPS
Model		 = nRF24L01
CRC Length	 = 16 bits
PA Power	 = PA_MAX
000000000000000011111111111111112222222222222222333333333333333344444444444444445555555555555555666666666666666677777777777777
0123456789abcdef0123456789abcdef0123456789abcdef0123456789abcdef0123456789abcdef0123456789abcdef0123456789abcdef0123456789abcd
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
```

kitplummer, 2020-04-28:
Would be good to get this to work out the box, but I'm not sure how to set up the Makefile.inc based on host/target.

kitplummer, 2020-04-30:
So, it isn't actually working.  I can't get the actual NRF Configuration to change, which is causing my radio to not work.

Here's my test code in case it might provide some insight:

```
Copyright (C) 2011 J. Coliz <maniacbug@ymail.com>

 This program is free software; you can redistribute it and/or
 modify it under the terms of the GNU General Public License
 version 2 as published by the Free Software Foundation.

 03/17/2013 : Charles-Henri Hallard (http://hallard.me)
              Modified to use with Arduipi board http://hallard.me/arduipi
						  Changed to use modified bcm2835 and RF24 library
TMRh20 2014 - Updated to work with optimized RF24 Arduino library

 */

#include <cstdlib>
#include <iostream>
#include <sstream>
#include <string>
#include <unistd.h>
#include <RF24/RF24.h>

using namespace std;
// RPi generic:
RF24 radio(RPI_V2_GPIO_P1_22, RPI_V2_GPIO_P1_24, BCM2835_SPI_SPEED_8MHZ);

int main(int argc, char** argv)
{
    cout << "RF24/examples/GettingStarted/\n";

    radio.begin();
   // radio.setChannel(5);
    radio.setCRCLength(RF24_CRC_16);
    //radio.setPALevel(RF24_PA_LOW);
    radio.setDataRate(RF24_2MBPS);
    // optionally, increase the delay between retries & # of retries
    radio.setRetries(15, 15);

    radio.openReadingPipe(1, 0xF0F0F0F0D2LL);

    radio.startListening();

    radio.printDetails();
    // forever loop
    while (1) {
        int value = 0;
        // if there is data ready
        if (!radio.available()) {
            // Dump the payloads until we've gotten everything
            // Fetch the payload, and see if this was the last one.
            radio.read(&value, sizeof(value));
            //if(value != 0) {
            if(value != 0) {
//		cout << "eCO2: " << value << "\n";
                printf("eCO2: %d\n", value);
            }
            //}
            delay(1000); //Delay after payload responded to, minimize RPi CPU time
        }
    } // forever loop

    return 0;
}
```

TMRh20, 2020-04-30:
With the following info:
```
[OK] machine detected: SoC=BCM2835, Type=RPi, CPU=aarch64.
[SECTION] Detecting DRIVER
  [OK] DRIVER detected:RPi.
```
It looks like you are using the BCM driver, which is using kind of a hack to support the RPI4.
You may have better luck with the SPIDEV driver, ( ./configure --driver=SPIDEV; make; sudo make install)

kitplummer, 2020-04-30:
Ah thanks @TMRh20 , getting closer.  Settings are correct, but now not receiving. :)

```
ubuntu@ubuntu:~/RF24/examples_linux$ sudo ./kitrx
RF24/examples/GettingStarted/
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0xf0f0f0f0d2 0xf0f0f0f0d2
RX_ADDR_P2-5	 = 0xc3 0xc4 0xc5 0xc6
TX_ADDR		 = 0xe7e7e7e7e7
RX_PW_P0-6	 = 0x20 0x20 0x00 0x00 0x00 0x00
EN_AA		 = 0x3f
EN_RXADDR	 = 0x03
RF_CH		 = 0x4c
RF_SETUP	 = 0x23
CONFIG		 = 0x0b
DYNPD/FEATURE	 = 0x00 0x00
Data Rate	 = 250KBPS
Model		 = nRF24L01+
CRC Length	 = 8 bits
PA Power	 = PA_LOW
```

Verifying the receive on another radio connected to a DigiSpark (Arduino sketch).

kitplummer, 2020-04-30:
Oops, that's my bad.  Am receiving things now, but doesn't look right.  Still workin' it.

TMRh20, 2020-04-30:
Remember RPi is running using 32 bits where Arduino is typically 8-bits, so
depending on code, could just be a data-type mismatch. Specifying variable
sizes (int16_t or int32_t etc) would clear that up.

kitplummer, 2020-05-01:
So, ultimately it was a combination of the SPIDEV, packing stuff in a struct, and matching all the datarate settings.  Am cookin' with fire now.  Thanks!

closed 2020-05-01 by kitplummer