nRF24/RF24 #592: Outdated documentation for SPIDEV

harrison3000, 2020-05-07:
Not really a bug per se, but the documentation for how to use spidev (other than /dev/spidev0.0) is very hard to find.

The constructor documentation at http://tmrh20.github.io/RF24/classRF24.html doesn't say how to do it, the comments on gettingstarted.cpp also don't explain it, I actually had to dig the code and only found the answer in the SPI::begin method

TMRh20, 2020-05-07:
There is some info in the documentation under Related Pages -> [Raspberry Pi](http://tmrh20.github.io/RF24/RPi.html)  
 
Maybe its a good idea to add a link to that page from the constructor documentation?

TMRh20, 2020-05-26:
http://tmrh20.github.io/RF24/classRF24.html#af564e19261afb56005d2411d0c7b05de

closed 2020-05-26 by TMRh20