nRF24/RF24 #594: 2 methods to release the CLK(G13) on NANO when SPI is not used

net234, 2020-05-11:
Hello 

On NANO and NANO-RF board,  SPI is on G13 so the LED_BUILTIN can't be used

I propose add two methods on NRF24 classe to allow release SPI and free G13.

This don't change anything in RF24 and may be useful.

I am new to Git so prefer to propose code here but i am ready to try fork but I will need help :)

net234

/****************************************************************************/
 * should be call before any access to RF24 if disableSPI() is used
 */
inline void RF24::enableSPI()
{
	_SPI.begin();
}

/****************************************************************************/
 * allow  digital.write(G13,LedStatus)  to access the Built In LED
 * enableSPI() must be call before any access to RF24 
 *  see example below
 */
inline void RF24::disableSPI()
{ 
	_SPI.end();
}

/************ exemple of use of G13 with radio.read **************
 * On your setup() add a radio.disableSPI() after your radio initialization
 *   radio.startListening();
 *   radio.disableSPI();                         // disable SPI after init
 *   digitalWrite(13, LOW);                   //  setup G13 as you want   
 *
 * On your loop :
 * 
 * bool saveG13 = digitalRead(13);    // save G13 state
 * radio.enableSPI();                           // enable SPI
 * 
 *                                                          // RF24 read job
 * if ( radio.available() )  {
 *    radio.read( &receveRadioMessage, sizeof(receveRadioMessage) );
 *    gotRadioMessage = true;
 * }
 *  radio.disableSPI();                         // disable SPI
 *   digitalWrite(13, saveG13);            // set G13 back
 *
 *  NOTE : this exemple is with no interrupt and only one RF24 on SPI (no other device)
 */

/****************************************************************************/
`

TMRh20, 2020-05-12:
I'm pretty sure you can just `#include <SPI.h>` in your sketch and then call `SPI.begin()` or `SPI.end()` directly as required. I don't see why this would be included in RF24.

net234, 2020-05-12:
I have try this at first :)   
It doesn't work because SPI only close the opened instance of the SPI object so you have to close the RF24 SPI instance.
This is why I am proposing this

TMRh20, 2020-07-29:
@net234 
> It doesn't work because SPI only close the opened instance of the SPI object so you have to close the RF24 SPI instance.

I actually had a chance to test this out and it works fine without modifying the RF24 library just as I suspected. The reason being is that a call to SPI.begin() merely sets the proper bits in the SPCR register.

To prove this out, just try the following code in the loop of your sketch:
```
  if(millis()-blink_timer > 5000){
    
    Serial.println("BLinking Begin");
    delay(500);
    Serial.println(SPCR,BIN);
    SPCR &= ~_BV(SPE);
    Serial.println(SPCR,BIN);
    digitalWrite(13,LOW);delay(500);    
    digitalWrite(13,HIGH);delay(500);
    digitalWrite(13,LOW);delay(500);
    digitalWrite(13,HIGH);delay(500);
    digitalWrite(13,LOW);delay(500);
    digitalWrite(13,HIGH);delay(500);
    digitalWrite(13,LOW);delay(500);
    digitalWrite(13,HIGH);delay(500);
    digitalWrite(13,LOW);
    SPCR |= _BV(SPE);
    Serial.println(SPCR,BIN);
    blink_timer = millis();
 }
```

The calls to `SPCR &= ~_BV(SPE);` and `SPCR |= _BV(SPE);` are essentialy the main components of SPI.end() and SPI.begin() since they disable and enable SPI respectively. You can replace them with calls to SPI.end() and SPI.begin, and the code functions fine.
Closing this issue as not necessary.

closed 2020-07-29 by TMRh20