nRF24/RF24 #598: Raspberry 4 Not working

alvaromoure, 2020-05-25:
I am unable to set up a receiver in a Raspberry Pi 4 with the python wrapper with an Arduino running the same library. Running the only example (pingpair_dyn.py) does not work either. Installation has been done correctly, using the SPI option. I would also recommend to add more python examples to the library.

Running the pingpair_dyn.py prints the following:
 sudo python3 pingpair_dyn.py 
pyRF24/examples/pingpair_dyn/
================ SPI Configuration ================
CSN Pin  	 = CE0 (PI Hardware Driven)
CE Pin  	 = Custom GPIO22
Clock Speed	 = 8 Mhz
================ NRF Configuration ================
STATUS		 = 0x00 RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=0 TX_FULL=0
RX_ADDR_P0-1	 = 0x0000000000 0x0000000000
RX_ADDR_P2-5	 = 0x00 0x00 0x00 0x00
TX_ADDR		 = 0x0000000000
RX_PW_P0-6	 = 0x00 0x00 0x00 0x00 0x00 0x00
EN_AA		 = 0x00
EN_RXADDR	 = 0x00
RF_CH		 = 0x00
RF_SETUP	 = 0x00
CONFIG		 = 0x00
DYNPD/FEATURE	 = 0x00 0x00
Data Rate	 = 1MBPS
Model		 = nRF24L01
CRC Length	 = Disabled
PA Power	 = PA_MIN

Connections have been double and tripled checked.

TMRh20, 2020-05-25:
All 0x00s means there is no SPI communication happening.
1. You can run a loopback test 
https://www.raspberrypi.org/documentation/hardware/raspberrypi/spi/README.md#troubleshooting
2. Can verify pin declarations + wiring: (Using SPIDEV driver) https://tmrh20.github.io/RF24/RPi.html
3. Verify hardware, test with 2xArduino etc if possible, switch radio devices
4. Get a bigger hammer?

alvaromoure, 2020-05-26:
1) Loopback test is successful:
pi@raspberrypi:~/Development $ ./spidev_test -D /dev/spidev0.0
spi mode: 0
bits per word: 8
max speed: 500000 Hz (500 KHz)

FF FF FF FF FF FF 
40 00 00 00 00 95 
FF FF FF FF FF FF 
FF FF FF FF FF FF 
FF FF FF FF FF FF 
DE AD BE EF BA AD 
F0 0D

2) Ping declaration seems to be fine, I appear to have spidev0.0 and sipdev0.1 in /dev. I therefore tried radio(22,0) and radio(22,1) in the declaration but none of them seem to work.

3) Tests with 2 Arduino were successful using your library. I am now trying to migrate the receiver to a raspberry pi.

4) hahahaha I hope that did work

Anyway, if you have any more suggestions or considerations I would highly appreciate that you told me. I will try now a fresh install.
Thanks in advance.

TMRh20, 2020-05-29:
> I thought radio(22,0) referred to GPIO22(pin nº15) (as is stated in the documentation)

Err, you are correct, I had it backwards. You would think I would know this stuff lol.  The BCM2835 enums are found [here](https://www.airspayce.com/mikem/bcm2835/group__constants.html) and are basically the same for SPIDEV.

Yeah, with the LNA+PA modules, I get SPI communication and printDetails() returns good info, but it won't work even at min power level. I haven't had a chance to test with a separate power supply.

alvaromoure, 2020-06-05:
Finally figured out everything and it's currently working. It had to be with the rf module (which was malfunctioning) and the python wrapper (the example available does not work with the Arduino one). I 'll probably make a YouTube tutorial on how to make it work using this setup (Arduino + Raspberry (Pa+ LNA) and its common problems. Also there is a tool which checks the health of the module. This is sometimes the main problem of people writing issues so pointing them to that tool may be useful. Anyway thanks for your help and of course.. THE LIBRARY!!

TMRh20, 2020-06-06:
Thanks for posting your results/info. If you do make a tutorial/video feel free to post a link here. We could use more RPi info/links in the docs.

closed 2020-06-06 by TMRh20