nRF24/RF24 #599: ATMEGA64 and NRF24L01 Can't get anything to work.

cyfite, 2020-05-30:
HI 
I have trouble with atmega64 and nrf240l01.i double checked my connections and my code works perfectly on UNO

TMRh20, 2020-05-30:
The lib hasn't been specifically tested with atmega64 as far as I know, but it should work if using an Arduino core. If you call `radio.printDetails();` after `radio.begin();` what is the output?

cyfite, 2020-05-30:
Unfortunately i dont have access to serial. how can i see radio.printDetails() output?

TMRh20, 2020-05-30:
Well I guess you can't without serial. You could connect an LED and use it to test specific settings.
ie:
```
if( radio.begin() == false){ 
    //FLASH LED HERE
 }
```
or
```
radio.setChannel(21);
if(radio.getChannel != 21){
    //FLASH LED HERE
}
```
Also if radio.begin() or radio.getChannel() are returning 0xFF or 0x00 then that is a good indication that SPI communication is not working. If you are getting good results, then it may be a communication or coding issue.

cyfite, 2020-05-31:
LED turned on!
so as you said SPI communication is not working.what can i do?

TMRh20, 2020-05-31:
Most likely things are going to be wiring and power supply. If its powered and connected properly, you should be getting SPI communication. Not a lot to it at that point.

closed 2020-08-11 by TMRh20