nRF24/RF24 #600: [Question] Not working at high amplification level

BJ4K3, 2020-06-03:
Hi, I started working with this library yesterday, after some experiments all hardware and software were connected each other. The problem is that if I try setting amplification level higher than LOW no data are delivered to the receiver. In fact if I set the amplification level to MIN or LOW all works fine, even without losing data, but at higher amplification level no data are delivered. I read online that the problem can be the power supply, so I powered the breakout board with the 3.3V regulator by an external power supply. Can someone help me solving this problem? Should I powering up the receiving module by an external power supply too? 

These are the sketches:

Sender: 
```
#include <SPI.h>
#include <nRF24L01.h>
#include <RF24.h>

//Arduino nano
#define CE_PIN   9
#define CSN_PIN 10

bool checkConn = false;

const byte nrfAddress[8] = "Node1";
RF24 radio(CE_PIN, CSN_PIN);

int blinkl = 0;

void setup()
{
  Serial.begin(9600);
  radio.begin();
  radio.setPALevel(RF24_PA_MAX);
  radio.setChannel(1);
  radio.setRetries(15, 10);
  radio.setDataRate(RF24_250KBPS);
  radio.openWritingPipe(nrfAddress);
}

void loop()
{
  if(radio.isChipConnected()){
    if(checkConn == false){
      checkConn = true;
      Serial.println("Connected");
    }
  }else{
    if(checkConn == false){
      checkConn = true;
      Serial.println("Not Connected");
    }
  }
  if(blinkl == 0){
    blinkl = 1;
  }else{
    blinkl = 0;
  }
  Serial.println(blinkl);
  bool rtre = radio.write(&blinkl, sizeof(blinkl));
  Serial.println("STATUS: " + String(rtre));
  delay(1000);
}
```

Receiver:
```
#include <DigitalIO.h>
#include <nRF24L01.h>
#include <RF24.h>

//ESP32
#define CE_PIN    12
#define CSN_PIN   14

bool checkConn = false;

const byte nrfAddress[8] = "Node1";
RF24 radio(CE_PIN, CSN_PIN);

#define LED 2

int blinkl = 0;

void setup()
{
  Serial.begin(9600);
  radio.begin();
  radio.setDataRate(RF24_250KBPS);  
  radio.setChannel(1);
  radio.openReadingPipe(1, nrfAddress);
  delay(100);
  pinMode(LED, OUTPUT);
}

void loop()
{
  radio.startListening();
  if(radio.isChipConnected()){
    if(checkConn == false){
      checkConn = true;
      Serial.println("Connected");
    }
  }else{
    if(checkConn == false){
      checkConn = true;
      Serial.println("Not Connected");
    }
  }
  while(radio.available()){
    int intSize = sizeof(blinkl);
    radio.read(&blinkl, intSize);
    Serial.println(blinkl);
    if(blinkl == 0){
      digitalWrite(LED, LOW);
    }else{
      digitalWrite(LED, HIGH);
    }
  }
  delay(100);
}
```

Thank you.

Avamander, 2020-06-03:
a) External regulator doesn't deliver enough current - decouple with electrolytic capacitors. 
b) External regulator is too noisy - decouple with ceramic capacitors. 
c) Receiver gets overloaded by the signal being too strong - increase distance or lower power

BJ4K3, 2020-06-03:
Thank you for you answer.
As you said, I discovered 10minutes ago that the problem was the signal saturation. 
Now the problem is that sometimes (60/70% of times) the receiver doesn't receive the data. Now I tried using in the receiver the NRF24L01 without PA (the first and small version with the printed antenna to be clear) and now all the data are received without problems. While in the transmitter I use the NRF24L01 with the PA.
Do you know why I had to use the PCB antenna version instead the PA version? If I have to keep this configuration the range will be narrowed?

Thank you.

Avamander, 2020-06-03:
> Do you know why I had to use the PCB antenna version instead the PA version?

The same answers as before really, the power consumption and the chances to overload a receiver are both higher. Respectively, the need to decouple is higher. 

In addition to that, there are a few very beefy nRF24 transmitters out there, strong emmissions near your power or data wires can exasperate any issues in your setup, shielding the transmitter and wires if they haven't been shielded yet, might be a good idea. 

The third possibility is that the PA module either has a counterfeit nRF24 or that there's some design issue, but it's hard to determine that based on comments.

BJ4K3, 2020-06-05:
Thank you for the answer, there're some methods to determine if this is a counterfeit nRF24 or not? I've just experimented again with both receiver and transmitter with the nRF24 with PCB antenna but they rarely exchange data. Did you see any issue in the sketches? Thanks.

BJ4K3, 2020-06-06:
Hi, I have just used 2 arduinos instead of one arduino nano and one esp32 and all worked fine. For the ESP32 I've edited the library changing the digitalIO SPI pins. I think that the problem is the digitalIO library itself. There're some trick to make it fully functional? Thanks.

Linux1231233, 2020-06-30:
Yes it is possible to get it functional. You just need to define pins to SPI. 

#define D5 14 // SCK to pin5 NRF
#define D6 12 // MISO to pin7 NRF
#define D7 13 // MOSI to pin6 NRF

TMRh20, 2020-08-11:
> there're some methods to determine if this is a counterfeit nRF24 or not?
Yeah, if you have it, its pretty much counterfeit. 

Any remaining issues here seem to be hardware related or with other software. Closing...

closed 2020-08-11 by TMRh20