nRF24/RF24 #601: [Question]GettingStarted crashes

KoMa1012, 2020-06-15:
Hello all,

I've been using the RF24 Lib for a while now with my Raspberry Pi 2B and my arduinos, it was working with GettingStarted and my own programm.
A While ago I upgraded to a NRF24L01+PA+LNA SMA on my Pi and switched over to an Arduino pro mini but keeping the cheap NRF24L01+ there. After that I had to reinstall my Pi (had too much unnecessary stuff installed and it was easier to erase and restart from scratch). After that there was 3 weeks break of doing nothing with the setup. I ran apt-get upgrade today and it was taking quite a while to finish, I rebooted the pi afterwards (just to make sure).
Today I started again and it seems like nothing is working, I run the gettingstarted on my arduino and on my pi. But it seems like the Pi is having trouble, I run "sudo ./gettingstarted" and sometimes it gets a message, sometimes not. After closing (ctrl+c) the programm and running "sudo ./gettingstarted" again it won't do anything, output:
`
pi@raspberrypi:~/rf24libs/RF24/examples_linux $ sudo ./gettingstarted
RF24/examples/GettingStarted/
================ SPI Configuration ================
CSN Pin          = CE0 (PI Hardware Driven)
CE Pin           = Custom GPIO22
Clock Speed      = 8 Mhz
================ NRF Configuration ================
STATUS           = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1     = 0xa8a8e1fc00 0xa8a8e1fcff
RX_ADDR_P2-5     = 0xc3 0xc4 0xc5 0xc6
TX_ADDR          = 0xa8a8e1fcff
RX_PW_P0-6       = 0x20 0x20 0x00 0x00 0x00 0x00
EN_AA            = 0x01
EN_RXADDR        = 0x03
RF_CH            = 0x4c
RF_SETUP         = 0x05
CONFIG           = 0x0e
DYNPD/FEATURE    = 0x00 0x00
Data Rate        = 1MBPS
Model            = nRF24L01+
CRC Length       = 16 bits
PA Power         = PA_HIGH

 ************ Role Setup ***********
Choose a role: Enter 0 for pong_back, 1 for ping_out (CTRL+C to exit)
>0
Role: Pong Back, awaiting transmission

Got payload(4) 139408...
Got payload(4) 139408...
^C
pi@raspberrypi:~/rf24libs/RF24/examples_linux $ sudo ./gettingstarted
RF24/examples/GettingStarted/
^C
pi@raspberrypi:~/rf24libs/RF24/examples_linux $ `

When I switch to transmit mode I get the message "RF24 HARDWARE FAIL: Radio not responding, verify pin connections, wiring, etc. failed." 

I'm following this wiring: [https://tutorials-raspberrypi.de/wp-content/uploads/Raspberry-Pi-nRF24L01-Steckplatine-600x496.png](url) assuming, that these are the VCC and GND pins.
[image]
I double checked the wiring (god knows how often) and I don't have an Idea what to do, to me it seems like everything is connected nicely, because I get nonzero values.

Do you guys have an Idea what I'm doing wrong here? I'm at a complete loss at the moment.

TMRh20, 2020-06-15:
> I'm following this wiring: https://tutorials-raspberrypi.de/wp-content/uploads/Raspberry-Pi-nRF24L01-Steckplatine-600x496.png assuming, that these are the VCC and GND pins.

It’s typically going to be the PA+LNA module. The RPi can’t source enough current, so a separate power supply would be required.

KoMa1012, 2020-06-15:
Just checked with two different NRF24L01+ modules until I double checked the soldering of the Arduino. There was a bad soldering point on this side, which makes me wonder why the arduino was reporting some data.
Please excuse my mistake!

closed 2020-06-15 by KoMa1012