nRF24/RF24 #605: RF24 with MegaCoreX and megaTinyCore

trusleymike, 2020-07-08:
Including the RF24 library in anything compiled for processors supported by MegaCoreX (MCUdude) and megaTinyCore (Spence Konde) throws an error relating to digitalWrite().  The code concerned is:

  https://github.com/nRF24/RF24/blob/af05e9d6d482ecc6aade40a8705b38f3b148c023/RF24.cpp#L14-L19
  
I enclose below a simple sketch which throws the error.

I understand the error relates to the use of the latest Arduino API (which MegaCoreX and megaTinyCore use).  However, the new API uses an enum for the second argument rather than an integer (as it used to), and this broke a lot of existing code out there.   I think that is the reason why both MCUdude and Spence Konde decided to ditch the new implementation and return to the classic one - to avoid breaking existing code.  Unfortunately this means that RF24 does not work with code for the latest generation of processors.

Can you suggest a workaround if you intend to stick to the enum method from the latest Arduino API?

------------------------------

Code and compiler output:
=================

#include "RF24.h"   // Library for nrf24L01+

void setup() {
}

void loop() {
}

C:\Users\Mike\Documents\Arduino\libraries\RF24-master\RF24.cpp: In function 'void digitalWrite(uint8_t, bool)':

C:\Users\Mike\Documents\Arduino\libraries\RF24-master\RF24.cpp:17:41: error: call of overloaded 'digitalWrite(uint8_t&, int)' is ambiguous

     digitalWrite(pin, value ? HIGH : LOW);
                                         ^
In file included from C:\Users\Mike\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.4\cores\coreX-corefiles/api/ArduinoAPI.h:52:0,

                 from C:\Users\Mike\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.4\cores\coreX-corefiles/Arduino.h:23,

                 from C:\Users\Mike\Documents\Arduino\libraries\RF24-master\RF24_config.h:60,

                 from C:\Users\Mike\Documents\Arduino\libraries\RF24-master\RF24.cpp:10:

C:\Users\Mike\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.4\cores\coreX-corefiles/api/Common.h:99:6: note: candidate: void digitalWrite(pin_size_t, uint8_t)

 void digitalWrite(pin_size_t pinNumber, uint8_t status);

      ^~~~~~~~~~~~

C:\Users\Mike\Documents\Arduino\libraries\RF24-master\RF24.cpp:15:13: note: candidate: void digitalWrite(uint8_t, bool)

 inline void digitalWrite(uint8_t pin, bool value)

TMRh20, 2020-07-08:
Looks like the original issue ( https://github.com/nRF24/RF24/issues/550 ) that initiated this code addition has been addressed in the Arduino Core API per [this commit](https://github.com/arduino/ArduinoCore-API/commits/9c2dbe25139cb68446b581ab505c0a4ab1b4bcdc/api/Compat.h)  

A bit of testing indicates that if I just remove the overload of digitalWrite - https://github.com/nRF24/RF24/blob/af05e9d6d482ecc6aade40a8705b38f3b148c023/RF24.cpp#L14-L19 

Then it compiles on both cores - Arduino Uno WiFi Rev2 (megaavr) and ATmega4809 (MegaCoreX)

So if you could just remove the overload code and test, I assume it should work fine, can you confirm?

trusleymike, 2020-07-08:
TMTh20: it does indeed!  I will close this issue.

Many thanks for you very speedy response.

closed 2020-07-08 by trusleymike

KishalayChakraborty, 2023-11-25:
It works. thanks