nRF24/RF24 #607: Using RF24 library on BCM2711 based Raspberry pi

Amir-HK, 2020-07-17:
Hi,

I want to use the RF24 library with Raspberry pi 4B, which is based on BCM2711, but the RF24 library use BCM2835 (.c and .h).
**how can I use this library with Raspberry pi 4B?**

Using the library without changes, I have faced some problems; for example, radio.write does not work!

I compared all of the registers with Arduino connected nrf24l01, which works correctly; the only difference is in the RF_SETUP register equal to 0x17. In contrast, on the Arduino side, the RF_SETUP register is 0x07, which means "force PLL lock signal" has been enabled in raspberry pi (based on datasheet explanation)!

I use Raspberry Pi OS, and connections are as below:
[image]

also, getstarted.cpp file is executed.

TMRh20, 2020-07-17:
It _should_ be working with RPi4, but the BCM2835 lib uses a bit of a workaround to support the RPi4.  You can try a lower SPI speed in case there is a problem with SPI, long wiring or something. `RF24 radio(22,0,1000000);`

I would also suggest trying with the SPIDEV driver, since it should pretty much work on all Linux platforms:
```
rf24libs/RF24:
./configure --driver=SPIDEV
make
sudo make install
```
Then recompile the example and test.

You might have to reload SPI after switching drivers:
```
sudo rmmod spi_bcm2711 (for my RPi B its spi-bcm2835)
sudo modprobe spi_bcm2711
```

I think it may be time to configure SPIDEV as the default driver, since it supports a wider range of hardware.
The radio constructor should be the same for both drivers `RF24 radio(22,0);`
Might have to get me one of those fancy RPI4s to play with sometime soon to see whats up with the BCM driver.

TMRh20, 2020-07-18:
First off, please don't open multiple issues for the same problem.

1. Per your other post, if the RPi can read data, the PLL lock issue may not be a problem. This would require some debugging of SPI transfers to see what is going on exactly.  
2. What kind of modules are you using? The PA+LNA high powered ones with antenna typically need a separate power supply to work with RPi. You can try setting `radio.setPALevel(RF24_PA_MIN);`
3. Typically you want a few meters distance between the modules for testing. Too close and there can be issues.
4. It sounds like everything is working for the most part, just some issues with radio communication. You can also try different channels in case of interference: `radio.setChannel(3);`

Amir-HK, 2020-07-19:
I checked the above points but my problem has not been solved.

**I am sure the problem is related to PLL_Lock bit in the RF_SETUP register, which is High!** 

I changed my NRF24l01 module (my module has not antenna and is 100m ranges module), and for first time testing, the PLL_Lock bit was Low and radio.write method work correctly on RPI4, but in the next time, PLL_Lock bit change to HIGH and radio.write does not work!

**So, can anyone tell me why PLL_Lock change in this Library?**

Thanks in advance.

TMRh20, 2020-07-21:
This is the first time this issue has come up, and I have no idea what is causing it. We just added a new function to the library to produce a constant carrier output, so if you update the library & rebuild/install, you should be able to call `radio.stopConstCarrier();` after radio.begin(). In theory that should manually disable the PLL lock configuration and work around the issue until we can figure out what is causing it.

closed 2020-07-21 by Amir-HK