nRF24/RF24 #612: [Question] Can this library work with the RF Nano?

FrightRisk, 2020-07-26:
There are a couple of versions of the "RF Nano" out there. A nano with an RF24 chip already on the board. The same code that works on a Nano with a separate RF24 works. They have their own "Mirf.h" library. Here are the pins. Could this cause a conflict in the library?

 D10 CE
 D11 MOSI
 D12 MISO
 D13 SCK

Here is an entire sketch if it helps. Very similar code, yet not really compatible with the RF24 library. I would hate to have to use a library other than RF24.

#include <SPI.h>
#include "Mirf.h"
#include "nRF24L01.h"
#include "MirfHardwareSpiDriver.h"
Nrf24l Mirf = Nrf24l(10, 9);
int value1,value2;

void setup()
{
    Serial.begin(9600);
    Mirf.spi = &MirfHardwareSpi;
    Mirf.init();
    //Set your own address (sender address) using 5 characters
    Mirf.setRADDR((byte *)"ABCDE"); 
}

void loop()
{
    Mirf.payload = sizeof(value1);
    Mirf.channel = 10;              //Set the channel used
    Mirf.config();
    if(Mirf.channel == 10)
    {
    Mirf.setTADDR((byte *)"FGHIJ");           //Set the receiver address
    value1 = 100;
    Mirf.send((byte *)&value1);                //Send instructions, send random number value
    Serial.print("Wait for sending.....");
    while (Mirf.isSending()) delay(1);        //Until you send successfully, exit the loop
    Serial.print("value1 Send success:");
    Serial.println(value1);
    delay(500);
    }
    Mirf.payload = sizeof(value2);
    Mirf.channel = 20;              //Set the channel used
    Mirf.config();
    if(Mirf.channel == 20)
    { 
    Mirf.setTADDR((byte *)"KLMNO");           //Set the receiver address
    value2 = 200;
    Mirf.send((byte *)&value2);                //Send instructions, send random number value
    Serial.print("Wait for sending.....");
    while (Mirf.isSending()) delay(1);        //Until you send successfully, exit the loop
    Serial.print("value2 Send success:");
    Serial.println(value2);
    delay(500);
    }
}

TMRh20, 2020-07-26:
> Could this cause a conflict in the library?

In this RF24 library? Nope.

> Very similar code, yet not really compatible with the RF24 library.

Why not? This library supports pretty much every configuration possible. 

When it comes down to it, this library can be made to work with pretty much any other nrf24l01 configuration, regardless of the software driving it. The mirf library has been around forever, but is not even available in the Arduino IDE, IMHO because it is so out of date. (See https://github.com/aaronds/arduino-nrf24l01/tree/master/Mirf)

All in all, the people making these devices should get their sh... stuff together and tell their users to use this library, since its going to have better support for more hardware and way more functionality. :)

FrightRisk, 2020-07-27:
I meant that the Mirf code is not compatibly with your commands. It isn't a 1 to 1 switchout and use the same commands. I would much rather use your library. I guess what I am trying to say is that I have sketches written using your library that work fine on Nanos with stand alone RF24 boards of different kinds. Those sketches are not working on the RF Nano. So I'm not sure where to start. Are there other settings I should be setting in software to account for this board? Internal wiring seems to be the same as an UNO. I am using the RF24.h and RFNetwork.h.  In reading your docs, I will try more debugging like the isChipConnected() method. Thanks for the help.

Avamander, 2020-07-27:
> I meant that the Mirf code is not compatibly with your commands. 

Yeah, there are some minor changes in function signatures. If you know what the Mirf function does, finding an equivalent in RF24 won't be difficult. 

> Are there other settings I should be setting in software to account for this board? 

Assuming it's using the regular SPI bus, then you probably only have to update the two pins given to the RF24 constructor, e.g. something like `RF24 radio(10, 9);` or `RF24 radio(9, 10);` instead of the current default of `RF24 radio(7,8);`.

TMRh20, 2020-07-27:
Can you link the version of the mirf library that you are using?

FrightRisk, 2020-07-27:
Well I feel stupid, but I was going by what the docs say are CE and CSN are for this RF Nano.  So I put in 10,9 and 9, 10.  I tried your excellent samples with the failure test and that got me further. So I changed your failure test program back to defaults of  7,8 and it worked! Is 7,8 a default for other systems?

TMRh20, 2020-07-27:
Something must be off with that result? Does it just print out OK or does it actually work? The mirf lib takes its constructor in the same way as RF24, with CE, CSN. The RF Nano hardware has the radio CE pin hardwired to digital pin 10, and CS hardwired to digital pin 9, meaning the RF24 constructor should probably be RF24 radio(10,9); exactly as it is with the RF Nano.

On AVR 328 devices (arduino nano), pin 10 is the 'hardware spi csn' pin, so that has usually been the default, but any pin can be used. It also happens that pins 9 & 10 are connected to the 16-bit timer PWM output. The **_only_** reason this library defaults to pins 7(CE), 8(CSN) is to make the timer pins available. Technically the radio CE & CSN pins can be connected to any digitalIO capable pins on the MCU, but on the RF Nano the radio is hardwired to the MCU, so you are stuck using whatever CE & CSN pins are wired up.

FrightRisk, 2020-07-28:
No, it works! What are the chances your random pins would be what this thing is hardcoded to?? I will take the magnifier and see if I can trace the traces. But your failure test lets me transmit from it and the other node sees it. Then I hit "r" to reverse the process and hit "t" on the node (A Mega running a board with an external antenna). Sends and receives. so I took my sketch with 6 Nanos (this is the only RF Nano I was seeing if I could replace the rest with), changed CE,CSN to 7,8 and everything works. I have an I2C expander board and all the servos work, another node has RGB leds and they change color. I knew if anything was working, it all would be working, but I tried it just because I didn't believe it was really working ;)

Avamander, 2020-07-28:
Good to know it works, marking the issue as solved.

closed 2020-07-28 by Avamander