nRF24/RF24 #614: [Question] stubs-hard.h missing

OLLYDOTDEV, 2020-08-05:
I have found an issue where I am missing stubs-hard.h 
does anyone know where I can download the lib for it 

image of the error
[image]

if anyone and help that would be great.

TMRh20, 2020-08-06:
Just a half-educated guess here, but thinking its going to be related to the line `-mfloat-abi=hard`
What is the hardware involved here, what is this running on?  

I would probably just try removing that mfloat-abi line from the makefile and see if it builds.
If not see https://stackoverflow.com/questions/49139125/fatal-error-gnu-stubs-soft-h-no-such-file-or-directory
You can also try specifying  ‘soft’ or ‘softfp’

OLLYDOTDEV, 2020-08-06:
Thanks a lot, mate. All I needed to do was set `-mfloat-abi=` to soft

closed 2020-08-06 by OLLYDOTDEV