nRF24/RF24 #615: Installing libary for Python 3

Fritzi16, 2020-08-10:
There is this linked guide for installation: http://tmrh20.github.io/RF24/Python.html
I tried to follow it and change the commands for python 3. I am on a Raspberry Pi 3 with a fresh installation of Raspberry Pi OS (Raspbian).So I used the following commands:

1.)
`sudo apt-get install python3-dev libboost-python-dev` //no libboost-python3-dev available

`sudo ln -s $(ls /usr/lib/arm-linux-gnueabihf/libboost_python-py3*.so | tail -1) /usr/lib/arm-linux-gnueabihf/libboost_python3.so `//didnt work at all. Always said "Data is already existing 

2.)
`sudo apt-get install python3-setuptools` //worked fine

3.)
`./setup.py build` ### **OR** `python3 setup.py build` //both brougt errors saying that there is no such data or directory. after some digging and trying something worked

4.)
`sudo ./setup.py install`  ### **OR** `python3 setup.py build` //also the same, worked after trying for a long time and digging in places

In the end, I am able to run following code with Python 2 but not with Pyhton3. 

`import time
import RPi.GPIO as GPIO
from RF24 import *` 

So in the end it was not a successfull installation... Could it be, that something changed over the time with the OS and thats the reason for failing?

Please help. Already trying everything for days -.-

Best regards, Fritzi16

TMRh20, 2020-08-10:
Here is what works for me on an older RPi3 that is pretty much up to date:

1. `sudo apt-get install python3-dev libboost-python-dev`
2. Issue#1, had to modify the line to:
`sudo ln -s $(ls /usr/lib/arm-linux-gnueabihf/libboost_python3-py3*.so | tail -1) /usr/lib/arm-linux-gnueabihf/libboost_python3.so`
got:
`ln: failed to create symbolic link '/usr/lib/arm-linux-gnueabihf/libboost_python3.so': File exists`
This is normal since I've run through this before
3. `sudo apt-get install python3-setuptools` 
4. `python3 setup.py build` I'm guessing your issue here was not building & installing the library in c++ first? This needs to be clearly stated in the python section.
5. `sudo python3 setup.py install`
6. Running the example with python3 gave error about RPi.GPIO, so `sudo apt-get install python3-dev python3-rpi.gpio`
7. `sudo python3 pingpair_dyn.py` Up and running..

So the takeaways here are I think I need to update the `sudo ln -s` linking line and link to the c++ build instructions in the python wrapper info?

Fritzi16, 2020-08-10:
Thank you sooooooo MUCH!!!!!!

deleted everyting again, used following commands to get the libary:

```
mkdir ~/rf24libs
cd ~/rf24libs
git clone https://github.com/TMRh20/RF24
cd ~/rf24libs/RF24
./configure
sudo make install
```
And after that everything you wrote worked perfectly fine! But I had to add a `cd pyRF24/` between your 3. and 4.

Thanks again! In the end I feel like a dumb idiot.

So to summ it up to a clear solution:

```
mkdir ~/rf24libs
cd ~/rf24libs
git clone https://github.com/TMRh20/RF24
cd ~/rf24libs/RF24
./configure
sudo make install
sudo apt-get install python3-dev libboost-python-dev
sudo ln -s $(ls /usr/lib/arm-linux-gnueabihf/libboost_python3-py3*.so | tail -1) /usr/lib/arm-linux-gnueabihf/libboost_python3.so
sudo apt-get install python3-setuptools
cd pyRF24/
python3 setup.py build
sudo python3 setup.py install
sudo apt-get install python3-dev python3-rpi.gpio
```
and the libary should be installed and work with Python 3

closed 2020-08-10 by Fritzi16

DIY89, 2023-09-22:
Best hint for me: https://github.com/andrewssobral/bgslibrary/issues/96