nRF24/RF24 #616: BLE Beacon example

Avamander, 2020-08-20:
It would be nice to have a BLE beacon example using this library. 

Here's a technical description of how it can be done: http://dmitry.gr/index.php?r=05.Projects&proj=11.%20Bluetooth%20LE%20fakery

TMRh20, 2020-08-20:
Are you familiar with https://github.com/floe/BTLE ? 

The code looks pretty similar to the link you posted, but most of the hard stuff is done in the library, so the user API is relatively simple. Its based on this library, so if we wanted to adapt an example, I think it should be relatively easy.

Avamander, 2020-08-20:
Hmm, I wasn't familiar with it.

I guess a really minimal example and a link to the BTLE library would be the best solution. Just to show what can be done and how to potentially do it even nicer.

TMRh20, 2020-08-21:
That sounds reasonable, show how to configure the radio and do it manually while linking the BTLE lib. I was intending to get into the existing examples recently for some review/cleanup, but got stuck digging into issues and fixing bugs instead. (Pretty happy with the results tho) I prob won't be able to help much on this for a while.

2bndy5, 2020-09-17:
there are several proof-of-concept examples. Although I keep coming back to [this blog post](https://d.lij.uno/misc-nrf24-ble.html) (a little more visual) which includes a link to [the source code on github](https://github.com/hwsink/nRF24_BLE/blob/master/Arduino/nRF24_BLE_advertizer_demo/nRF24_BLE_advertizer_demo.ino) that was used in the post. However, that code doesn't use this library as a dependency, rather it uses a minimal structure to control the nRF24L01 with an Arduino. In the way of adding an example to this library as an "official" proof-of-concept, were you thinking about an advertising example or a listening example (or both)?

2bndy5, 2020-09-18:
I know this is a little off-topic, but TBH I'd be more interested to see a Gazelle-based implementation that might act as a OTA bridge to the micro-bit (or other nRF58xx-based devices), specifically an example towards the [makecode.com's radio blocks](https://makecode.microbit.org/reference/radio). However that is a different issue and may require a separate repo due to the [pairing](https://developer.nordicsemi.com/nRF5_SDK/nRF51_SDK_v5.x.x/doc/5.2.0/html/a00141.html) & [link layer](https://developer.nordicsemi.com/nRF5_SDK/nRF51_SDK_v5.x.x/doc/5.2.0/html/a00140.html) differences

2bndy5, 2020-09-29:
I've been adding this concept to my CircuitPython library and I noticed that the BTLE library that @TMRh20 mentioned does not offer broadcasting the "TX power level" (GAP assigned number `0x0A`) which only occupies 3 bytes (including pa_level value) and could be used by BLE scanning applications to determine the relative proximity ("perceived pa_level" - "broadcasted pa_level") to the broadcasting nRF24L01. It might be cool to work this into the example for this library.

So far I've managed to max out the FIFO when transmitting the device name as "nRF24L01", the "TX power level", and a battery charge (as integer percentage). It should be noted that iPhones may require a different PDU type (`0x40`) than Androids expect (`0x42`), but that's a bug-fix I'm waiting for someone to confirm. Literally any other device will pick up advertisements using either PDU type (Windows PC, MAC desktop, & Chromebook verified).

2bndy5, 2020-10-20:
@Avamander would you be ok if we just add a link to the [BTLE library](https://github.com/floe/BTLE) in the docs (somewhere)? Because there are [issues with certain iPhones](https://github.com/floe/BTLE/issues/3) that I would like to avoid on this repo. Also, that library does more than I would be willing to support (nudge to the #658 about quantity harder to maintain), namely receiving BLE payloads (which involves trying to de-whiten, swap bits, & decipher from BLE data structures).

p.s. Does it strike anyone else as weird that RF24Network layer doesn't employ channel hopping like BLE and WiFi? Or is it reliable enough on 1 channel?