nRF24/RF24 #621: [Question] Why powerUp() is called in startListening()?

Seoho, 2020-08-27:
Usually, RF module is not in powerDown() mode when we switch to startListening() from stopListening(). Why powerUp() should be called in startListening() function? 

void RF24::startListening(void)
{
    #if !defined(RF24_TINY) && !defined(LITTLEWIRE)
    powerUp();
    #endif
    /* Notes Once ready for next release
    * 1. Can update stopListening() to use config_reg var and ack_payloads_enabled var instead of SPI rx/tx
    * 2. Update txDelay defaults: 240 for 2MBPS, 280 for 1MBPS, 505 for 250KBPS per initial testing
    * 3. Allows time for slower devices to update with the faster startListening() function prior to updating stopListening() & adjusting txDelay
    */
    config_reg |= _BV(PRIM_RX);
    write_register(NRF_CONFIG,config_reg);
    write_register(NRF_STATUS, _BV(RX_DR) | _BV(TX_DS) | _BV(MAX_RT));
    ce(HIGH);
    // Restore the pipe0 adddress, if exists
    if (pipe0_reading_address[0] > 0) {
        write_register(RX_ADDR_P0, pipe0_reading_address, addr_width);
    } else {
        closeReadingPipe(0);
    }

    if(ack_payloads_enabled){
        flush_tx();
    }

}

TMRh20, 2020-08-28:
Good question, I was thinking about this recently when reviewing the
library, and decided not to think about it anymore.

The only real reasons are because it's kind of a remnant from the past when
the lib would power down the radio after every call to write()
Also to preserve previous 'expected' behaviour.

I don't know, the radio is powered up on begin() so removing it would
probably have very little impact on users.

2bndy5, 2020-10-02:
removing it would also need a note in the docs on `startListening()` to warn users that calling `powerDown()` after `stopListening()` would require them to also call `powerUp()` before (not after) calling `startListening()` again. Obviously the radio needs to be powered up to enter RX mode, but sequentially speaking, the radio needs to be powered up before `startListening()` asserts the CE pin. I understand the concern about this issue as powering up the radio requires an additional wait time (see [table 16 in datasheet](http://tmrh20.github.io/RF24/tmrh20/nRF24L01P_datasheet_v1.pdf#G1161881) about mandatory wait times), especially since this library has no internal attribute to keep track of the power mode without directly writing/reading the PWR_UP flag (in the CONFIG register) to manipulate the radio's operating mode. Currently `powerUp()` reads and then writes this flag before waiting the mandatory time (if it needed to power up) which amount to a total of 15 ms from power down mode (5 ms if already powered up), but I'm not sure why this library waits a whole 5 ms when going from power down to Standby-I. Remember also that each SPI transaction waits 5 ms (`csDelay`) for stability reasons, and `startListening()` also has to wait 130 microseconds for the radio to enter "active RX" mode after asserting the CE pin to do so.

wmarkow, 2020-10-02:
>but I'm not sure why this library waits a whole 5 ms when going from power down to Standby-I.

I do not understand this question. Is your concern that the 5ms is to small or to big or that it should never wait? I think it is just an extra safty mechanism, so when the ```powerUp``` method is done then we have sure that the chip is in standby mode and we can call ```startListening```. For example this piece of user's code will mostly work:

```
rf24.powerUp();
rf24.startListening();
```
Without this delay in ```powerUp``` the ```startListening()``` may fail as the chip is not in standby yet. Does it make sense?

2bndy5, 2020-10-02:
[image]

sorry I should specified that it is too long. I agree that we should be waiting the mandatory time for the radio to establish StandBy-I mode. I think 1.5 ms is a safer bet, but 2 ms at most is still enough.

wmarkow, 2020-10-02:
Ok, I have it from Git history:
 * with this commit 2f7e4489bad411ee79f06 the delay was initially set to 2ms
 * but later with this commit 3455e489f4c88504bd6 it was increased to 5ms because of some kind hanging condition

Please also read log messages for those two commits.

2bndy5, 2020-10-02:
> I wouldn't be surprised if some batch of original chips don't stick to the docs. 

The clones don't (bk242x or AKA RFM7x chips). The counterfeits really don't (Si24R1 targeted). Thanks for the git blame. I also think I read somewhere about the poor precision on some Arduino clones in relation to the external (or even internal -- if they dared) crystal oscillators when using `delay()` & `delayMicroseconds()`, but I can't remember where (so long ago).

2bndy5, 2020-10-17:
>especially since this library has no internal attribute to keep track of the power mode without directly writing/reading the PWR_UP flag (in the CONFIG register) to manipulate the radio's operating mode

I just realized that this library actually does use an internal [`config_reg`](https://github.com/nRF24/RF24/blob/a3f9ace8c530138bd10211a08bc498f05187edb7/RF24.h#L92) variable to avoid doing a `delay(5)` for every call to [`powerUp()`](https://github.com/nRF24/RF24/blob/a3f9ace8c530138bd10211a08bc498f05187edb7/RF24.cpp#L758). This means it doesn't hurt to leave `startListening()` the way it is, especially if the MCU calls `powerDown()` before calling `startListening()`. A note in the docs about needing to make sure the radio is powered up (before calling `startListening()`) up would be unnecessary due to current implementation.

2bndy5, 2021-04-22:
closing this as it has been thoroughly discussed

closed 2021-04-22 by 2bndy5