nRF24/RF24 #622: [Question] compile error

OLLYDOTDEV, 2020-09-05:
Hey 
wondering if anyone is able to help me 

```
#include <cstdlib>
#include <iostream>
#include <sstream>
#include <string>
#include <unistd.h>
#include <RF24/RF24.h>

void setup(void) {
  delay(2000);
  radio.begin(); 
  
  radio.setChannel(125);
  radio.setPALevel(RF24_PA_LOW);       
  radio.setDataRate(RF24_250KBPS);
  radio.setAutoAck(false);                    
  
  radio.setRetries(15, 15);    
  radio.setCRCLength(RF24_CRC_8);
  
  radio.openWritingPipe(pipes[1]);
  radio.openReadingPipe(1, pipes[0]);

  radio.startListening();           
  radio.printDetails();   
  radio.powerUp();    
}

```

here is the relevant code for this 

I have already installed the NRF24 library via ./install.sh from http://tmrh20.github.io/RF24Installer/RPi/install.sh 

[image]

my guess is something to do with not finding the RF24.h file 
I am using `g++ -marm -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -mfloat-abi=soft -Ofast -Wall -pthread   BULK_READ.cpp -o BULK_READ` to compile the code 
if you need more information just say so. 

any help will be appreciated

TMRh20, 2020-09-05:
add -lrf24-bcm

OLLYDOTDEV, 2020-09-06:
Well I tried adding that still didn't work 

```
# g++ -marm -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -mfloat-abi=soft -Ofast -Wall -lrf24-bcm -pthread BULK_READ.cpp -o BULK_READ
/usr/bin/ld: /tmp/ccwsBp3o.o: in function `setup()':
BULK_READ.cpp:(.text+0x10): undefined reference to `__msleep'
/usr/bin/ld: BULK_READ.cpp:(.text+0x18): undefined reference to `RF24::begin()'
/usr/bin/ld: BULK_READ.cpp:(.text+0x24): undefined reference to `RF24::setChannel(unsigned char)'
/usr/bin/ld: BULK_READ.cpp:(.text+0x34): undefined reference to `RF24::setPALevel(unsigned char, bool)'
/usr/bin/ld: BULK_READ.cpp:(.text+0x40): undefined reference to `RF24::setDataRate(rf24_datarate_e)'
/usr/bin/ld: BULK_READ.cpp:(.text+0x4c): undefined reference to `RF24::setAutoAck(bool)'
/usr/bin/ld: BULK_READ.cpp:(.text+0x5c): undefined reference to `RF24::setRetries(unsigned char, unsigned char)'
/usr/bin/ld: BULK_READ.cpp:(.text+0x68): undefined reference to `RF24::setCRCLength(rf24_crclength_e)'
/usr/bin/ld: BULK_READ.cpp:(.text+0x78): undefined reference to `RF24::openWritingPipe(unsigned long long)'
/usr/bin/ld: BULK_READ.cpp:(.text+0x8c): undefined reference to `RF24::openReadingPipe(unsigned char, unsigned long long)'
/usr/bin/ld: BULK_READ.cpp:(.text+0x94): undefined reference to `RF24::startListening()'
/usr/bin/ld: BULK_READ.cpp:(.text+0x9c): undefined reference to `RF24::printDetails()'
/usr/bin/ld: BULK_READ.cpp:(.text+0xa8): undefined reference to `RF24::powerUp()'
/usr/bin/ld: /tmp/ccwsBp3o.o: in function `loop()':
BULK_READ.cpp:(.text+0xf0): undefined reference to `RF24::read(void*, unsigned char)'
/usr/bin/ld: BULK_READ.cpp:(.text+0x100): undefined reference to `RF24::available()'
/usr/bin/ld: BULK_READ.cpp:(.text+0x250): undefined reference to `__msleep'
/usr/bin/ld: /tmp/ccwsBp3o.o: in function `RF24::~RF24()':
BULK_READ.cpp:(.text._ZN4RF24D2Ev[_ZN4RF24D5Ev]+0x20): undefined reference to `SPI::~SPI()'
/usr/bin/ld: /tmp/ccwsBp3o.o: in function `RF24::~RF24()':
BULK_READ.cpp:(.text._ZN4RF24D0Ev[_ZN4RF24D5Ev]+0x20): undefined reference to `SPI::~SPI()'
/usr/bin/ld: /tmp/ccwsBp3o.o: in function `_GLOBAL__sub_I_radio':
BULK_READ.cpp:(.text.startup+0x5c): undefined reference to `RF24::RF24(unsigned short, unsigned short, unsigned int)'
collect2: error: ld returned 1 exit status
```
in short same error.
i have also tried added -L  `g++ -marm -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -mfloat-abi=soft -Ofast -L/usr/local/include/RF24 -L/usr/local/lib -lrf24 -pthread BULK_READ.cpp -o BULK_READ
`
as I thought that I might not be finding the library and it still didn't work.

so yeah...

OLLYDOTDEV, 2020-09-06:
well with the tag you gave me i was able to find the resources to get it to work
`g++ -marm -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -mfloat-abi=soft -Ofast -Wall -pthread  -I/usr/local/include/RF24/.. -I.. -L/usr/local/lib BULK_READ.cpp -lrf24 -o BULK_READ
`

Keep up the great work

closed 2020-09-06 by OLLYDOTDEV