nRF24/RF24 #623: radio.setAutoAck(false) is preventing my receiver from receiving anything

amdkt7, 2020-09-07:
I have several receivers, some are closer than others and the one furthest away misses transmissions when the other closer receivers are also listening. I wanted to disable the ACK from the closer receivers so that only the furthest receiver could send the ACK when it did get the data. But when I disable AutoAck then I don't receive anything on that receiver. Is this a bug, or am I doing something wrong?

I'm running Arduino on a Atmel 328p micro at 8MHZ, 3.3 volts. I am using the latest library too.

my code:

#include <nRF24L01.h>
#include <RF24.h>
#include <printf.h>

boolean newData = false;//used to verify that data has been received
char data[14]; //holds data as received from the radio
byte radioAddress[5] = {'C', 'S', 'C', 'A', 'A'};

RF24 radio(6, 7);//CE,CSN

void setup() {
  Serial.begin(115200); 
  printf_begin();
  radio.begin();
  radio.setAutoAck(false);
  radio.openReadingPipe(0, radioAddress);
  radio.setPALevel(RF24_PA_MIN);
  radio.setDataRate( RF24_250KBPS );
  radio.startListening();
  radio.printDetails();
}

void loop() {
  getData();
  if (newData == true) {
    Serial.println(data);
    newData = false;
  }
}

void getData() {
  if ( radio.available() ) {
    radio.read( &data, sizeof(data) );
    newData = true;
  }
}

TMRh20, 2020-09-08:
I think you are just misunderstanding/misusing the ESB/AA functionality. With AA its generally 1->1 communication per address. Disable it on at least one pipe on all devices for multicasting.

One of these two links should explain it further:
https://tmrh20.blogspot.com/2016/08/rf24-addressing-review-of-nrf24l01.html
http://tmrh20.github.io/RF24/tmrh20/nRF24L01_datasheet_v2.pdf

amdkt7, 2020-09-08:
I will dig into those links. I looked at the top link already and it seems to be the best explanation I have seen yet. Thanks for the assistance. I will read it all as soon as possible.

 
 

https://tmrh20.blogspot.com/2016/08/rf24-addressing-review-of-nrf24l01.html
http://tmrh20.github.io/RF24/tmrh20/nRF24L01_datasheet_v2.pdf

amdkt7, 2020-09-09:
So if I need AA I need to send to each radio one at a time, each receiver with a different address/pipe? How can I setup the TX to send to several RX?

TMRh20, 2020-09-09:
Please don't take this the wrong way, but this is not a tech. support forum, its for library issues, bugs, development etc.

In short, use a single address for all devices & disable auto-ack. All the required info and more is right there.

closed 2020-09-09 by TMRh20