nRF24/RF24 #624: [Question] how to incorporate in C-project?

focussing, 2020-09-09:
Hello,
I have a project which consists only of C files, and I need to incorporate your beautiful C++ library.
What would be the best way to do this?

Looking forward to hearing from you.

Avamander, 2020-09-09:
You would have to rewrite it in C, unfortunately. There have been efforts at C-ifying the library, but it's a lot of work and there's not too much demand. You can see the previous effort here: https://github.com/nRF24/RF24/pull/330 Maybe you could salvage it and make a new PR?

focussing, 2020-09-14:
Thanks for answering.
I took an IDE and simply put all .c and .cpp files in one directory. Then `make` creates a well working executable.

closed 2020-09-14 by focussing