nRF24/RF24 #625: Raspberry pi 3 B TMRh20 RF24 RX_ADDR_P0-1 wrong Address

bprovvis, 2020-09-11:
hello
1-my sketch get_start return wrong RX_ADDR_P0-1
	after radio.begin()
	RX_ADDR_P0-1	 = 0xf3f3f3f3f3 0xe1e1e1e1e1
	after
	        radio.openWritingPipe(pipes[1]);
	        radio.openReadingPipe(1, pipes[0]);
	
	RX_ADDR_P0-1	 = 0x3232372719 0x3232372718
	
	instead of 	RX_ADDR_P0-1	 = 0x65646f4e31 0x65646f4e32 - Node1 Node2
2-    radio.begin() return false
bool RF24::begin(void)
.
.
     write_register(NRF_CONFIG, (_BV(EN_CRC) | _BV(CRCO)) );    -- 1100 ?
     config_reg = read_register(NRF_CONFIG);
  
     printf(("config_reg after read_register :%d \n",config_reg);
     config_reg value = 6  110 instead of  1100
     powerUp();
     return config_reg == (_BV(EN_CRC) | _BV(CRCO) | _BV(PWR_UP)) ? true : false; --return false

see details problems_RF24.txt
[problems_RF24.txt](https://github.com/nRF24/RF24/files/5207668/problems_RF24.txt)

help me please

TMRh20, 2020-09-12:
Standard troubleshooting, likely the module is not connected properly.
See:
https://github.com/nRF24/RF24/issues/598
https://github.com/nRF24/RF24/issues/532

bprovvis, 2020-09-18:
hi
I had to do several tests
1-I checked the connections to the module on Rpi OK OK
2- i tried on Rpi RF24 radio(22,0);------>radio(22,0,8000000);
    and RX_ADDR_P0-1 is OK

uint64_t pipes = { 0xF0F0F0F000LL};

 ================ SPI Configuration ================
CSN Pin  	 = 0
CE Pin  	 = Custom GPIO22
SPI Speedz	 = 8 Mhz
================ NRF Configuration ================
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0xf0f0f0f000 0xf0f0f0f000
RX_ADDR_P2-5	 = 0xc3 0xc4 0xc5 0xc6
TX_ADDR		 = 0xf0f0f0f000
RX_PW_P0-6	 = 0x20 0x20 0x00 0x00 0x00 0x00
EN_AA		 = 0x00
EN_RXADDR	 = 0x03
RF_CH		 = 0x4c
RF_SETUP	 = 0x21
CONFIG		 = 0x0e
DYNPD/FEATURE	 = 0x00 0x00
Data Rate	 = 250KBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_MIN

3-  master Rpi --->slave Arduino

|------  - |          radio.write                                
| Rpi       |                                              
|MASTER|       send to slave 0 "Message to slave 0"
|---------|                                              

|----------|         radio.read                           
| Arduino |                                              
| Slave 0  |         receive "Message to slave 0" OK   
|----------|                                              

|----------|         radio.write to MASTER                
| Arduino|                                              
| Slave 0  |         send to MASTER "00000,0000,0076,50"
|----------|                                              

|---------|          radio.read                                
| Rpi       |                                                    
|MASTER|       receive from slave 0 "88888>8888>88??>?8"     error on receive
|------     |                                              

slave send "0" 00110000 master receive "8" 00111000
slave send "," 00101100 master receive ">" 00111110
slave send "76" 00110111 00110110 master receive "??" 00111111 00111111
slave send "50" 00110101 00110000 master receive "?8" 00111111 00111000

I do not understand why
i tried  radio.disableCRC()  but the result is the same

thanks for an answer

Avamander, 2020-09-18:
Does this occur with unmodified example sketches?

bprovvis, 2020-09-18:
This test 
sketch on raspberry 
[RF24_MASTER.txt](https://github.com/nRF24/RF24/files/5245636/RF24_MASTER.txt)
schetch on arduino nano
[SLAVE_0.txt](https://github.com/nRF24/RF24/files/5245664/SLAVE_0.txt)

**log on MASTER**

Begin() OK 
================ SPI Configuration ================
CSN Pin  	 = 0
CE Pin  	 = Custom GPIO22
SPI Speedz	 = 8 Mhz
================ NRF Configuration ================
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0xf0f0f0f000 0xf0f0f0f000
RX_ADDR_P2-5	 = 0xc3 0xc4 0xc5 0xc6
TX_ADDR		 = 0xf0f0f0f000
RX_PW_P0-6	 = 0x20 0x20 0x00 0x00 0x00 0x00
EN_AA		 = 0x00
EN_RXADDR	 = 0x03
RF_CH		 = 0x4c
RF_SETUP	 = 0x21
CONFIG		 = 0x0e
DYNPD/FEATURE	 = 0x00 0x00
Data Rate	 = 250KBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_MIN
RF24 connesso 
Write ok...to Slave 0 Conteggio OK 1 
Ricevuto risposta ....88888>8888>88??>?8

**log on SLAVE_0**

RF24 Connesso
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0xf0f0f0f000 0xf0f0f0f000
RX_ADDR_P2-5	 = 0xc3 0xc4 0xc5 0xc6
TX_ADDR		 = 0xf0f0f0f000
RX_PW_P0-6	 = 0x20 0x20 0x00 0x00 0x00 0x00
EN_AA		 = 0x00
EN_RXADDR	 = 0x02
RF_CH		 = 0x4c
RF_SETUP	 = 0x21
CONFIG		 = 0x0f
DYNPD/FEATURE	 = 0x00 0x00
Data Rate	 = 250KBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_MIN
Ricevuto....Messaggio inv. allo slave 0
Ora invio...00000,0000,0076,50

thanks

bprovvis, 2020-09-21:
I redid the wiring on Rpi
i removed  GPIO Breakout for Raspberry 
I directly connected the module nrf24l01 to rpi
 receive from slave 0 "88888>8888>88??>?8" error on receive RESOLVED !!!!!!
thanks

2bndy5, 2020-09-21:
Excellent! Overcoming problems definitely makes the achievement that much sweeter.

closed 2020-09-21 by 2bndy5