nRF24/RF24 #628: Raspberry pi 4 not supported

BelhassenHamdi, 2020-09-17:
Communication is not established correctly, digging more into details I found that the RF_SETUP is not set correctly and few other constant parameters keep changing during listening and writing which means the two peers are not able to reach each others. Also setting up manually the PA power and the Data rate is not really working.

Used both python and C++ example codes for the pingpair and its also not working (you can use it to reproduce the error).
I am working on docker but was able to reproduce the error on the host too.

for communication I am using raspberry pi 4 and arduino nano but tried with double Raspi 4 didn't work. Same code works for me in raspi 3 and 2.

2bndy5, 2020-09-17:
> I found that the RF_SETUP is not set correctly and few other constant parameters keep changing during listening and writing

Could you expand on this?

BelhassenHamdi, 2020-09-17:
used setup:
```
radio.begin()
self.network.begin(90, master_node)
radio.setDataRate(RF24_1MBPS)
radio.enableAckPayload()
radio.setRetries(5, 15)
radio.setPALevel(RF24_PA_HIGH)
radio.setCRCLength(RF24_CRC_16)
```

code:
```
While True:
      *********** some code *********
      network.update()
      radio.printDetails()
      *********** some code *********
```
The rest of the code is very simple listen to the network read message and decode it.

output:
```
Radio Network Started...
================ SPI Configuration ================
CSN Pin  	 = 0
CE Pin  	 = Custom GPIO25
SPI Speedz	 = 10 Mhz
================ NRF Configuration ================
STATUS		 = 0x07 RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=3 TX_FULL=1
RX_ADDR_P0-1	 = 0xe6e6e6e6e1 0xe6e6e6e61e
RX_ADDR_P2-5	 = 0x19 0xe7 0x1f 0xf1
TX_ADDR		 = 0xe6e6e61ff1
RX_PW_P0-6	 = 0x10 0x10 0x10 0x10 0x10 0x10
EN_AA		 = 0x0f
EN_RXADDR	 = 0x00
RF_CH		 = 0x2d
RF_SETUP	 = 0x02
CONFIG		 = 0x07
DYNPD/FEATURE	 = 0x0f 0x03
Data Rate	 = 1MBPS
Model		 = nRF24L01
CRC Length	 = 16 bits
PA Power	 = PA_MIN
RF24 HARDWARE FAIL: Radio not responding, verify pin connections, wiring, etc.
RF24 HARDWARE FAIL: Radio not responding, verify pin connections, wiring, etc.
failed.
================ SPI Configuration ================
CSN Pin  	 = 0
CE Pin  	 = Custom GPIO25
SPI Speedz	 = 10 Mhz
================ NRF Configuration ================
STATUS		 = 0x07 RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=3 TX_FULL=1
RX_ADDR_P0-1	 = 0xe6e6e6e6e1 0xe6e6e6e61e
RX_ADDR_P2-5	 = 0x19 0xe7 0x1f 0xf1
TX_ADDR		 = 0xe6e6e61ff1
RX_PW_P0-6	 = 0x10 0x10 0x10 0x10 0x10 0x10
EN_AA		 = 0x03
EN_RXADDR	 = 0x00
RF_CH		 = 0x2d
RF_SETUP	 = 0x08
CONFIG		 = 0x03
DYNPD/FEATURE	 = 0x0f 0x03
Data Rate	 = 2MBPS
Model		 = nRF24L01
CRC Length	 = 8 bits
PA Power	 = PA_HIGH
Tx failed
failed.
================ SPI Configuration ================
CSN Pin  	 = 0
CE Pin  	 = Custom GPIO25
SPI Speedz	 = 10 Mhz
================ NRF Configuration ================
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0xccccccccc3 0xcccccccc3c
RX_ADDR_P2-5	 = 0x33 0xce 0x3e 0xe3
TX_ADDR		 = 0xcccccc3ee3
RX_PW_P0-6	 = 0x20 0x20 0x20 0x20 0x20 0x20
EN_AA		 = 0x06
EN_RXADDR	 = 0x01
RF_CH		 = 0x5a
RF_SETUP	 = 0x18
CONFIG		 = 0x03
DYNPD/FEATURE	 = 0x1f 0x06
Data Rate	 = 2MBPS
Model		 = nRF24L01+
CRC Length	 = 8 bits
PA Power	 = PA_MIN
Tx failed
failed.
================ SPI Configuration ================
CSN Pin  	 = 0
CE Pin  	 = Custom GPIO25
SPI Speedz	 = 10 Mhz
================ NRF Configuration ================
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0xccccccccc3 0xcccccccc3c
RX_ADDR_P2-5	 = 0x33 0xce 0x3e 0xe3
TX_ADDR		 = 0xcccccc3ee3
RX_PW_P0-6	 = 0x20 0x20 0x20 0x20 0x20 0x20
EN_AA		 = 0x06
EN_RXADDR	 = 0x01
RF_CH		 = 0x5a
RF_SETUP	 = 0x18
CONFIG		 = 0x03
DYNPD/FEATURE	 = 0x1f 0x06
Data Rate	 = 2MBPS
Model		 = nRF24L01+
CRC Length	 = 8 bits
PA Power	 = PA_MIN
.
.
.
```

As you can see everything changes from one iteration to the next.

2bndy5, 2020-09-18:
I see what you mean; that's strange. You didn't mention that you were using the [RF24Network library](https://github.com/nRF24/RF24Network). I'm not familiar with that library, but this issue might belong on the [RF24Network repo](https://github.com/nRF24/RF24Network/issues).

TMRh20, 2020-09-18:
What driver are you using for RF24? Have you tried using SPIDEV? 
```
./configure --driver=SPIDEV;make;sudo make install
```

BelhassenHamdi, 2020-09-18:
@TMRh20  I am using RPi but I think I tried them all and didn't work I will recheck.

@2bndy5 I thinks it's more related to the RF24 than the RF24Network since Network uses RF24.

2bndy5, 2020-09-18:
ugh. it feels like the python wrapper for this library has been rather consistently plagued with linking errors. I do plan on porting the RF24Network to pure python for availability via pip3, but that idea hasn't escaped my dreams yet (have to familiarize myself with the C++ source).

BelhassenHamdi, 2020-09-18:
> ugh. it feels like the python wrapper for this library has been rather consistently plagued with linking errors. I do plan on porting the RF24Network to pure python for availability via pip3, but that idea hasn't escaped my dreams yet (have to familiarize myself with the C++ source).

That would be very nice to see.

2bndy5, 2020-09-19:
This issue has turned into a "python wrapper not working on RPi4" for me, so bare with me as this post may be a little off topic.

@BelhassenHamdi I've been trying to look into [the error you posted](https://github.com/nRF24/RF24/issues/628#issuecomment-694707123), but I keep asking myself: "what cmd did you use to get that error?" I personally don't have a RPi4, so the python wrapper checks out for me (cannot reproduce error). I also never had the privilege of playing with an Odroid, so that issue you referenced is of little help (right now). The [cmd that @TMRh20 suggested](https://github.com/nRF24/RF24/issues/628#issuecomment-694614965) would not have output anything related to Python's dist-packages directory. I'm guessing the error came from `python3 setup.py build` or `python3 setup.py install`. (which should be equivalent to `pip3 install .` in the directory that setup.py exists). Could you also post what version of libboost is installed on RPi4 (`apt-cache policy libboost-python-dev` should get that info)?

It is possible that this error could be a libboost-python-dev problem. @TMRh20 The latest version (v1.71) does not have cmake files for python2 anymore... Rightfully so, python2 hit end-of-life in January 2020; pip has been warning users for over a year now, but most debian-based distros still include python2 and probably will for some time. My RPi2 installed libboost-python-dev v1.67 which does have cmake files for python2.

I've been digging through the `pyRf24` directory, and pylint keeps warning me about polymorphism errors (`distutils.unixccompiler` and `pyRF24.crossunixccompiler` modules), which might be related to general scoping, or python's not-so-strict policy on singletons, or simply artifacts from older versions of python (2 or 3). I should note that my local python version is v3.7 which seems to be the same on @BelhassenHamdi RPi4.

BelhassenHamdi, 2020-09-19:
@2bndy5 thanks for the time invested, I also had the same doubts in the begining that it could be related to the python wrapper, so as I mentioned I went one step lower and tried the original implemention ( C++) of the example codes pingpair, helloworld_rx/_tx but it still has the same issue. So obviously this excludes the python wrapper hypothesis.

2bndy5, 2020-09-19:
Quite a relief. Thank you for clarifying.

BelhassenHamdi, 2020-09-19:
Hi @TMRh20 can you provide any updates or suggestions? Thanks

2bndy5, 2020-09-19:
Maybe try WiringPi instead. [These are the build options](https://github.com/nRF24/RF24/blob/d0627715a2e9ff626e7bfbb4d89215f94c9bc8cd/configure#L177-L192).

Does the spi bus behave correctly for other SPI devices? Not sure if/how you can test this (maybe a loopback test).

krestyan, 2020-11-05:
I wanted to plan a hub around rpi and nrf24l01 and before committing to rpi4 found this issue, are there any updates regarding this?

TMRh20, 2020-11-05:
The RPi4 should be working with the SPIDEV driver but I don’t know if it has been specifically tested.

2bndy5, 2020-11-05:
Library has been working on my new RPi4 with a fresh install of Raspberry Pi OS (formerly known as Raspbian). I've been meaning to update this thread, but I have no suggestion for @BelhassenHamdi

combs, 2020-12-26:
also working on my Pi 4 with the SPIDEV driver, using the Python wrapper with Cython.

2bndy5, 2020-12-26:
I believe @BelhassenHamdi has misconfigured system on his individual RPi4 because I've responded to more than a few other people that got this library working on the RPi4... Still don't know what to suggest without knowing everything else that's setup on his RPi4.

BelhassenHamdi, 2020-12-27:
@2bndy5 Thanks for the update I didn't try since some time. It could be that an raspberry pi os update has fixed the issue. I will check that.

2bndy5, 2020-12-27:
@BelhassenHamdi the RF24 libs have had updates also

2bndy5, 2021-03-31:
@BelhassenHamdi you ever get around to re-testing this? I'm looking to close this as the title is a bit misleading in general.

2bndy5, 2022-01-30:
@BelhassenHamdi Is [this pypi account](https://pypi.org/user/hbe/) yours? I'm asking because the uploaded RF24 & RF24Network packages are broken and violate our [GNU Public License agreement](https://github.com/nRF24/RF24/blob/master/LICENSE).

---
> That would be very nice to see.

I have released a pure python API that employs RF24Network/RF24Mesh protocol as part of the [circuitpython-nrf24l01 pkg](https://pypi.org/project/circuitpython-nrf24l01/).