nRF24/RF24 #630: [Question] Can this library work without Arduino?

TarekkMA, 2020-09-19:
I have an ESP32 that I will be running with the ESP-IDF with no Arduino, so how can I use this library with the build system of espressif build system?

Avamander, 2020-09-19:
The library relies on the Arduino HAL on the ESP platform. So if you want to use it totally without anything Arduino, you'd either have to add support for your current SPI library or you would have to rewrite the Arduino SPI HAL to the extent it's used here.

TarekkMA, 2020-09-19:
Okay I will try to add Arduino as a component to my project and see if I can get it to work

TarekkMA, 2020-09-19:
@Avamander the esp arduino seems to be only working on older version of the ESP-IDF. So I guess I will have to support the esp SPI library.

Can you point me on how I can do that? Should I implement the functions in the https://github.com/nRF24/RF24/blob/master/utility/Template/spi.h to work properly on the ESP-IDF only? or should I do something else?

Avamander, 2020-09-19:
Something like that, yes. You can also check out `ATXMegaD3` folder, it's built a bit more "ground-up".

closed 2020-09-19 by TarekkMA