nRF24/RF24 #636: ignoring a return value in interrupt.c file

2bndy5, 2020-09-21:
I noticed a [warning in the build CI action ](https://github.com/nRF24/RF24/runs/1137835057?check_suite_focus=true#step:5:13)about ignoring a return value in  RF24/utility/RPi/interrupt.c file. Which refers to [this line](https://github.com/nRF24/RF24/blob/d0627715a2e9ff626e7bfbb4d89215f94c9bc8cd/utility/RPi/interrupt.c#L156).

To my understanding, `read()` will return the amount of characters read. The way this file is written, I think that if this function throws an exception, then the build should stop. Which might be why we're ignoring the returned data. Please correct me if I've assumed too much.

It doesn't seem important to fix. I'd just like to eliminate the message from the log.
No idea the correct course of action. Maybe assign the returned value to scoped variable that doesn't get used?

Avamander, 2020-09-21:
> Maybe assign the returned value to scoped variable that doesn't get used?

Then it's not impossible that you'll just get a warning about a variable being written but never used.

Actual error handling would be the correct course of action, if the result ignoring isn't intended, in which case the warning should probably just be suppressed.

2bndy5, 2020-09-22:
git blame originates from @Oitzu . by referencing [this commit](https://github.com/nRF24/RF24/commit/67d0285fc5ceae2b6f50f453ee9926490e629bd4)

Oitzu, 2020-09-22:
The reads at this point are just to discard anything in th fd at this point.
It's unlikely to fail, but I guess it would be better practise to handle error.

2bndy5, 2020-10-20:
ok I managed to fix this one (on my master branch). @Oitzu you're approach to cast the result was close, but since gcc 4.6 your solution has been throwing the `[-Wunused-result]` warning. After taking a cue from [this stackoverflow answer](https://stackoverflow.com/a/13999461), I casted the return value + 1 as void. And no more warning in the CI actions.

@avamander Do you aggree with this solution?

Avamander, 2020-10-20:
I guess it's okay

2bndy5, 2020-10-20:
>I guess it's okay

second opinions?

2bndy5, 2020-12-14:
Closing this for now.

closed 2020-12-14 by 2bndy5