nRF24/RF24 #638: [Question] Odd algorythm in getDynamicPayloadSize()

2bndy5, 2020-09-24:
### Situation
The following snippet is from the [`getDynamicPayloadSize()` definition](https://github.com/nRF24/RF24/blob/820236e6128e295df99a1a7b6b36ea75d2e52030/RF24.cpp#L1058-L1063):
```c++
    // result is the data returned from the SPI command R_RX_PL_WID
    if (result > 32) { // how could this ever happen?
         flush_rx();
         delay(2);
         return 0;
     }
     return result;
```

### Rationality leading to the question
Technically this conditional statement would never resolve to `true`. The nRF24L01 RX FIFO can only store up to 32 bytes in any 1 of the 3 levels. If the payload was received from another nRF24L01, then this _will_ never happen. If a payload of length greater than 32 was received from an ISM (& ESB protocol) compatible radio, then the excess data (33<sup>rd</sup> and 34<sup>th</sup> bytes) in the payload is likely to be perceived as CRC data, and the entire packet is **_most likely_** discarded (excluding the PID info in the packet header). I say "most likely" in the last sentence because I assume that the excess payload data doesn't happen to exactly match the CRC checksum (false-positive), but that is very unlikely. CRC is enabled automatically (by the nRF24L01) when enabling the Auto-ACK feature which is required for dynamic payloads to be enabled (for RX node).

### The question
In any case, why flush the entire RX FIFO? Furthermore, why check if the payload length is greater than 32 bytes at all? This just seems like excessive over-correction. The only time I can foresee getting a value greater than 32 is when the SPI transaction got corrupted either due to strong electromagnetic flux or if bad/faulty pull up/down resistor connections on SPI bus. In offchance that `getDynamicPayloadSize()` does return a number larger than 32 and the MCU uses that value as `len` parameter to `read()` (as exemplified in many example sketched) see #632 

### Code Origin
git blames [this commit](https://github.com/nRF24/RF24/commit/ebcd0d1d0b3061fcb57444e1dbe5829ef25705cd) (back in 2012) from @TMRh20 with description saying:
>Radio getDynamicPayloadSize() - Per the datasheet, added fix to flush
corrupted packets

### Additional Context
I've been looking through the datasheets and I can confirm that the payload length (when dynamic payloads are enabled) is counted by how many bytes get stored to the RX FIFO (concerning 1<sup>st</sup> available 1 level -- max 32 bytes). This number would be different from the payload length data stored in the packet header (even if the payload is larger than 32 bytes and CRC passed). 

### I'm human as well
I have misinterpreted the datasheet in many ways before mastering this device. I do see the datasheet saying (in multiple places) that a single packet is automatically discarded (by the nRF24L01) for various reasons (mostly due to duplicate transmissions when ACK packet is lost). I don't see it saying this has to be done by the MCU. @TMRh20 please elaborate "per datasheet" comment in commit description.

Avamander, 2020-09-24:
You said "Technically this conditional statement would never resolve to `true`" but that is assuming ideal conditions. Some SPI bus error might cause exactly the thing TMRh20 saw when making that commit. I would not remove it unless absolutely sure.

2bndy5, 2020-10-19:
Found [this snippet in the pingpair_dyn.ino](https://github.com/nRF24/RF24/blob/a3f9ace8c530138bd10211a08bc498f05187edb7/examples/pingpair_dyn/pingpair_dyn.ino#L172-L175) example:
```c++
      // If a corrupt dynamic payload is received, it will be flushed
      if (!len) {
        return; 
      }
```

2bndy5, 2020-10-27:
closing this as leaving the code in question doesn't hurt anything. Its just a bit over-assertive

closed 2020-10-27 by 2bndy5

2bndy5, 2020-11-03:
@TMRh20 I mentioned the dynamic payload size returning 0 on this issue.