nRF24/RF24 #644: python3 setup.py build failed with errors

roysG, 2020-10-08:
Having issues with ACK not working correctly? Please see [common issues](https://github.com/nRF24/RF24/blob/master/COMMON_ISSUES.md).

**Describe the bug**
A clear and concise description of what the bug is.

Please include:
1. pi@opengate:~/Desktop/RF24-master/pyRF24 $ python setup.py build
running build
running build_ext
building 'RF24' extension
creating build/temp.linux-armv7l-2.7
arm-linux-gnueabihf-gcc -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -marm -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -mfloat-abi=hard -L/usr/local/lib -I/usr/local/include -fPIC -I/usr/include/python2.7 -c pyRF24.cpp -o build/temp.linux-armv7l-2.7/pyRF24.o
cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
pyRF24.cpp: In function ‘void startFastWrite_wrap2(RF24&, boost::python::api::object, bool, bool)’:
pyRF24.cpp:92:107: error: no matching function for call to ‘RF24::startFastWrite(char*, int, const bool&, bool&)’
 tes_or_bytearray_str(buf), get_bytes_or_bytearray_ln(buf), multicast, startTx);
                                                                              ^

In file included from pyRF24.cpp:1:
/usr/local/include/RF24/RF24.h:822:8: note: candidate: ‘void RF24::startFastWrite(const void*, uint8_t, bool)’
   void startFastWrite( const void* buf, uint8_t len, const bool multicast );
        ^~~~~~~~~~~~~~
/usr/local/include/RF24/RF24.h:822:8: note:   candidate expects 3 arguments, 4 provided
pyRF24.cpp: In function ‘void init_module_RF24()’:
pyRF24.cpp:179:36: error: ‘RPI_BPLUS_GPIO_J8_03’ was not declared in this scope
     .value("RPI_BPLUS_GPIO_J8_03", RPI_BPLUS_GPIO_J8_03)
                                    ^~~~~~~~~~~~~~~~~~~~
pyRF24.cpp:179:36: note: suggested alternative: ‘RPI_V2_GPIO_P5_03’
     .value("RPI_BPLUS_GPIO_J8_03", RPI_BPLUS_GPIO_J8_03)
                                    ^~~~~~~~~~~~~~~~~~~~
                                    RPI_V2_GPIO_P5_03
pyRF24.cpp:180:36: error: ‘RPI_BPLUS_GPIO_J8_05’ was not declared in this scope
     .value("RPI_BPLUS_GPIO_J8_05", RPI_BPLUS_GPIO_J8_05)
                                    ^~~~~~~~~~~~~~~~~~~~
pyRF24.cpp:180:36: note: suggested alternative: ‘RPI_V2_GPIO_P5_05’
     .value("RPI_BPLUS_GPIO_J8_05", RPI_BPLUS_GPIO_J8_05)
                                    ^~~~~~~~~~~~~~~~~~~~
                                    RPI_V2_GPIO_P5_05
pyRF24.cpp:181:36: error: ‘RPI_BPLUS_GPIO_J8_07’ was not declared in this scope
     .value("RPI_BPLUS_GPIO_J8_07", RPI_BPLUS_GPIO_J8_07)
                                    ^~~~~~~~~~~~~~~~~~~~
pyRF24.cpp:181:36: note: suggested alternative: ‘RPI_V2_GPIO_P1_07’
     .value("RPI_BPLUS_GPIO_J8_07", RPI_BPLUS_GPIO_J8_07)
                                    ^~~~~~~~~~~~~~~~~~~~
                                    RPI_V2_GPIO_P1_07
pyRF24.cpp:182:36: error: ‘RPI_BPLUS_GPIO_J8_08’ was not declared in this scope
     .value("RPI_BPLUS_GPIO_J8_08", RPI_BPLUS_GPIO_J8_08)
                                    ^~~~~~~~~~~~~~~~~~~~
pyRF24.cpp:182:36: note: suggested alternative: ‘RPI_V2_GPIO_P1_08’
     .value("RPI_BPLUS_GPIO_J8_08", RPI_BPLUS_GPIO_J8_08)
                                    ^~~~~~~~~~~~~~~~~~~~
                                    RPI_V2_GPIO_P1_08
pyRF24.cpp:183:36: error: ‘RPI_BPLUS_GPIO_J8_10’ was not declared in this scope
     .value("RPI_BPLUS_GPIO_J8_10", RPI_BPLUS_GPIO_J8_10)
                                    ^~~~~~~~~~~~~~~~~~~~
pyRF24.cpp:183:36: note: suggested alternative: ‘RPI_V2_GPIO_P1_10’
     .value("RPI_BPLUS_GPIO_J8_10", RPI_BPLUS_GPIO_J8_10)
                                    ^~~~~~~~~~~~~~~~~~~~
                                    RPI_V2_GPIO_P1_10
pyRF24.cpp:184:36: error: ‘RPI_BPLUS_GPIO_J8_11’ was not declared in this scope
     .value("RPI_BPLUS_GPIO_J8_11", RPI_BPLUS_GPIO_J8_11)
                                    ^~~~~~~~~~~~~~~~~~~~
pyRF24.cpp:184:36: note: suggested alternative: ‘RPI_V2_GPIO_P1_11’
     .value("RPI_BPLUS_GPIO_J8_11", RPI_BPLUS_GPIO_J8_11)
                                    ^~~~~~~~~~~~~~~~~~~~
                                    RPI_V2_GPIO_P1_11
pyRF24.cpp:185:36: error: ‘RPI_BPLUS_GPIO_J8_12’ was not declared in this scope
     .value("RPI_BPLUS_GPIO_J8_12", RPI_BPLUS_GPIO_J8_12)
                                    ^~~~~~~~~~~~~~~~~~~~
pyRF24.cpp:185:36: note: suggested alternative: ‘RPI_V2_GPIO_P1_12’
[... 144 lines cut ...]
                                                                     ^

In file included from pyRF24.cpp:1:
/usr/local/include/RF24/RF24.h:785:9: note: candidates are: ‘bool RF24::txStandBy(uint32_t)’
    bool txStandBy(uint32_t timeout);
         ^~~~~~~~~
/usr/local/include/RF24/RF24.h:765:9: note:                 ‘bool RF24::txStandBy()’
    bool txStandBy();
         ^~~~~~~~~
pyRF24.cpp:306:41: error: ‘startConstCarrier’ is not a member of ‘RF24’
         .def("startConstCarrier",&RF24::startConstCarrier, ( bp::arg("level"), bp::arg("channel")))
                                         ^~~~~~~~~~~~~~~~~
pyRF24.cpp:307:40: error: ‘stopConstCarrier’ is not a member of ‘RF24’
         .def("stopConstCarrier",&RF24::stopConstCarrier)
                                        ^~~~~~~~~~~~~~~~
pyRF24.cpp:315:51: error: ‘failureDetected’ is not a member of ‘RF24’
         .def_readwrite( "failureDetected", &RF24::failureDetected );}
                                                   ^~~~~~~~~~~~~~~
error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

2. build setup should success
3. raspberry pi 4

**Additional context**
Add any other context about the problem here.

2bndy5, 2020-10-08:
If you don't plan on using RF24Network & RF24Mesh, then I highly recommend using `pip3 install circuitpython-nrf24l01`. The API is different though, so you'll have to [read the docs for that library](http://circuitpython-nrf24l01.rtfd.io/)

Actually this isn't the first time someone has reported build errors for the python wrapper on a Raspberry Pi 4. Seen as how there are build errors for practically every line in the python wrapper, I would guess that you didn't [install the C++ library first](https://tmrh20.github.io/RF24/CrossCompile.html). Then you can retry [installing the python wrapper for this library](https://tmrh20.github.io/RF24/Python.html).

roysG, 2020-10-09:
Do i have to install the "Linux cross-compilation"?

closed 2020-10-09 by roysG

2bndy5, 2020-10-09:
>Do i have to install the "Linux cross-compilation"?

Probably not if there are no more build errors.

Have you run any examples? Usually this is the next step to make sure your SPI pins are communicating with the nRF24L01.