nRF24/RF24 #645: RX_ADDR stick on 0xffffffffff

roysG, 2020-10-09:
Having issues with ACK not working correctly? Please see [common issues](https://github.com/nRF24/RF24/blob/master/COMMON_ISSUES.md).

**Describe the bug**
A clear and concise description of what the bug is.

Please include:
1. >> %Run roy.py
/home/pi/Downloads/nrf/lib_nrf24.py:378: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  self.GPIO.setup(self.ce_pin, self.GPIO.OUT)
STATUS	 = 0xff RX_DR=1 TX_DS=1 MAX_RT=1 RX_P_NO=7 TX_FULL=1
RX_ADDR_P0-1	�� =
 0xffffffffff 0xffffffffff
RX_ADDR_P2-5	�� =
0xff
0xff
0xff
0xff

TX_ADDR		 =
 0xffffffffff
RX_PW_P0-6	�� =
0xff
0xff
0xff
0xff
0xff
0xff

EN_AA		 =
0xff

EN_RXADDR	�� =
0xff

RF_CH		 =
0xff

RF_SETUP	�� =
0xff

CONFIG		 =
0xff

DYNPD/FEATURE	�� =
0xff
0xff

Data Rate	 = 1MBPS
Model		 = nRF24L01
CRC Length	 = 16 bits
PA Power	 = PA_HIGH
>>> 
2. The "RX_ADDR_P0-1" should get his unique id, but i always get: " 0xffffffffff 0xffffffffff", i also tried with another rf24 device. 
3. rf24

What may cause to this problem?

Avamander, 2020-10-09:
Module is incorrectly connected or not powered.

roysG, 2020-10-10:
Do you have some example how to connect it raspberry?

I tried to followed posts show how to connect it, but it seems that it was
not good.

Maybe the driver not installed properly on my system?

Can you give me the steps for installing this drivers?

2bndy5, 2020-10-10:
I already sent you links to the install steps in #644. I highly recommend a 100 microfarad capacitor between the nRF24L01 VCC and GND pins. The following table is a simplified (copy from the [bottom of the docs main page](https://tmrh20.github.io/RF24/index.html)) uses the pins used in all the library examples

| nRF24L01 pin | Raspberry Pi pin |
| -- | -- |
| GND | gnd |
| VCC | 3v3 |
| CE | gpio22 (ce0) | 
| CSN | gpio8 |
| SCK | gpio11 (sckl) |
| MOSI | gpio10 (mosi) |
| MISO | gpio9 (miso) |
| IRQ | not typically connected |

[image]

[image]

roysG, 2020-10-10:
Thanks, now i get the correct values!

sudo ./gettingstarted
RF24/examples/GettingStarted/
================ SPI Configuration ================
CSN Pin  	 = 0
CE Pin  	 = Custom GPIO22
SPI Speedz	 = 10 Mhz
================ NRF Configuration ================
STATUS		 = 0x80 RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=0 TX_FULL=0
RX_ADDR_P0-1	 = 0x8183818180 0x8181838180
RX_ADDR_P2-5	 = 0x80 0x80 0x80 0x80
TX_ADDR		 = 0x8383838080
RX_PW_P0-6	 = 0x80 0x80 0x80 0x80 0x80 0x80
EN_AA		 = 0x80
EN_RXADDR	 = 0x80
RF_CH		 = 0x80
RF_SETUP	 = 0x80
CONFIG		 = 0x80
DYNPD/FEATURE	 = 0x80 0x80
Data Rate	 = 1MBPS
Model		 = nRF24L01
CRC Length	 = 8 bits
PA Power	 = PA_MIN

 ************ Role Setup ***********

Now i am trying to run the gettingStarted.
I defined thr raspberry as reciever and arduino as transmitter.

I see the line: 
const uint8_t pipes[][6] = {"1Node", "2Node"};

1. Do i need to change it to my values?
2. Does pipes can be defined by any id i decide?

roysG, 2020-10-10:
The raspberry now works propely.
It seems that there is problem with the arduino tansmit.
This is the aruduino otuput:

Sent 35132952, Got response 0, Round-trip delay 96096 microseconds
Now sending
failed

2bndy5, 2020-10-10:
you need to change [this line](https://github.com/nRF24/RF24/blob/6e5195acddb0aaf5b4b2aa4d198bd707ce84bd57/examples/GettingStarted/GettingStarted.ino#L22) in arduino (or raspberry Pi, but not both).

This is getting to the point where you just need to [READ THE DOCUMENTATION](https://tmrh20.github.io/RF24/annotated.html) and any comments in the example code. If English isn't you're native language, then use "google translate" that is natively available in the chrome browser. So far we've walked you through what [any youtube video can tell you about using this library](https://www.youtube.com/results?search_query=RF24).

closed 2020-12-14 by TMRh20