nRF24/RF24 #648: [Question]MISO MOSI and HMISO, HMOSI

pavluchenkor, 2020-10-11:
Good afternoon, following the documentation, the node has MISO MOSI etc. and HMISO, HMOSI, etc. How can I use MISO MOSI? Everything works fine for me, but I would like to free the ports for other purposes.
I checked, turned on NRF 24 in MISO MOSI, left CN on port 4 (GPIO 2), indicated RF24 radio (2, SCK) ;, it seemed to me so correct, but radio.isChipConnected (); will give me a false.
[image]

2bndy5, 2020-10-11:
This is more of a ESP8266 question rather than a nRF24L01 question. [Try this advice](https://arduino-esp8266.readthedocs.io/en/2.7.4_a/libraries.html#spi) from the Arduino core for the ESP8266. In my experience, the nodeMCU boards weren't labeled (nor documented) for use with Arduino IDE. It was originally designed to run a LUA interpreter using the [nodeMCU firmware](https://github.com/nodemcu/nodemcu-firmware).

pavluchenkor, 2020-10-11:
Of course, this is a question for nodeMCU, the NRF24 module works fine, but I'm sure it can be connected this way. Half a year ago, when I did not know anything about NRF24, I found one lesson where it was included in this way, but unfortunately I cannot find it and I have no code left.
Now for the tests.
Either I'm trying the wrong way or your advice doesn't work.
Connection
* S1 MOSI = SD1
* S0 MISO = SD0
* SK SCLK = CLK
CE 2
CSN 0
the code

RF24 radio (2.0); //
void setup (void)
{
   SPI.pins (6, 7, 8, 0);
   SPI.begin ();
   radio.begin ();
   bool result = radio.isChipConnected ();
...
}
and it doesn't work

pavluchenkor, 2020-10-11:
I found such an example on the Internet, but I can't find anything like that anymore, and this one doesn't work for me.
In the description of the class, I see a little different

class RF24 {
private:
     #ifdef SOFTSPI
     SoftSPI <SOFT_SPI_MISO_PIN, SOFT_SPI_MOSI_PIN, SOFT_SPI_SCK_PIN, SPI_MODE> spi;

but I don't understand how to specify CE and CSN then

#include "nRF24L01.h"
#include "RF24.h"
char msg[6] = "hello";
RF24 radio(12, 14, 26, 25, 27);
const uint64_t pipe = 0xE8E8F0F0E1LL;
 
void setup(void) {
  Serial.begin(115200);
  radio.begin();
  radio.setChannel(2);
  radio.setPayloadSize(7);
  radio.setDataRate(RF24_250KBPS);
  radio.openWritingPipe(pipe);
}
void loop(void) {
  Serial.println("send ...");
  radio.write(msg, 6);
  delay(3000);
}

2bndy5, 2020-10-11:
I can't help you. Providing a link to the advice was "generous" of me. Try asking this question on [the repo that enables esp8266 for Arduino IDE](https://github.com/esp8266/Arduino/issues). As you said
>Of course, this is a question for nodeMCU, the NRF24 module works fine

this is simply the wrong repo to ask this question

2bndy5, 2020-10-11:
BTW, of course this won't work
```c++
RF24 radio(12, 14, 26, 25, 27);
```
because... well just [read the docs](https://tmrh20.github.io/RF24/classRF24.html#abe8f878e238644252ccf5ec27fba611d).
Be aware that any "unofficial" examples may not be compatible with the current release of this library (and this fork of maniacBug's original code -- if unofficial example is older than 2014).

2bndy5, 2021-03-20:
@pavluchenkor I'm working on this feature to use a secondary SPI bus by overloading `begin()`... I think its ready for testing (need testers). The [overload-begin branch](https://github.com/nRF24/RF24/tree/overload-begin) has the changes. I've also added [example source code specific for the ESP8266 (NodeMCU)](https://github.com/nRF24/RF24/blob/overload-begin/docs/arduino.md#esp8266-example).

closed 2021-04-11 by 2bndy5