nRF24/RF24 #659: Add example to work with NRFLite library

tong67, 2020-10-19:
For a project I needed some free pins for the receiver side connected to ATTiny85. The library NRFLite (https://github.com/dparson55/NRFLite) provides a configuration where only 2 pins are required. However the API is different and different defaults are used. Some investigation and experiments were required to find the correct settings. The transmitting side finally was RF24 python library on RPI. The attached sketch and python script works with the Basic_RX examples provided by the NRFLite library.
Can these files be added so other users can use the same combination?

[Basic_TX_RF24.ino.txt](https://github.com/nRF24/RF24/files/5403638/Basic_TX_RF24.ino.txt)
[Basic_TX.py.txt](https://github.com/nRF24/RF24/files/5403639/Basic_TX.py.txt)

2bndy5, 2020-10-19:
Generally you would open an Pull Request (PR for short) to add files to a library. What's wrong with this library's `rf24ping85.ino` example (I see you authored it)?

These lovely Basic_TX examples are proof that #658 is long overdue.

2bndy5, 2020-10-19:
Basic_TX.py example has namespacing problems. I mentioned why `from RF24 import *` is problematic for this reason in #634

2bndy5, 2020-10-21:
@tong67 The timingSearch3pin.ino example isn't compiling anymore. Could you take a look at it (since this was your contribution)?

tong67, 2020-10-22:
>I'd actually be more interested in implementing nerdRalph's 2-wire implementation (though I think `getStatus()` would need some TLC as full duplex communication may be compromised with MOSI & MISO pins sharing the same connection).

I agree, but I say that too much of a challenge. Having a way to combine the two implementation could make is easier to realize. This looks like a good first step to me

> 
> > different defaults are used
> 
> could you specify what defaults are different? Please tell me dynamic payloads are enabled by default (something I wish this library would do).

All commands I had to add after radio.begin():
 const uint64_t pipes[2] = { 0x0004030201LL, 0x0104030201LL };
 radio.setChannel(100);          // channel 100
 radio.setPALevel(RF24_PA_MAX);  // MAX power
 radio.setDataRate(RF24_2MBPS);  // 2MBPS
 radio.setRetries(1, 15);        // 250+1*250us delay, 15 retries
 radio.enableDynamicAck();       // Enable auto ack
 radio.enableAckPayload();       // enable payload with ack and dynamic payload length
 radio.setCRCLength(RF24_CRC_8); // CRC8

The first bytes of pipe addresses are configurable in NRFLite, the other parts of the pipe addresses not.
> 
> These lovely Basic_TX examples are proof that #658 is long overdue.

tong67, 2020-10-22:
> Basic_TX.py example has namespacing problems. I mentioned why `from RF24 import *` is problematic for this reason in #634
> 
> > `millis = lambda: int(round(time.time() * 1000)) & 0xffffffff`
> 
> you could use `time.monotonic()` for arbitrary milliseconds timer.

I will update the python script (and also the existing one I used as starting point that has the same import * issue).
Will make it a pull request ....
> 
> > `pipes = [0x0004030201, 0x0104030201]`
> 
> See #496 about using `0x00` as addresses' first byte (its a problem with this library)

The NRFLite Basic_RX example expect 0x00 as first byte. Will add comments that here for demo but should be prevented in normal usage with reference to #496

>@tong67 The timingSearch3pin.ino example isn't compiling anymore. Could you take a look at it (since this was your contribution)?

Will have a look at it and combine the changes in pull request

2bndy5, 2020-10-22:
>radio.setPALevel(RF24_PA_MAX); // MAX power

This library's default is the same
>radio.enableDynamicAck(); // Enable auto ack

My next PR (& hopefully the net release) will use this as default.
>radio.enableAckPayload(); // enable payload with ack and dynamic payload length

I've been considering making this default as well, but I need to do some more testing.
>radio.setRetries(1, 15); // 250+1*250us delay, 15 retries

This is not advised according to the datasheet. the delay should be set to longer than 500 us
>The NRFLite Basic_RX example expect 0x00 as first byte.

This is just not a good idea considering the packet's OTA preamble. [Seriously give this article a read](http://maniacalbits.blogspot.com/2013/04/rf24-addressing-nrf24l01-radios-require.html)
>Will have a look at it and combine the changes in pull request

I asked you to look at it because my plate is a bit full for this library, and that code looks like you copied the entire library source code into the example. I'm sorry, but no; I will not support an example that I didn't write,. Also, I'm not familiar with NerRalph's custom SPI functions that make 2 or 3 wires possible. I only saw the electrical schematic to see how it was done, which is what interested me to begin with.

I have to warn you that the examples in this library *may be* getting replaced. The examples in this library (in my opinion) should really be about using this library. Any other examples that boast compatibility with another library should be hosted by another repo or the other library.

If you want, we can add a link in the docs to your repo where you host these peculiar examples.

tong67, 2020-11-01:
The build error is caused by combination of selected for and version of this library. After #646 (removal of SPI) the code only compiles correctly with ATTinyCore from SpenceKonde. No modifications are required.

Because the provided examples will not be included I will not attempt to change them and make a pull request. I will wait till the NRFLite repo has included the code examples to work with this library to tell which link can be added to this repo to refer to the examples how the repo's can with each other.

2bndy5, 2020-11-02:
>No modifications are required.

Are you saying I can remove the redefined SPI functions from "timingSearch3pin.ino" example?

tong67, 2020-11-08:
No, I reacted on the remark that the sketch was not compiling anymore. As stated in the comments in the sketch it is based on another tiny core. It also uses another pin as serial output which is determined by the core being used. Sketch should be completely re-examined with different core being selected as default for RF24

2bndy5, 2020-11-08:
Library now depends on the popular SpenceKonde ATTinyCore.

[Tagging my comment from #646](https://github.com/nRF24/RF24/pull/646#issuecomment-717082135)

[Here is a link to the file with the changes](https://github.com/2bndy5/RF24/blob/master/examples/rf24_ATTiny/timingSearch3pin/timingSearch3pin.ino) I made when I last looked into it.

I suspect that a macro should be defined in RF24_config.h so that the the right redefined SPI functions are used...

2bndy5, 2020-11-22:
@tong67 I've been looking into the timingSearch3pin.ino example, and I got it to compile for the SpenceKonde ATTinyCore, but I haven't tested it on hardware yet (going to use my digispark).

I used `#include "nRF24L01.h"` instead of copying all those values into the sketch. Also, I had to get rid of the custom defined SPI class and use the SpenceKonde's `__ATTinyX4__` & `__ATTinyX4__` macros to detect the ATTinyCore to set the `CE_PIN` & `CSN_PIN` accordingly. Lastly, I changed 2 lines from
```c++
PORTB |= (1 << PINB2);
PORTB &= ~(1 << PINB2);
```
to
```c++
digitalWrite(SCK, mode);
```
in the `csn()` function because the SpenceKonde ATTinyCore's SPI library already has defined the SCK pin (which I think is already setup as an output pin due to the "fake" SPI implementation on ATTiny chips).

Do you have any objections to these changes (was there a reason you originally wrote the sketch to be "self-contained")? I plan on reducing the program size/structure a bit as well, but is there a reason that you don't test multiple values for `csnHighSettle` & `csnLowSettle` based on consecutive failure? IMO, it seems counterproductive to only test the values that are hard coded in the sketch.

I also see that the example doesn't set a SPI frequency which should affect the CSN state. So, what are your thoughts on the SPI frequency? Sorry for all the questions, but I think most of the comments in the sketch are actually copied from NerdRalph's code (I'm adamant for explicit comments).

p.s. the ATTinyCore that your examples link to hasn't been updated in over 5 years, so I think this maintenance is long overdue.

tong67, 2020-11-29:
The sketch was made self-contained because multiple libraries exist(ed) which could be used to enable 3 pin mode for ATtiny85. Because the sketch is now part of this library it make sense to let it depend on the library provided API. So no objects to that. I would not replace the bit handling of eg. PORTB from direct bit manipulation into use of digitalWrite(). The timing of both constructions is very different and we should stick as much as possible to the code used within the library which is direct bit manipulaton.
For the SPI frequency I have no idea. The current code does not set it to a value. I have no idea what the SPI library with the SpenceKonde ATTinyCore will use as SPI frequency. Again I would go for sticking as much as possible to that what is used normally.

When you start experimenting, be aware that the Serial output pins of the two cores are different. In the tiny-core from CodingBadly it is pin 2 (PB3). The SpenceKonde  ATtinyCore uses pin 5 (PB0) as TX and pin 6 (PB1) as RX which could also interfere your sketch when pin 5 or 6 are used for other purposes.

2bndy5, 2020-11-30:
>The timing of both constructions is very different and we should stick as much as possible to the code used within the library which is direct bit manipulaton.

I assume you mean "instructions" instead of "constructions". Which "library" are you referring to -- SpenceKonde or RF24?

I changed the `PORTB`  lines to `digitalWrite()` because I wasn't sure if the SpenceKonde library actually used the same defined macros for portB. However, reverting back to directly manipulating PB2 bit in the PORTB register seems to compile fine with SpenceKonde library.
- [link to SpenceKonde's `digitalWrite()` definition for the ATTiny85](https://github.com/SpenceKonde/ATTinyCore/blob/99c47b3116d847a1d69731c1ab121c8bc219b146/avr/cores/tiny/wiring_digital.c#L136)
    - notice this function does take an extra step to disable PWM on specified pin (if applicable -- I think PWM only applies to pin 0 and pin 1).
- still can't find where PORTB is defined in SpenceKonde's ATTinyCore, but I think it exits in the "avr" library that ships with Arduino IDE because it is used as a reference variable (`&PORTB`) in several files of the SpenceKonde ATTinyCore.

As for SPI frequency, I'm pretty sure the clock speed of the ATTiny chip directly affects the SPI frequency due to USI SPI technique in the ATTiny. The `RF24::begin()` function will override the SPI frequency passed to the RF24 c'tor (`_spi_speed` parameter) based on the MCU frequency.

I will also look into the Serial I/O pins. I plan on using a Digispark, but I can't remember what pins are used by the USB connection. I wish I wasn't doing this maintenance on my own because I clearly don't know a lot about the ATTiny85. So far, I feel like I'm waving my hand in a dark room trying to find the light switch.

closed 2021-06-12 by 2bndy5