nRF24/RF24 #666: Radio.write return always true

bprovvis, 2020-10-20:
hello
my project

raspberry MASTER ----->Arduino uno SLAVE

It works fine but in raspberry Radio.write return always true even when Arduino SLAVE is turned off

my code
  if(radio.begin() ) {
  printf("Begin() OK \n") ;
  }
  radio.setDataRate(RF24_250KBPS);
  radio.setPALevel(RF24_PA_MIN );
  radio.setAutoAck(false);
  radio.setRetries(15,15);
    radio.openWritingPipe(pipes_for);
    radio.openReadingPipe(1,pipes_for);

 // First, stop listening so we can talk.
    radio.stopListening();
    bool ok = radio.write( &messaggio_to_slave, sizeof(messaggio_to_slave) );

    if (ok){
      cont_ok++;

      printf("Write ok...to Slave ");printf("%d",i);printf(" Conteggio OK ");printf("%d \n",cont_ok);
      }
   else
   {
      cont_fail++;

      printf("failed Write... to slave ");printf("%d",i);printf("Conteggio ");printf("%d \n",cont_fail);
      }

thanks for any help

Avamander, 2020-10-20:
I saw this once when my module's power supply was garbage. How's yours?

2bndy5, 2020-10-20:
>radio.setAutoAck(false);

`write()` will always return true because you're telling the master to not wait for an acknowledgement from the slave node. Therefore, the only thing left to report is if the master attempted to send anything (which will always be true). [Read the docs about `setAutoAck()`](https://tmrh20.github.io/RF24/classRF24.html#aec71746d59da978bcbb975167886a2cc).

bprovvis, 2020-11-06:
hello
sorry for the delay I had some problems
i try  radio.setAutoAck(true); and is all OK
thanks

closed 2020-11-06 by bprovvis