nRF24/RF24 #671: startListening() calls closePipe(0)

2bndy5, 2020-10-22:
I don't throw the "bug" label around lightly, but I've been racking my brain and triple checking the datasheet. This actually relates to #496 as it would only happen when the uint8_t array `pipe0_reading_address`'s first byte is 0. In #496, @justinBR pointed out that `startListening()` executes the following code:
```c++
    if (pipe0_reading_address[0] > 0) {
        write_register(RX_ADDR_P0, pipe0_reading_address, addr_width);
    } else {
        closeReadingPipe(0);  // THIS LINE IS A PROBLEM
    }
```
which stems from [maniacBug's initial commit titled "added"](https://github.com/nRF24/RF24/commit/de083c964d9aeeb9fda7485c39fed27443cd617c)

## What was the original intention???
I can't understand the intention to close pipe 0 (its power-on-reset value makes it open). For a function whose sole purpose is to enable RX mode, it should not be closing any RX pipes. I realize that pipe 0 is also used for TX-ing ACK packets, but that functionality is automated via the nRF24L01's firmware (see [list item 2 on this page](https://tmrh20.github.io/RF24/tmrh20/nRF24L01P_datasheet_v1.pdf#page=26)). Also, the address for ACK packets is fetched from the RX_ADDR_P# register related to the pipe that received the instigating payload (not the [TX_ADDR register](https://tmrh20.github.io/RF24/tmrh20/nRF24L01P_datasheet_v1.pdf#page=57) -- see also [the first paragraph on this page](https://tmrh20.github.io/RF24/tmrh20/nRF24L01P_datasheet_v1.pdf#page=39)). This is why the nRF24L01 *can* listen to 6 devices simultaneously instead of 5. I know the docs about [`openReadingPipe()`](http://tmrh20.github.io/RF24/classRF24.html#a9edc910ccc1ffcff56814b08faca5535) say:
>Pipe 0 is also used by the writing pipe so should typically be avoided as a reading pipe.

Nordic Semiconductors didn't design pipe 0 to be a TX only pipe for good reasons. Additionally, the datasheet never warns against using pipe 0 for RX mode.

1. Code to reproduce
```c++
uint64_t address[2] = {0x00FBFBFBFBLL, 0xFBFBFBFBFBLL};
// 0x00FBFBFBFB  is a malformed address due to the 0x00 byte
radio.openReadingPipe(0, address[0]);
radio.openReadingPipe(1, address[1]);
radio.startListening();
```

2. Expected behavior
After the call to `startListening()` pipe 0 should be open for receiving, but it gets closed because the pipe's address (& its shadow copy `pipe0_reading_address`) started with a 0. The status for pipe 1 is open as expected after calling `startListening()`.

3. What device(es) are you using? Please specify make and model.
Doesn't matter. This is a conceptual bug limiting adequate/proper usage of the transceiver.

## Suggested Solution
1. Simply remove the else block that closes pipe 0.
2. We should also remove the docs' discouragement about using pipe 0 as a reading pipe as I've already stated that pipe 0's usage for TX operations during RX mode is automated in the nRF24L01 firmware (when auto-ack is enabled). Instead the docs about `openWritingPipe()` should mention how pipe 0 needs to be re-appropriated (using another call to `openWritingPipe()`) for RX-ing ACK packets if `openReadingPipe()` and/or `startListening()` where called after setting the TX address (even if the TX address hasn't changed).

## Additional context
I want to be clear that this will not solve #496, but it *might* limit that issue's occurrence. [The docs](http://tmrh20.github.io/RF24/classRF24.html#a9edc910ccc1ffcff56814b08faca5535), [TMRh20's comment in #496](https://github.com/nRF24/RF24/issues/496#issuecomment-671169580),  and the erroneous code at the top clearly state/show why using the integer 0 for the first byte in the address written to pipe 0 is strongly discouraged.

In trying to make sense of this call `closeReadingPipe(0)`, I checked with RadioHead's RH_NRF24.cpp file, and [it also doesn't close any RX pipes when enabling RX mode](https://github.com/hallard/RadioHead/blob/efd2e258bb9f230d96ccbe81679222fc96a21101/RH_NRF24.cpp#L161).

Avamander, 2020-10-22:
> I've already stated that pipe 0's usage for TX operations during RX mode is automated in the nRF24L01 firmware (when auto-ack is enabled)

This does sound like a nice footgun to be honest.

2bndy5, 2020-10-22:
>is the behaviour consistent with the nRF24L01 non-plus and maybe even clones?

I'm not sure. Preliminary guess (due to the copy and paste I've seen in those datasheets): Yes. But I have to do more research on that (I don't own any clones to test). It would be nice to have a list of supported clones somewhere in the docs (just to have a clear starting point for that question).

>footgun

you mean on maniacBug's behalf? I spent the last 3 hours trying to make sense of this, so my brain is constantly questioning what I know or may not know.

I'm also wondering if this would effect the intention of RF24Network & Rf24Mesh, but I haven't gotten through reading all the code for those.

Avamander, 2020-10-22:
> you mean on maniacBug's behalf? I spent the last 3 hours trying to make sense of this, so my brain is constantly questioning what I know or may not know.

When someone disables autoack and has to debug that it behaves differently.

2bndy5, 2020-10-22:
per non-plus datasheet, this behavior is consistent

2bndy5, 2020-10-22:
I recently added a commit titled "8 paragraphs that say keep auto-ack enabled" -- its a clarification in the setAutoAck() docs.

Avamander, 2020-10-22:
It probably should be written in the documentation of the function that disables autoack that it has side-effects. Was that where you added the paragraphs?

2bndy5, 2020-10-22:
[here you go](https://github.com/2bndy5/RF24/commit/727c3e194af224cb72a3dfa38f36424fa07ccf2a)

2bndy5, 2020-10-22:
Looks like the Si24R1 datasheet says the same thing, but I'm trying to translate that from chinese (I don't read chinese -- my google translate does). Do you know of a copy for that datasheet that's in english?

2bndy5, 2020-10-22:
@Avamander don't forget that this issue only becomes a problem if the first byte is 0 for the address on pipe 0 -- so its mostly a library issue stemming from uint8_t friendly NULL vs 0.

TMRh20, 2020-10-27:
Long thread and again I’m having problems following the discussion, so hopefully this isn’t redundant info, but the TX address is assigned for reading on pipe0 whenever sending happens. 

It needs to be closed so that the TX device does not end up listening on the same pipe as the recipient after sending is completed.

2bndy5, 2020-10-27:
>the TX address is assigned for reading on pipe0 whenever sending happens

Yes. Per the datasheet this is required, but only for when auto-ack is enabled. This isn't needed if the TX node doesn't care about ACK packets.

>It needs to be closed so that the TX device does not end up listening on the same pipe as the recipient after sending is completed.

Ok, that explains the intention. However, this problem should be avoided with docs, not code. Technically the RF24Network nodes can listen to as many as 6 child nodes as long as the pipe 0 address is re-written to RX_ADDR_P0 register after exiting TX mode. I used the RF24Network as an example to demonstrate the benefit of using all pipes for RX mode.

The following is from my Multiceiver example (to address #658) in which the `role` variable is a number from 0-5 for TX nodes, or it is a negative number for the RX node (this code tested out fine):
```c++
void setRole() {
  if (role < 0) {
    // For the RX node

    // Set the addresses for all pipes to TX nodes
    for (uint8_t i = 0; i < 6; ++i)
      radio.openReadingPipe(i, address[i]);

    radio.startListening(); // put radio in RX mode

  } else {   // BOUNDS CHECKING IS DONE FROM `if (Serial.available())` BLOCK
    // For the TX node

    // set the payload's nodeID & reset the payload's identifying number
    payload.nodeID = role;
    payload.payloadID = 0;

    // Set the address on pipe 0 to the RX node.
    radio.stopListening(); // put radio in TX mode
    radio.openWritingPipe(address[role]);

    // According to the datasheet, the auto-retry features's delay value should
    // be "skewed" to allow the RX node to receive 1 transmission at a time.
    // So, use varying delay between retry attempts and 15 (at most) retry attempts
    radio.setRetries(((role * 3) % 12) + 3, 15); // maximum value is 15 for both args
  }
} // setRole
```

TMRh20, 2020-11-16:
> I can't understand the intention to close pipe 0

The original intention was that if the user was not using pipe0 as a reading pipe, the radio would not end up listening on pipe0 using the TX address. This will cause issues with auto-ack if the library does not close the TX RX pipe that is using the address of the previous recipient. This is why the issue should be addressed in code, not docs IMHO. 

Its more of a user friendly capability than a necessary requirement I understand, but if not handled in code, the user would need a separate function to close it, or be forced to use pipe0 as a reading pipe, no?

2bndy5, 2020-11-16:
Yes, it would be more user friendly to close the pipe 0 when entering RX mode, especially if the user isn't monitoring what pipe received the payload. However, concerning the code snippet in the OP, this library only closes pipe 0 when the RX address written to pipe 0 has a 0 in the LSByte (malformed address). That means the library isn't actually closing pipe 0 after transitioning from TX mode when a well-formed TX address is used on pipe 0 (which counteracts the original intention you specify).
```c++
uint8_t address[2][6] = {"1Node", "2Node"};
radio.openWritingPipe(address[0]);
radio.openReadingPipe(1, address[1]);
radio.stopListening();  // pipe 0 and 1 are open for RX, but only pipe 0 is used in TX mode
radio.startListening(); // pipe 0 and 1 are open for RX because the address on pipe 0 didn't start with "\0"
```
I fully appreciate this dialogue because its a great learning opportunity for me!

2bndy5, 2020-11-16:
I just noticed this line at the bottom of `stopListening()`:
```c++
write_register(EN_RXADDR, read_register(EN_RXADDR) | _BV(pgm_read_byte(&child_pipe_enable[0]))); // Enable RX on pipe0
```
This is getting hard to track, I have to do some thought experiments.

TMRh20, 2020-11-16:
> However, concerning the code snippet in the OP, this library only closes pipe 0 when the RX address written to pipe 0 has a 0 in the LSByte (malformed address). That means the library isn't actually [...]

That is exactly the intention, except the lib should close pipe0 when the RX addy written to pipe0 is malformed **OR is not present.**

2bndy5, 2020-11-16:
ok I'll leave the `closeReadingPipe(0)` call in, but I've already expanded the docs to notify the user of the underlying behavior. Is it ok if I leave the modified docs?

updated docs on [`startListening()`](https://2bndy5.github.io/RF24/classRF24.html#a30a2733a3889bdc331fe2d2f4f0f7b39), [`stopListening()`](https://2bndy5.github.io/RF24/classRF24.html#a6f144d73fc447c8ac2d1a4166210fd88), [`openWritingPipe()`](https://2bndy5.github.io/RF24/classRF24.html#af2e409e62d49a23e372a70b904ae30e1), and [`openReadingPipe()`](https://2bndy5.github.io/RF24/classRF24.html#a9edc910ccc1ffcff56814b08faca5535)

TMRh20, 2020-11-16:
Yup the docs look good

2bndy5, 2020-11-16:
if there was a private member that saved the state of the EN_AA register, we could limit this behavior to only when auto-ack is enabled for pipe 0 (similar to the `config_reg` private member). But that is an issue I'm putting off till after the next release (more likely just for my fork).

combs, 2020-11-25:
Forgive me if I am missing something obvious. I'm wondering if this change to startListening() would be an improvement:

```
     ...
     config_reg |= _BV(PRIM_RX);
     write_register(NRF_CONFIG,config_reg);
     write_register(NRF_STATUS, _BV(RX_DR) | _BV(TX_DS) | _BV(MAX_RT));
     ce(HIGH);
     write_register(RX_ADDR_P0, pipe0_reading_address, addr_width);

     if(ack_payloads_enabled){
         flush_tx();
     }
```

In other words, always restoring pipe 0 reading address from `pipe0_reading_address`, instead of trying to sense whether it is set by LSB==0. (`pipe0_reading_address` is set to `0x0000000000` by the constructor.)

- Someone using pipe 0 rx is fine, as Pipe 0 address would be restored from the value stashed in the `pipe0_reading_address`.

- Someone who had never called `openReadingPipe` for pipe 0 would have pipe 0's RX address set to 0x0, so if they don't close the pipe explicitly, they could receive messages addressed to `0x0000000000`. 

- Pipe 0 will not be accidentally closed by the library if the address LSB contains 0.

- A tiny bit less logic to run on MCUs. But, maybe one more SPI transaction if pipe 0 rx is unused.

I think this would be a little more "noob friendly" as the possibility of receiving random messages at `0x0000000000` after TX w/AA seems less likely and less problematic to me than a noob trying to use an address with 0x00 LSB and having their pipes stealth-closed. 

I've been wrestling with some strange problems with pipe 0 with (barf) Si24R1 modules, and this idea came out of me digging into that. This library is seriously cool and useful, and I hope I'm helping. Please let me know if I'm missing something obvious with this suggestion or if you have any other feedback.

2bndy5, 2020-11-26:
>I think this would be a little more "noob friendly" as the possibility of receiving random messages at 0x0000000000 after TX w/AA seems less likely and less problematic to me than a noob trying to use an address with 0x00 LSB and having their pipes stealth-closed.

This actually relates to #496 . Personally, I'd prefer 
```c++
if (memcmp(pipe0_reading_address, 0x0000000000, addr_width) == 0)
    // Restore pipe 0 address
else
    closeReadingPipe(0);
```
to better prevent undesired stealthy closure of pipe 0. I understand that this code wouldn't work as is because `0x0000000000` is not an rvalue, but I digress.

The only reason we use `pipe0_reading_address` is to restore the state of pipe 0 to it's original state when the user hasn't explicitly designated it for RX behavior. Granted, `startListening()` & `stopListening()` currently presume that auto-ack is always enabled.

>so if they don't close the pipe explicitly, they could receive messages addressed to 0x0000000000

Setting aside the malformation of a zeroed-out address, I don't see how this is advantageous. Did you have a use case in mind for this?

>A tiny bit less logic to run on MCUs. But, maybe one more SPI transaction if pipe 0 rx is unused.

I'm not worried about the quantity of logic (if that wasn't obvious from my `memcmp()` suggestion). I am a bit sensitive about unnecessary SPI transactions because they can slow things down.

combs, 2020-11-27:
Ahh, I understand the existing logic more if SPI transactions are costly. I can see how this would reduce throughput to reduce the address each time. Thank you

2bndy5, 2020-11-27:
No problem. I do appreciate these discussions as it provides a great opportunity for learning (for all parties).

closed 2020-12-14 by 2bndy5