nRF24/RF24 #672: [Request] Remove useless macros

2bndy5, 2020-10-24:
I have an idea that could save some memory. It involves using the actual pipe numbers instead of some macros in `nRF24L01.h`. The macros in question are:
```c++
/* By using the actual the actual pipe number, we ca get rid of these */
// for use with the DYNPD register
#define DPL_P5      5
#define DPL_P4      4
#define DPL_P3      3
#define DPL_P2      2
#define DPL_P1      1
#define DPL_P0      0

// for use with the EN_RXADDR register
#define ERX_P5      5
#define ERX_P4      4
#define ERX_P3      3
#define ERX_P2      2
#define ERX_P1      1
#define ERX_P0      0

// for use with the EN_AA register
#define ENAA_P5     5
#define ENAA_P4     4
#define ENAA_P3     3
#define ENAA_P2     2
#define ENAA_P1     1
#define ENAA_P0     0

/* most of these can be discarded; I've marked the ones to keep */
// macros for register offsets
#define RX_PW_P0    0x11  // KEEP THIS ONE
#define RX_PW_P1    0x12
#define RX_PW_P2    0x13
#define RX_PW_P3    0x14
#define RX_PW_P4    0x15
#define RX_PW_P5    0x16
#define RX_ADDR_P0  0x0A  // KEEP THIS ONE
#define RX_ADDR_P1  0x0B
#define RX_ADDR_P2  0x0C
#define RX_ADDR_P3  0x0D
#define RX_ADDR_P4  0x0E
#define RX_ADDR_P5  0x0F

/* These have the same value because they're the same register offset, */
#define CD          0x09  // the library only uses this one
#define RPD         0x09

/* this one is never used */
#define LNA_HCURR   0
```
The comments I've added to the above snippet talks about what I'd like to do to get rid of them, Except the macros I marked with the comment `// KEEP THIS ONE`. When getting rid of the subsequent macros we can use 
`RX_ADDR_P0 + pipe_number` or `RX_PW_P0 + pipe_number` where applicable.

## Why is this important?
To enable control per pipe of dynamic  and static payload lengths, I need to add quite a bit (4 functions and 1-7 bytes). I've been avoiding implementing the control per pipe features because I'm afraid of increasing the memory allocated. But with these useless macros gone, I could trade their allocation for implementing the control per pipe features. I realize that the trade off wouldn't be exact, but I would feel less guilty about it

Avamander, 2020-10-24:
Preprocessor directives that aren't used do not use any memory.

2bndy5, 2020-10-24:
ok. my bad. I was really hoping to find the space for the control per pipe features

closed 2020-10-24 by 2bndy5