nRF24/RF24 #676: CMake Support

AndruePeters, 2020-11-06:
**Describe the problem**
Configuring RF24 as part of another project is not 'simple'. My current method calls a script the configures and 
installs RF24 on the host system. However, I'd like to be able to use this repo as I would other libraries such as: 
with a package manager, or using git submodules. One of the present issues regards the build system which
requires running the configure script and then using Make.

**Describe the solution you'd like**
The solution would be to combine the configuration and build system and configuration. CMake provides an interface
to handle this. CMake has also transitioned to becoming industry standard for build systems. The interface for building using
CMake can be similar to the current interface for the configure script; e.g., `cmake . -Bbuild -Ddriver=RPi`

2bndy5, 2020-11-07:
>I can show examples as needed.

Yes please. BTW, a PR is always welcome.

AndruePeters, 2020-11-09:
Just got back from an anniversary trip and catching up on work. I'll give an example within the next couple days. 
I don't mind submitting a pull request, but I wanted to see if this idea would be encouraged first. 

It will take some time, but shouldn't be too bad. In the meantime, look at https://github.com/lefticus/cpp_starter_project and how that CMakeLists.txt is setup. Pay attention to the use of `project_options` and `project_warnings`. These are "fake" libraries that are only used to attach compiler options and warnings to the rest of the project. It provides a modular method of doing so. This is the general style I'd use to setup the CMake stuff.

2bndy5, 2020-12-10:
I can think of a couple reasons to switch to CMake:
1. doxygen doesn't parse the *Makefile* correctly since it doesn't have a file extension, and doxygen assumes it is a C source code file by default. I couldn't get doxygen to ignore parsing the *Makefile* without excluding it from the docs entirely. Thus, doxygen outputs a warning about reaching the EoF while inside nested comments; actually, (I think) this is because it sees `/*` as the beginning of a comment even though they're part of a directory/file path. 
2. building the python wrapper with boost.python is currently being done a via the *Makefile.inc*, but using pybind11 (a separate issue I'm looking into) allows using CMake directly from *setup.py*

Reason 1 is more important because we can have doxygen elevate warnings to errors (using Doxyfile's WARN_AS_ERROR tag) and immediately fail generating docs. Without failing fast on warnings, the doxygen build action I put together will always result in "passed" if no actual errors are encountered. Doxygen however also treats *.txt files as C source code, but we can explicitly force *CMakeLists.txt* to be parsed like a python file because the comment syntax is the same (actual *CMakeLists.txt* content will have no bearing on documentation if the comments are done in a way that doesn't trigger doxygen's parsing).

AndruePeters, 2020-12-16:
This is in the works. Going slower than I intended.
Are there any style guidelines to apply to this?

2bndy5, 2020-12-17:
>This is in the works. Going slower than I intended.

No rush. I *really* appreciate you taking the initiative. Remember, good work is never finished...

>Are there any style guidelines to apply to this?

We have contributing guidelines, but I'm not sure how much the NASA C-style guide or PEP8 would apply to CMake

Avamander, 2020-12-17:
CMakefiles have their own style usually. We can just agree on spaces, not tabs right now and leave the rest for later.

2bndy5, 2021-03-22:
I'm still exploring the CMake docs... The [generation of the _/utility/includes.h_ file from the _/configure_ bash script](https://github.com/nRF24/RF24/blob/1187c56e4820785c1464062856ff783cd8d26d86/configure#L457) might be rendered obsolete because I see we can use CMake to specify which extra support folder (from _/utility_ directory) to include in the build process, but [_/RF24_config.h_ would need to compensate](https://github.com/nRF24/RF24/blob/1187c56e4820785c1464062856ff783cd8d26d86/RF24_config.h#L53) for this idea. I'm still brainstorming here; I have yet to try CMake out.

AndruePeters, 2021-03-22:
That sounds like it would be a good way. I lost all the stuff I had done.
I'll try to get something up there within the next two weeks, though. The
basics weren't complicated.

2bndy5, 2021-03-22:
@AndruePeters if you want, I can create a branch called "use-cmake"

AndruePeters, 2021-03-24:
Yes please.  Sent from Mail for Windows 10 From: BrendanSent: Monday, March 22, 2021 1:54 PMTo: nRF24/RF24Cc: Andrue Peters; MentionSubject: Re: [nRF24/RF24] CMake Support (#676) @AndruePeters if you want, I can create a branch called "use-cmake"

2bndy5, 2021-03-25:
The "use-cmake' branch is created.

I just found out that CMake can take in a _config.h.in_, replace @embedded_variables@ with CMake's variables' values (including ones commented out), and output the result as _config.h_. This would actually be better than modifying the RF24_config.h manually (like I mentioned above).

Still learning here, but I'm determined enough to try this stuff out (on a separate/simple repo). 

I'm still not that much help (obviously), but it doesn't sound like you need it.

AndruePeters, 2021-03-25:
Thanks. I'll start working there on it there. I'll take a look at that file
you were mentioning. You think it's okay if it doesn't support SSH builds
at this point?

2bndy5, 2021-03-25:
> You think it's okay if it doesn't support SSH builds at this point?

We can patch it up later. Personally, I think it's useful, but I don't like that it requires disabling security measures

AndruePeters, 2021-04-07:
Sweet. I'm on this now. Got it forked and working on it. Sorry about all the delays. Full time job + grad school + family = crazy busy.

AndruePeters, 2021-04-07:
Is there a reason there's currently no source directory? I'm mostly curious on the decisions for the project structure.

2bndy5, 2021-04-07:
>Is there a reason there's currently no source directory?

Because the lib is so old... Using a src directory wasn't compatible with older versions of the Arduino IDE.

>Sweet. I'm on this now.

That's so good to hear. I almost pinged you yesterday because I'm trying to incorporate examples for the Pico SDK (which tells on CMake), and I have a vague idea of how to do it. Mostly concerned about building for Linux vs building for Pico SDK.

AndruePeters, 2021-04-11:
The approach I'm taking is fairly flexible. It's mostly as simple as defining a variable that checks for what version and then adds the proper files. 

I know nothing about Pico SDK, but I think this approach should work. I'll try to have a basic example done by Friday.

2bndy5, 2021-04-11:
we just merged in a new feature to master (it shouldn't affect Linux builds - excluding work one the python wrapper).

@AndruePeters could you make a commit to your fork (I assume the use-cmake branch would be most applicable), so I can at least _see_ your direction. Maybe I can mimic the same direction on the rp2xxx branch (for Pico SDK support). The work doesn't have to be complete; I guess I'm just not good with "teasers".

AndruePeters, 2021-04-14:
I just pushed to my user branch. CMake itself is configuring everything. I'm still testing the actual build process, though.

Currently to run:
`cmake . -Bbuild -DRF24_DRIVER=wiringPi `

You can look in utility/CMakeLists.txt for how it's currently being handled.

2bndy5, 2021-04-14:
@kripton, please have a look at [@AndruePeters fine work](https://github.com/AndruePeters/RF24/tree/use-cmake). This is what I was originally trying to do on the rp2xxx branch. 

@AndruePeters I'm getting a build error at [this line in _RF24_config.h_](https://github.com/AndruePeters/RF24/blob/a9fe9defff4d3d5e9c12c7a762c9467a0ba16249/RF24_config.h#L53). I still propose the idea to modify this line so that CMake will automatically adjust this when copying the file into the binary source tree. Or maybe there's a better way (I haven't tested this idea yet).
### CMakeLists.txt
```text
add_subdirectory(utility)
configure_file(RF24_config.h RF24_config..h)
```
[link to CMake docs about `configure_file` cmd](https://cmake.org/cmake/help/v3.20/command/configure_file.html)

### RF24_config.h
```cpp
#include "utility/@RF24_DRIVER@/RF24_arch_config.h"
```
--------------
Its hard to decide what value the `RF24_DRIVER` var should default to; I'm leaning toward `SPIDEV` since its based on the Linux kernel, but the _/configure_ script defaults to a value based on what processor is detected. 

See #642 also as the absence of the arm-linux-gnueabihf compiler nullifies some of the current [CCFLAGS being set in the _/configure_ script](https://github.com/nRF24/RF24/blob/0b60c0b457ac75c60fbe9f9a508223cab703509a/configure#L147) (exemplified in [this comment's posted output](https://github.com/nRF24/RF24/issues/642#issuecomment-803759895)).

AndruePeters, 2021-04-15:
That's an easy fix. I'll get that up before midnight probably. 
Issue was forgetting to add the directory to the include list.

AndruePeters, 2021-04-15:
I can make it default to SPIDEV. The current design allows a wide range of flexibility with how RF24_DRIVER gets set. I can implement the config script's behavior from within CMake and have it also default based upon the chip.

I can fix the build error just fine like I mentioned in the previous comment. However, I don't completely follow what you mean about modifying the line for copying into the binary source tree. Can you explain?

2bndy5, 2021-04-15:
>Can you explain?

Its preformed by CMake's `configure_file` cmd. Basically, it takes the 1st arg as the input file from the src dir (root folder for RF24), and it replaces any syntax `@VAR_NAME@` or `{VAR_NAME}` with the var value when it copies the input file to the build dir. The 2nd arg is the output file name used in the build dir. All of this is explained in the [CMake docs](https://cmake.org/cmake/help/v3.20/command/configure_file.html), but I got the idea reading through the [CMake basic tutorial](https://cmake.org/cmake/help/v3.20/guide/tutorial/index.html#adding-a-version-number-and-configured-header-file). The only thing that might be a problem is the fact that `@RF24_DRIVER@` would be enclosed in quotes, but since the CMake docs boast that the `configure_file` cmd also works for commented defines, I hope it would also work inside quotes.

>I can implement the config script's behavior from within CMake and have it also default based upon the chip

I figured as much. I think we could also make use of CMake's [`find_library`](https://cmake.org/cmake/help/v3.20/command/find_library.html) cmd to detect if WiringPi, MRAA, or LittleWire libs are installed.

AndruePeters, 2021-04-15:
Oh. Gotcha. I don't think it's necessary here if this is the only reason for wanting it. I think it will bring more confusion that anything. The build error was mostly an oversight from my part. 

Speaking of the `find_library`, what should be done about WiringPi? I really miss that it will be gone, but I'm about to switch away from it in my own projects. What's the official status of support for here, and should I still add it?

2bndy5, 2021-04-15:
>That's an easy fix. I'll get that up before midnight probably.
Issue was forgetting to add the directory to the include list.

@AndruePeters The problem is that _/utility/includes.h_ doesn't exist (on github - [see gitignore](https://github.com/nRF24/RF24/blob/0b60c0b457ac75c60fbe9f9a508223cab703509a/.gitignore#L22)). If the user never runs the _/configure_ script, then the file never gets copied from the appropriate driver dir in _/utility_. So, I'm not confident that simply including the _/utility_ dir is the right fix. Did I understand that fix correctly?

>What's the official status of support for [WiringPi] here, and should I still add it?

Yes, RF24 lib still supports it indiscriminately (as to which unofficial fork is installed - our workflow used [CoRfr/wiringPi](https://github.com/CoRfr/WiringPi))I remember a while back that @TMRh20 mentioned one of the network libs (RF24Gateway) doesn't support WiringPi.

>I really miss that it will be gone

Its not gone; its been re-incarnated by a multitude of offshoots. It seems Gordon's attempt to retire the lib was marginally ignored 😞 .

kripton, 2021-04-18:
Current "use-cmake"-branch on top of current "rp2xxx"-branch and modified to make the two approaches (building a shared library for Linux and including RF24 into a pico-sdk based project) working: https://github.com/nRF24/RF24/issues/727#issuecomment-822050290. Sorry for cross-posting, just wanted to show that it's possible and as a response to https://github.com/nRF24/RF24/issues/676#issuecomment-817236011 ;)

2bndy5, 2021-04-18:
Erroneous output posted by @kripton in https://github.com/nRF24/RF24/issues/727#issuecomment-822063678:
>RF24_config.h:59:14: fatal error: utility/includes.h: No such file or directory

@andruePeters this is the error I was referring to in https://github.com/nRF24/RF24/issues/676#issuecomment-819922691. Although, the line number is probably off a little because its using the rp2xxx branch's _RF24_config.h_.

AndruePeters, 2021-04-18:
Yeah. I'll take a look at these. 

Please critique this idea. I don't think that the user should necessarily have to run a configure script, and I'm trying to think of a decent design that does not require the configure script. My view is that everything should be handled from within CMake, and that the user can define the proper CMake Variables (or default to auto config) within their own project.

So, if I was in a project for some Linux project using RF24, I'd probably expect to `set (RF24_RPI_DRIVER ON)`, or even pass this as part of building the project. My view of the use of CMake is to due away with the configure script. Obviously, this makes the CMake slightly more complicated. However, this is still doable. 

I implemented CMake at my job where we have a lot of custom code generation scripts that are ran. I think we should find a way to remove the utility file completely, or just see if you'd be okay with creating a script that only creates this file. That script can be called by CMake as part of the compilation or generation process. 

For the complexity of my job, a combination of both was used. For this, the latter might be sufficient. I just like removing unnecessary moving parts as needed. Thoughts?

2bndy5, 2021-04-18:
> My view of the use of CMake is to do away with the configure script. Obviously, this makes the CMake slightly more complicated. However, this is still doable.

Yes, I completely agree. We could also remove the current _Makefile_ in the root folder.

>I think we should find a way to remove the utility file completely

This was my intention when I proposed using Cmake's `configure_file()`. But, like I said (multiple times now), [the line in _RF24_config.h_](https://github.com/nRF24/RF24/blob/b0a91545dc5608649e3a6aafcb7357d256d17973/RF24_config.h#L59)
```cpp
#elif (!defined (ARDUINO)) // Any non-arduino device is handled via configure/Makefile
    // The configure script detects device and copies the correct includes.h file to /utility/includes.h
    // This behavior can be overridden by calling configure with respective parameters
    // The includes.h file defines either RF24_RPi, MRAA, LITTLEWIRE or RF24_SPIDEV and includes the correct RF24_arch_config.h file
    #include "utility/includes.h"
```
needs to be modified because it is currently written under the assumption that the _configure_ script was run prior to `make; sudo make install`

---------
>or just see if you'd be okay with creating a script that only creates this file. That script can be called by CMake as part of the compilation or generation process.

This idea sounds like you want to port the _configure_ script to a new _configure_linux_driver.cmake_ script that's called from a _CMakeLists.txt_. The current _configure_ script copies the _includes.h_ file from the a driver's subdirectory into the utility directory. 
```bash
echo "[SECTION] Preparing configuration."
cp ${DRIVER_DIR}/includes.h ${ARCH_DIR}/includes.h
```
Since the only error we've come across is because this line isn't executed by CMake, I'm interested in your ideas about this.

2bndy5, 2021-04-18:
If we go with porting the _configure_ script (in its entirety) to a _configure_linux_driver.cmake_ script, then we might be able to keep the existing _configure_ and _Makefile_ files for compatibility with user projects that only use the `./configure; make; sudo make install` paradigm (something I know @Avamander would appreciate). This way the _RF24_config.h_ file wouldn't need to be modified at all (assuming we copy the _includes.h_ file from the driver subdirectory into the _utility_ directory).

AndruePeters, 2021-04-20:
I'm all for @2bndy5 's idea; at least at this point in the process. Whatever y'all decide in the future is different. For now keeping the _configure_ and _Makefile_ file for compatibility would be good.

AndruePeters, 2021-04-26:
Pushed an update to my fork that should fix some issues. 
The header file is now copied into ${CMAKE_BINARY_DIR}/utility.

If you run `cmake . -Bbuild`, then `build` will be your binary directory. 

One of the updates includes changing some include statements. I don't have the means to test the different drivers, but wiringPi did work.

2bndy5, 2021-05-05:
just tried your latest changes and this is the error output I get:
```shell
PS C:\Users\%username%\Documents\GitHub\RF24> cmake -G "NMake Makefiles" . -DRF24_DRIVER=SPIDEV -Bbuild
-- The CXX compiler identification is MSVC 19.28.29914.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/VC/Tools/MSVC/14.28.29910/bin/Hostx86/x86/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Warning at cmake/Cache.cmake:28 (message):
  ccache is enabled but was not found.  Not using it
Call Stack (most recent call first):
  CMakeLists.txt:23 (include)

Using driver: SPIDEV
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/%username%/Documents/GitHub/RF24/build
PS C:\Users\%username%\Documents\GitHub\RF24> cd build
PS C:\Users\%username%\Documents\GitHub\RF24\build> nmake

Microsoft (R) Program Maintenance Utility Version 14.28.29914.0
Copyright (C) Microsoft Corporation.  All rights reserved.

[ 16%] Building CXX object CMakeFiles/RF24.dir/RF24.cpp.obj
RF24.cpp
C:\Users\%username%\Documents\GitHub\RF24\RF24_config.h(53): fatal error C1083: Cannot open include file: 'utility/includes.h': No such file or directory
NMAKE : fatal error U1077: '"C:\Program Files\CMake\bin\cmake.exe"' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.28.29910\bin\HostX86\x86\nmake.exe"' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.28.29910\bin\HostX86\x86\nmake.exe"' : return code '0x2'
Stop.
```
I don't know what the result of copying the includes.h file should look like, but it doesn't seem to be copying the includes.h file? I see in the build folder there are:

- build\CMakeFiles\RF24.dir
- build\utility\SPIDEV\CMakeFiles\SPIDEVDriver.dir

2bndy5, 2021-05-10:
I recently looked into having 2 differently named CMakeLists.txt files (for the unreleased RF24Log lib), and it turns out CMake doesn't allow you to pick a different name for the CMakeLists.txt files. The suggested alternatives (found when googling) were:
- use an `OPTION` to customize the CMakeLists.txt behavior (this is most preferred)
- use subdirectories that each have a CMakeLists.txt file for different desired behaviors (this may seem less transparent and would involve maintaining more than 1 CMakeLists.txt file)

@kripton I like the tactic you implemented in the top-level CMakeLists.txt file (involving `if (PICO_SDK_PATH) ... return()`). Is there a reason to change this approach?

kripton, 2021-05-10:
No specific reason to change the approach. I just thought that it might be cleaner since the two styles to build/use the library are totally independent. So of course we can also leave it with the `if (PICO_SDK_PATH) ...` solution in the CMakeLists.txt file.

About the names: It would not matter if the name of the cmake-file for pico-sdk based projects is not CMakeLists.txt. That only matters if you actually call `cmake` in/for the RF24 library. For pico-sdk projects, one needs to `include()` the file anyways, specifying the file name: https://github.com/kripton/rp2040-dongle/blob/kripton-rf24/src/CMakeLists.txt#L14
So the name wouldn't really matter, one just has to know it.

But as I said, the solution I proposed is still perfectly fine with me :)

2bndy5, 2021-05-11:
@AndruePeters I took another stab at building the lib on your fork's "use-cmake" branch, and I got it to work by changing a coupe things.
1. destination of copied _includes.h_ is now in `${CMAKE_SOURCE_DIR}/utility` (not `${CMAKE_BINARY_DIR}/utility`). This change also nullified the `target_include_directories()` line in _copy_includes.cmake_.
2. all `#include` statements that point to files within the RF24 lib now use quotes instead of angle brackets. This change had to be applied to _RF24_config.h_ (line 53 only), and all _includes.h_ and _RF24_arch_config.h_ files within the RPi, SPIDEV, MRAA, wiringPi, and LittleWire driver folders.
3. I also changed the casing for the project name as that's what's used to create symlinks. For instance, RF24 lib currently installs _librf24.so_, but with a project title "RF24", the symlink name(s) would be _libRF24.so_

Our next (and hopefully final) challenge will be writing install rules and getting the _examples_linux_ folder to build. Currently, the `./configure; make; sudo make install` paradigm creates several interlinked _librf24.so_ files:
- _librf24.so.1_
- _librf24.so.1.4_
- _librf24.so.1.4.0_

I haven't been able to mimic this output using CMake. Although, I don't think versioned symlinks are a detrimental requirement, but I could be wrong because of users updating their RF24 lib's installation...

Reminder: We need to run `/sbin/ldconfig` after installing the RF24 lib to be able to link the examples to the RF24 lib's installation (not the lib's source).

2bndy5, 2021-05-11:
I just figured out that `set_target_properties(rf24 PROPERTIES VERSION 1.4.0)` will produce 2 different symlinks:
1. librf24.so
2. librf24.so.1.4.0

Still trying to find out how to create a major and major+minor symlinks in addition to the symlinks listed above...

2bndy5, 2021-05-12:
Also forgot to mention in my last comment that the _interrupt.c_ sources don't compile with a C compiler because it eventually `#include <stdexcept>`  which is a C++ std lib. When explicitly setting the LANGUAGES property as only CXX (as AndruePeters implemented), the project skipped any *.c  source files (even when they were explicitly named as part of the project's source files). After switching to the implicit default value of the LANGUAGES property (set to `C CXX`), I ran into this problem.

The compiler traced the includes (using SPIDEV as the utility driver but also applies to RPi utility driver):
_interrupt.c_ ➡️ _interrupt.h_ ➡️ _RF24_arch_config.h_ ➡️ _spi.h_ ➡️ stdexcept

Changing the suffix from _interrupt.c_ to _interrupt.cpp_ and setting the LANGUAGES property to `C CXX` fixed this problem.

2bndy5, 2021-05-12:
ok, I've got the RF24 lib building and installing almost exactly like the current build system does. The `RF24_DRIVER` var is automatically set depending on if MRAA, wiringPi, or LittleWire libs' installation are found. If the Broadcomm CPU (specifically excluding RPi3 & RPi4 like the configure script) is detected, then the `RF24_DRIVER` defaults to the RPi driver. Otherwise `RF24_DRIVER` will default to the SPIDEV driver when all else fails to be detected. Additionally, I've given the _examples_linux_ folder a separate CMake project to build the examples from the installed shared RF24 lib symlink. So far, testing has been successful! Watch out for a new workflow that tests the CMake build system for Linux platforms 😉 .

@AndruePeters Could you look over the [rp2xxx branch](https://github.com/nRF24/RF24/tree/rp2xxx) and tell me if I did anything unconventional or wrong. The biggest modifications are in the _/utility/CMakeLists.txt_ and the top-level _CMakeLists.txt_. I'm thinking about making another .cmake module to compartmentalize the CPU detection and set compiler-friendly flags accordingly (see paragraph below).

## direct solution to #642
The only thing left that I want to figure out before a PR is how to determine which GNU compiler CMake has selected. If its using the arm-gnu compiler, it still shows as GNU in the CMAKE_COMPILER_ID var. However, the [current configure script injects optimization flags to the compiler](https://github.com/nRF24/RF24/blob/0b60c0b457ac75c60fbe9f9a508223cab703509a/configure#L64-L175) (which at the time assumed using the arm-gnu compiler), but these optimization flags are not accepted by the x86/x64 GNU compiler (which is currently shipped with Ubuntu when running on a RPi).

2bndy5, 2021-05-12:
# 🤦🏼‍♂️ 
I forgot about the python wrapper's install method which relies on the _configure_ script's generated _Makefile.inc_ for compiler flags and lib version number. The docs say that the RF24 lib needs to be fully installed (symlink and all) before installing the python wrapper, but I have my doubts about that because the _setup.py_ file invokes the compiler via python's std libs.

Hmm

We could just mandate that the lib be built traditionally (using _configure_ script) for installing the python wrapper until I've had time to explore pybind11 more intensively (tagging nRF24/pyRF24#3).

2bndy5, 2021-05-19:
Here is a useful [link to the CMake FAQ about selecting a compiler](https://gitlab.kitware.com/cmake/community/-/wikis/FAQ#how-do-i-use-a-different-compiler); this is a must-read because having CMakeLists.txt force using a specific compiler (via CMake's `set()` cmd) is frowned upon.

AndruePeters, 2021-05-26:
Hey,

I hope I've given enough for y'all to take and work on. I've realized I currently just don't have enough time to properly work and test these changes. Engineering job + family + grad school, etc. As long as this thread isn't closed, I can follow and comment on things as needed. But, I just don't have the time to do the work (despite being the one to open the issue).

This book is a great resource on using modern CMake.

The only design critique I have was regarding the location of the utility.h file. I can't seem to find the comment addressing that. I think the best fix it to keep it in CMAKE_BINARY_DIR as it keeps a clean build that way. The fix is just to add CMAKE_BINARY_DIR to the includes for that particular target.

AndruePeters, 2021-05-26:
> ok, I've got the RF24 lib building and installing almost exactly like the current build system does. The `RF24_DRIVER` var is automatically set depending on if MRAA, wiringPi, or LittleWire libs' [...]

I didn't see anything with a cursory glance. Take a long at my previous post on here for a good book recommendation.

2bndy5, 2021-05-30:
### About integrating remote installation of the RF24 lib (from host machine to target machine)
>>You think it's okay if it doesn't support SSH builds at this point?
>
>Yes, we can patch it up later. Personally, I think it's useful, but I don't like that it requires disabling security measures

@AndruePeters I think if I figure out how to specify where the linker looks for cross-compiling symlinks (& installation headers like  with _/usr/local/include/RF24_), we can at least have a set of files to copy over remotely with ssh (probably using a separate bash script with elevated sudo privileges).

On the other hand, if we host our own PPA repo that aids in distributing RF24 libs (via `apt`/`yum` or maybe just whatever OS-specific package manager), then remotely building/installing the RF24 libs wouldn't be necessary. But I have no idea how to set up a PPA repo.

2bndy5, 2021-05-30:
@AndruePeters Is the _Doxygen.cmake_ module more useful to end-users? It certainly isn't useful for us because
1. we already have a Doxyfile generated, and it is uniquely tweaked to
   - avoid duplicate docs for the multiple similar utility driver sources 
   - force-output some docs that are specific to non-Linux platforms
   - parse the ino files as C++ examples
2. the Doxyfile tags set in the _Doxygen.cmake_ module aren't consistent with our Doxyfile's tags. Although this point can be rectified, I personally don't like having to read through [CMake docs about the _FindDoxygen_ module](https://cmake.org/cmake/help/latest/module/FindDoxygen.html) and then cross-reference that with the (more applicable/useful) Doxygen docs.
3. we don't use graphs, so the dot lib isn't really required
4. (considering a project may have RF24 as a submodule) end-users shouldn't need to build the RF24 docs as part of their project's docs. Unless, the project is intentionally pinned to a certain version of RF24, but there is such a big concern among nRF24 org owners for backward compatibility that this shouldn't be a problem.

I'm leaning toward removing this _Doxygen.cmake_ module, but I'm trying to imagine a use case for end-users (in a similar convenience that the testing/fuzzing options provide) - maybe for outputting PDF, XML, or latex. 

Alternatively, we could modify the _Doxygen.cmake_ module to execute the `doxygen` cmd from a `cmake -D ENABLE_DOXYGEN=ON` cmd, but this seems (to me) like it would be more complicated than it needs to be. This idea is basically [what the PicoSDK does for building docs from CMake](https://github.com/raspberrypi/pico-sdk/blob/afc10f3599c27147a6f34781b7102d86f58aa5f6/docs/CMakeLists.txt#L52-L56)

-------------
### Concerning the tests folder
The current _/tests_ folder is an unused relic from the ManiacBug dev era (prior to 2014), and it is only meant for the Arduino platform. Furthermore, the _/tests_ folder may be broken since Arduino CLI doesn't use jam files and doesn't support the pde file format. I'm mentioning this because I noticed the cmd `add_subdirectory(test)` for which the "test" folder doesn't actually exist... yet; I have an idea to test a dummy object of the RF24 class but that's not a priority right now.

@TMRh20 I'm removing the _/tests_ folder (on the rp2xxx branch) to avoid confusion with the CMake implementation's optional testing mechanisms.

2bndy5, 2021-05-30:
YEAH! I got the python wrapper to install from a CMake install of the RF24 lib 💯 . I also modified _setup.py_ to check if the RF24 lib is installed and raise an exception if the _/usr/local/lib/librf24.so_ file isn't found. The new exception prompt will tell users to install the RF24 lib first. This should avoid some confusion, but the `python3 setup.py build` step now requires `sudo` permission. If you look at [the new CMake CI steps](https://github.com/nRF24/RF24/runs/2704089521?check_suite_focus=true#step:13:1), I've devised a way to alias the symlink for boost.python that is agnostic to the compiler/OS. This also means that we can now properly test build/install the python wrapper in the linux_build workflow (had trouble doing this with the traditional build system). The Linux & python wrapper's install instructions in the docs are updated 😉 .

Reminder: the python wrapper cannot be cross-compiled when the RF24 lib is installed with CMake.

2bndy5, 2021-05-31:
ok, I've employed a toolchain file in CMake to cross-compile the lib with ease. Currently, only the arm platform is configured for cross-compiling, but more toolchain files can easily be added with some Copy-n-Paste skills. I've also created a [new "dpkg-deb" workflow](https://github.com/nRF24/RF24/actions/workflows/build_dpkg.yml) to package the lib for arm platform, but it currently only compiles the lib using `SPIDEV` & `RPi` drivers. My hope is that doing it this way will allow us to distribute Debian install-able packages (for `SPIDEV` & `RPi` drivers) with new releases (& nullify the current remote install method over ssh). Distributing deb packages with MRAA, wiringPi, and LittleWire drivers is going to be a bit trickier - I'd like to distribute the built drivers as part of _librf24-MRAA_arm.deb_, but that may not be the best practice (IDK).

I encourage anyone to download [the saved artifacts from the workflow](https://github.com/nRF24/RF24/actions/runs/893199621#artifacts) and try installing the deb file on a RPi (examples still need to be built though)... I'm going to sleep now 💤

2bndy5, 2021-06-11:
@Avamander I've been looking for a CMake style guide that we can link to and best suites contributions. [I like this one](https://gist.github.com/mbinna/c61dbb39bca0e4fb7d1f73b0d66a4fd1). Although it doesn't really talk about syntax guidelines like indentation size & where the closing parenthesis should go for multi-line cmds (CMake docs use a mix of styles). There is also an [official guideline in CMake docs](https://cmake.org/cmake/help/v3.20/manual/cmake-developer.7.html), but it focuses more on writing a "Find-Module" script and variable naming. Maybe we can explicitly specify 4-space indentations and refer to both docs in our CONTRIBUTING.md

2bndy5, 2021-08-13:
@AndruePeters The _RF24\cmake\CompilerWarnings.cmake_ doesn't seem to have the expected affect... 

I started building the new python wrappers using CMake, and the compiler warnings prevents building RF24 as a linked lib. However, compiling RF24 as a standalone lib doesn't have any warnings raised as errors (no warnings at all in fact).

The sheer amount of warnings raised as errors feel like a rude awakening (there is a lot). As it stands, I don't think the `WARINGS_AS_ERRORS` option should default to `TRUE`.

The following is 1 warning that should be ignored for cross-platform reasons
```shell
/mnt/c/Users/%USERNAME%/Documents/GitHub/pyRF24/RF24/RF24.cpp: In member function ‘void RF24::csn(bool)’:
/mnt/c/Users/%USERNAME%/Documents/GitHub/pyRF24/RF24/RF24.cpp:16:21: warning: unused parameter ‘mode’ [-Wunused-parameter]
   16 | void RF24::csn(bool mode)
      |                ~~~~~^~~~
```
It would be too much work to workaround this warning (which is raised as an error by default).

AndruePeters, 2021-08-13:
I understand that. Jason Turner and some other top C++ people recommend to treat warnings as errors. It's always been something I enable for releases, but disable for normal development. I find it gets in the way a lot, but it does force developers to submit quality code. Figure out what warnings you think should stay enabled for this code-base, and then force very PR to compile with these flags. With that said, I tend to turn it off until I'm ready for a release.

So, to address the warning:
1) Remove the -Wunused from CompilerWarnings.cmake.
   I tend to shy away from this approach because it's helped catch several bugs.

2) `(void)parameter`
   Use this approach to suppress the warnings where you want. 
   This approach might be somewhat cumbersome for the initial changes, but I believe it to be the best approach.

2bndy5, 2021-08-13:
I think the seconds approach actually raises a separate set of warnings (definitely will raise a `-Wold-style-cast` but I can get around that).

I've done the first approach already, but the warning still gets raised. I think `-Wpedantic` may be reasserting the `Wunused-parameter`

AndruePeters, 2021-08-13:
Use a static cast

static_cast<void>(argument);

2bndy5, 2021-08-13:
yep that did it!

All I have left to fix are all the `-Wconversion` warnings.

2bndy5, 2021-08-13:
Well, I fixed all the warnings... I still think its strange that the warnings don't show up when RF24 lib is built alone (they only showed up when I used RF24 lib as a dependency). Furthermore, I've been linking built/installed RF24 lib to RF24Network & RF24Mesh (all using CMake now), and the warnings never showed. I did copy (& `include()`'d) the _CompilerWarnings.cmake_ module you submitted to the other RF24* libs' CMake-4-Linux branches.

AndruePeters, 2021-08-13:
Are you adding `project_warnings` to the `target_link_libraries` calls for those libraries? 

I hate conversion warnings. They're so important, but also a pain with most interfaces.

2bndy5, 2021-08-13:
Yep. Actually I had to rename the `project_options` & `project_warnings` to `${LibTargetName}_project_*` so they could be more specific to the project when building more than 1 lib at a time

2bndy5, 2021-08-23:
I have to use c-style casts for passing a `pgm_read_ptr()` types to `printf()` for certain platforms - Teensy platforms crash when using `const_cast` while all other platforms (including Linux) only like `const_cast`. I'm disabling the extra compiler warnings until teensyduino core updates their toolchain (which is rumored to happen soon).

AndruePeters, 2021-08-23:
Only disable the warnings you specifically need to. Also, const_const is
very tricky to use properly. It has huge potential to cause UB.

https://youtu.be/iuLwHoMFP_Y

2bndy5, 2021-08-23:
ok here is the details for the problem I came across:
**deprecated-avr-comp/avr/pgmspace.h** (from [ArduinoCore-API](https://github.com/arduino/ArduinoCore-API/blob/master/api/deprecated-avr-comp/avr/pgmspace.h) - literally all boards since 2016 use this)
```c
// no printf_P defined
#define pgm_read_ptr(addr) (*(void *const *)(addr))
```
**avr/pgmspace.h** (from [Teensyduino-core](https://github.com/PaulStoffregen/cores/blob/master/teensy3/avr/pgmspace.h) for teensy3 as that's what [the errors site](https://github.com/nRF24/RF24/runs/3396450878?check_suite_focus=true))
```c
#define printf_P(...) printf(__VA_ARGS__)
// ....
#define pgm_read_ptr(addr) ({ \
	typeof(addr) _addr = (addr); \
	*(void * const *)(_addr); \
})
```
 There is some confusion in the RF24 src
- _utilities/Teensy/RF24_arch_config.h_
   ```c
   #define printf Serial.printf
   ```
   While the Teensyduino-core's Print.h (parent header to any `Serial` objects)
   ```cpp
    // format warnings are too pedantic - disable until newer toolchain offers better...
    // https://forum.pjrc.com/threads/62473?p=256873&viewfull=1#post256873
    int printf(const char *format, ...) /*__attribute__ ((format (printf, 2, 3)))*/;
    int printf(const __FlashStringHelper *format, ...);
   ```
- all calls in `printDetails()` (this where the errors originate) use
   ```cpp
   // global variables
   static const PROGMEM char rf24_model_e_str_0[] = "nRF24L01";
   static const PROGMEM char rf24_model_e_str_1[] = "nRF24L01+";
   static const PROGMEM char * const some_debug_strings[] = {
     rf24_model_e_str_0,
     rf24_model_e_str_1,
   };
   void printDetails() {
       printf_P((char*)pgm_read_ptr(some_debug_strings[index]), ...);
   }
   ```
   While RF24_config.h conditionally defines
   ```c
   #else //Everything else
       #include <Arduino.h>
   // ...
       #if !defined pgm_read_ptr || defined ARDUINO_ARCH_MBED
           #define pgm_read_ptr(p) (*(p))
       #endif
   ```

I feel like this belongs in a separate issue, but posting it here (for now). I have disabled the `-Wold-style-cast` only (re-enable the rest by default).