nRF24/RF24 #677: Error setting options in python3

tigertone, 2020-11-07:
I’m trying to change some of the default options (using the same command that worked with an older version of the code). The pingpair_dyn example runs as expected, but when I add the following line...

radio.setPALevel(RF24_PA_MAX);

Then I get the error below. Is there something simple wrong?

pyRF24/examples/pingpair_dyn/
================ SPI Configuration ================
CSN Pin          = 0
CE Pin           = Custom GPIO22
SPI Speedz       = 10 Mhz
================ NRF Configuration ================
STATUS           = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1     = 0xf0f0f0f0e1 0xf0f0f0f0d2
RX_ADDR_P2-5     = 0xc3 0xc4 0xc5 0xc6
TX_ADDR          = 0xf0f0f0f0e1
RX_PW_P0-6       = 0x20 0x20 0x00 0x00 0x00 0x00
EN_AA            = 0x3f
EN_RXADDR        = 0x02
RF_CH            = 0x4c
RF_SETUP         = 0x07
CONFIG           = 0x0e
DYNPD/FEATURE    = 0x3f 0x04
Data Rate        = 1MBPS
Model            = nRF24L01+
CRC Length       = 16 bits
PA Power         = PA_MAX
Traceback (most recent call last):
  File "pingpair_dyn.py", line 65, in <module>
  radio.setPALevel(RF24_PA_MAX);
Boost.Python.ArgumentError: Python argument types in
    None.setPALevel(RF24, rf24_pa_dbm_e)
did not match C++ signature:
    setPALevel(RF24 {lvalue}, unsigned char, bool level)

TMRh20, 2020-11-07:
the api changed recently. It should work like that, but allows setting the
LNA gain now:

setPALevel(RF24 {lvalue}, unsigned char, bool level)

or

radio.setPALevel(RF24_PA_MAX,1);

tigertone, 2020-11-07:
It works now. Thanks!

2bndy5, 2020-11-07:
boost.python doesn't handle default values for optional variables very well. It would require extra "wrapping" to compensate for the API change that was added.

2bndy5, 2020-11-10:
I just tested fix in above referenced commit; this will not be a problem as of next release. You'll be able to use 1 arg or 2 args for `setPALevel()`. 🤓

closed 2020-12-14 by 2bndy5