nRF24/RF24 #682: my TX FIFO remains full

Hmissa, 2020-11-12:
Hi, 
I have adapted the RF24 with my STM32, but I had a problem when sending it, it always sends the first data and ignores the other data.
I had as a conclusion that the TX FIFO always remains full.
When I added the flush_tx function at the end of the write function, my application runs very well.

Any comment?

TMRh20, 2020-11-12:
Does the happen if you run any of the examples? Can you modify an example or post a small code snippet that demonstrates this behaviour?

2bndy5, 2020-11-12:
What does `write()` return?
     - If it always returns false, then the payload will always remain in the TX FIFO. Thus, every consecutive call to `write()` will keep retrying to send the artifact payload in the top level while writing a new payload to any available levels of the TX FIFO. 

The `write()` function:
1. writes a new payload to the TX FIFO
2. holds CE pin HiGH until transmission fails or succeeds, then returns the CE pin LOW
3. clears the MAX_RT flag (which indicates that last transmission failed) upon function end. This step is needed for continued operation of the TX FIFO.

Please tell me you didn't use `setAutoAck(false)` on only the receiving node(s) because that is a sure way to get the behavior you described. If this is not the case, then your transmitter is not getting the ACK packets it's waiting for, which is usually a power stability problem for the receiving node (or it can also be caused by too many receivers listening to the same address at the same time).

Hmissa, 2020-11-16:
@TMRh20  The example is Getting started
I did'nt have access to my hardware this week, next week I modify an example

Hmissa, 2020-11-16:
@2bndy5 : Concerning the problem of power supply stability I put a capacity of 100uF between VCC and GND.
For the other remarks I will check them next week.
Thanks

2bndy5, 2020-11-20:
@Hmissa I forgot about [this `if` block that clears the TX FIFO](https://github.com/nRF24/RF24/blob/a0cd707137607842f0ef782f12f2e40409a12d38/RF24.cpp#L819-L822) on failed transmissions. So, my initial assessment does not hold up (unless your application is using a mix of `write()` and other `write*()` calls). It would help to see your code (if possible) and know how your module behaves when running the getting started example.

Hmissa, 2021-01-04:
@2bndy5  I use flush_tx() just before the return 1 of the write function after checking the transmission success and I don't have a failed transmission.

Regarding your remark about setAutoAck (false), I put it in both nodes and it worked fine (without flush_tx). (Thanks)

2bndy5, 2021-01-04:
Does that mean this is resolved? Can we close this issue?

Hmissa, 2021-01-04:
Yes it is resolved

closed 2021-01-04 by Hmissa