nRF24/RF24 #683: [Question] 5ms delay in powerUp()?

jscrane, 2020-11-15:
Hi guys,
I have a battery-powered sensor whose loop() is:
- read adc (light + battery)
- read a DHT22 sensor
- transmit data

The third step is basically:
- powerUp
- stopListening
- send data
- powerDown

Digging into why the third step takes over 8ms, the main culprit seems to be the unconditional 5ms delay in powerUp(). Is there any value in skipping this in order to do something useful for 5ms? (Coincidentally, this is the time-to-read the DHT22 sensor.)

Thanks,
Steve

2bndy5, 2020-11-15:
I [talked about this specific delay](https://github.com/nRF24/RF24/issues/621#issuecomment-702580233) on #621 The thread is worth a read (I posted some pretty specific info pertaining to the datasheet).

Ultimately, we concluded that it should be 5 ms to compensate for some boards' lack of precision in implementing `delay()` or `delayMicroseconds()`.  Are you using your ATTiny85 for this?

jscrane, 2020-11-15:
(It's an ATTiny84 but yeah...) 

I read that thread, it is interesting; but maybe we shouldn't penalize everyone for a few bad boards? Just a though.

So I made a small change to the signature of powerUp() [in my branch](https://github.com/jscrane/RF24-1/commit/5fe7dde1c74a1312ddc944938f13dee96906c9f5). This allows me to start powering-up while reading the sensor at the same time. As a result, the loop time goes down from 14ms to 9ms, which is a big deal!

You could also imagine defaulting the delay to 5ms but allowing to pass it in to find out if 2ms was enough in a particular case?

2bndy5, 2020-11-15:
>maybe we shouldn't penalize everyone for a few bad boards? 

you just stuck my biggest nusance with this library. I'm of the mind that the software shouldn't overcompensate for faulty hardware (clones and counterfeits included).

>You could also imagine defaulting the delay to 5ms but allowing to pass it in to find out if 2ms was enough in a particular case?

you mean like `radio.powerUp(1500);` for trying to delay 1.5 ms instead of default 5 ms?

jscrane, 2020-11-15:
Yes, exactly!

2bndy5, 2020-11-15:
If your radio has an oscillator soldiered into it, then you might be able to get away with 1 ms or less (150 us minimum).

2bndy5, 2020-11-17:
I've been giving the powerUp param more thought. I only have 1 objection: internal calls to `powerUp()`. Currently this is done in [`stopListening()`](https://github.com/nRF24/RF24/blob/a0cd707137607842f0ef782f12f2e40409a12d38/RF24.cpp#L737) and in [`startListening()`](https://github.com/nRF24/RF24/blob/a0cd707137607842f0ef782f12f2e40409a12d38/RF24.cpp#L691). It might be better to have a `RF24_PWR_UP_DELAY` macro in *RF24_config.h* instead of an optional parameter to `powerUp()`

IMO the `startFastWrite()` & `startWrite()` should check the `config_reg` private member to make sure that TX mode is enabled (& radio is powered up). To compensate for when the radio isn't configured properly by the user, I would:
- add a `stopListening()` call
or
- make `startFastWrite()` & `startWrite()` return false (to prevent altering the TX FIFO)

My CircuitPython library does the former

jscrane, 2020-11-18:
There are a few more delays in RF24.cpp which could use this treatment too. Would certainly improve code readability.

2bndy5, 2020-11-18:
The delay on startup is required because the datasheet says that a PoR event takes 100 ms to complete. I see a `delay(200)` for the `XMEGA_D3` macro, but I think that's done specifically for certain boards.

The `txDelay` is there because C++ is so freaking fast, but the user could be able to add this themselves if they see an ACK payload getting cut short when calling `stopListening()`.

The delay in `errNotify()` is probably there so it doesn't get triggered more than once per call to`write*()`. Same goes for the delay after `errNotify()` is called. 

I don't fully understand the delay in `getDynamicPayloadSize()`, but it would only get triggered by a bad SPI connection. Personally, I don't think `errNotify()` or this delay should exist as they only overcompensate for faulty SPI connections (which should be indicated by `begin()` anyway).

All other delays are standard (per datasheet) or used for debouncing (`csDelay`). It's only the delay in `powerUp()` that could be detrimental for speedy application.

Although,
```c++
// Must allow the radio time to settle else configuration bits will not necessarily stick.
    // This is actually only required following power up but some settling time also appears to
    // be required after resets too. For full coverage, we'll always assume the worst.
    // Enabling 16b CRC is by far the most obvious case if the wrong timing is used - or skipped.
    // Technically we require 4.5ms + 14us as a worst case. We'll just call it 5ms for good measure.
    // WARNING: Delay is based on P-variant whereby non-P *may* require different timing.
    delay(5);
```
I'm not really sure what this is about. I don't recall the datasheet saying anything about this. In fact this is what made me think the `csDelay` was 5ms instead of 5 us (up until a couple weeks ago).

2bndy5, 2020-12-07:
Closing this due to merging of #690 .

When re-working the *timingSearch3pin.ino* for the SpenceKonde ATTinyCore, I noticed that the [settling delay times in `csn()`](https://github.com/nRF24/RF24/blob/8b2f2a952d7ac9c389be6d5a39a0ca0178a3234a/RF24.cpp#L25) could also use a macro in *utility/ATTiny/RF24_arch_config.h*. 

Additionally, the *RF24_config.h* uses macros not defined in the SpenceKonde ATTinyCore to [declare the RF24_TINY macro](https://github.com/nRF24/RF24/blob/8b2f2a952d7ac9c389be6d5a39a0ca0178a3234a/RF24_config.h#L56), but this is actually related to #646

closed 2020-12-07 by 2bndy5