nRF24/RF24 #684: CSN Pin = 0 on getting started linux

Mbasttias, 2020-11-16:
hello, I just downloaded and installed the library but when I use some getting started the pin configuration is not correct and the communication is not established with another node.

the hardware configuration message is as follows

> ================ SPI Configuration ================
> CSN Pin  	 = 0
> CE Pin  	 = Custom GPIO22
> SPI Speedz	 = 10 Mhz

when you should dial 

> CSN Pin          = CE0 (PI Hardware Driven)

Besides, if I want to try any of the examples with another node anyway, the communication is not established either as sender or receiver. Previously I had used this library in another RPi and I had not given that problem. Look for several possible solutions related to the constructor RF24 radio(22, 0), , but I couldn't fix it.

2bndy5, 2020-11-16:
>CSN Pin = 0

This is referring to CE0 (AKA GPIO8).

>the communication is not established 

This could mean a lot of things. However, if [`radio.begin()`](https://github.com/nRF24/RF24/blob/a0cd707137607842f0ef782f12f2e40409a12d38/examples_linux/gettingstarted.cpp#L67) returns true, then the SPI bus is properly setup. In fact, the output you quoted is from using `radio.printDetails()` which directly reads values from the radio's registers via SPI bus). But, you didn't post any of the registers' info (found in the section labeled "NRF Configuration") from that function's output, so I can't say for sure if SPI is behaving properly.

Did you [change this line](https://github.com/nRF24/RF24/blob/a0cd707137607842f0ef782f12f2e40409a12d38/examples_linux/gettingstarted.cpp#L51) in either (not both) of the gettingStarted.cpp examples? Without doing so, each node is trying to communicate to the same unique node instead of each other.

2bndy5, 2020-11-16:
Just realized the getting started example's `radioNumber` variable on Linux is already set up to communicate to the arduino example's `radioNumber`

>Previously I had used this library in another RPi and I had not given that problem.

It depends how long ago you last tried the library (there have been updates to the c'tor)

We simply need more details to help you here.

Mbasttias, 2020-11-17:
They are right that the communication was not established due to the radio number, I had not noticed that detail and the communication was established without problems, but I still can not change what the impression of `radio.printDetails()`

2bndy5, 2020-11-17:
>I still can not change what the impression of radio.printDetails()

Why is it so important that it CE0 instead of 0? Part of the reason is because we need the same code to work on arduino and Linux. The other part of the reason is because there is more than 1 SPI bus on RPi (although it takes some extra work to enable a second or third SPI bus).

2bndy5, 2020-11-18:
@Mbasttias I found [the commit that changed the output for CSN pin's label](https://github.com/nRF24/RF24/commit/ed35bd93cc981ca1e118556a2e530954cdb80125). This was done just before release of [v1.3.7](https://github.com/nRF24/RF24/releases/tag/v1.3.7). I can't endorse rolling back to v1.3.6 because you would lose the various important bugfixes in the later releases

2bndy5, 2020-12-14:
Output for `printDetails()` (& `printPrettyDetails()`) now prints "/dev/spidev<bus_number>.<bus_cs_pin>" about the CSN pin to avoid any further confusion. 
So calling 
```c++
RF24 radio(22, 0);
radio.begin();
radio.printDetails();
```
should say
>CSN Pin = /dev/spidev0.0

@Mbasttias I"m closing this now. If you have a better suggestion that isn't specific to the Raspberry Pi, feel free to re-open this issue and post it here.

closed 2020-12-14 by 2bndy5