nRF24/RF24 #686: [Request] Move REGISTER_MASK to print_byte_register() & replace spiTrans() with write_register()

2bndy5, 2020-11-17:
I'm trying to reduce the code size of `getDynamicPayloadSize()` by using the `read_register()` function, but the use of REGISTER_MASK is the only thing in my way. In fact it would only be useful in the `print_byte_register()` as that is the only time that a macro defined in *nRF24L01.h* is not passed to `read_register()`. So, [this line](https://github.com/nRF24/RF24/blob/a0cd707137607842f0ef782f12f2e40409a12d38/RF24.cpp#L400) would be changed to:
```c++
        printf_P(PSTR(" 0x%02x"), read_register(reg++ & REGISTER_MASK));
```
Same thing applies to `write_register()`; every call to `write_register()` passes a macro defined from *nRF24L01.h*, so REGISTER_MASK is completely useless.

I'd also like to get rid of `RF24::spiTrans()` because we can get the same job done with `write_register()` by adding a `bool is_cmd_only=false` parameter to the `write_register()` declaration. This would affect:
1. `flush_tx()`
2. `flush_rx()`
3. `reUseTX()`
4. `get_status()`

I don't really see the usefulness of returning the status byte for the first 2 affected functions (see also the additional context below). Its completely useless in the python wrapper because the macros in *nRF24L01.h* aren't exposed to python. In fact, `reUseTX()` ignores the returned status byte already, but I will modify `flush_tx()`, `flush_rx()`, and `get_status()` to continue returning the status byte for backward compatibility.

Of course `write_register()` would need some tweaking:
```c++
void RF24::write_register(uint8_t reg, uint8_t value, bool is_cmd_only)
{
    if (is_cmd_only) {
        IF_SERIAL_DEBUG(printf_P(PSTR("write_register(%02x)\r\n"), reg));
        beginTransaction();
        status = _SPI.transfer(W_REGISTER | reg);
        endTransaction();
    }
    else {
        IF_SERIAL_DEBUG(printf_P(PSTR("write_register(%02x,%02x)\r\n"), reg, value));
        #if defined(RF24_LINUX)
        beginTransaction();
        uint8_t * prx = spi_rxbuff;
        uint8_t * ptx = spi_txbuff;
        *ptx++ = (W_REGISTER | reg);
        *ptx = value;

        _SPI.transfernb((char *)spi_txbuff, (char *)spi_rxbuff, 2);
        status = *prx++; // status is 1st byte of receive buffer
        endTransaction();
        #else // !defined(RF24_LINUX)

        beginTransaction();
        status = _SPI.transfer(W_REGISTER | reg);
        _SPI.transfer(value);
        endTransaction();
    }
    #endif // !defined(RF24_LINUX)
}
```
### Additional context
- Remember that #678 proposes that the status byte be saved in every SPI transaction, meaning any RF24::function can access the status byte after any SPI transaction.
- Notice [from the datasheet](http://tmrh20.github.io/RF24/tmrh20/nRF24L01P_datasheet_v1.pdf#G1137736) that `W_REGISTER  | reg` will not affect the SPI cmds as the bit that is asserted by W_REGISTER is already asserted in all of the SPI cmds. Only exception is the ACTIVATE cmd, but that is only used by the `toggle_features()` function which already directly accesses the SPI bus.
- Before raising this issue, I've already reduced `writeAckPayload()` to use 
    ```c++
    write_payload(current, len, W_ACK_PAYLOAD | (pipe & 0x07))`.
    ```
    There was nothing special about `writeAckPayload()` that made it differ from `write_payload()`.

TMRh20, 2020-11-17:
Pretty detailed but seems to make sense & if it all works, it works. 👍

2bndy5, 2020-11-17:
>Pretty detailed 

Sorry it started out simple, but I kept finding small changes that needed a thread for changelog reasons

closed 2020-12-14 by 2bndy5