nRF24/RF24 #687: [Request] make printDetails() more human readable

2bndy5, 2020-11-17:
I find it a little annoying when I have to open the data sheet and translate what `printDetails()` is ouputting.

### show don't tell
Instead of gettting long-winded here, I'll show the output from my circuitpython library (modified for this issue):
```
Is a plus variant_________True
Channel___________________76 (~ 2.476 GHz)
RF Data Rate______________1 Mbps
RF Power Amplifier________-12 dbm
RF Low Noise Amplifier____Enabled
CRC bytes_________________2
Address length____________5 bytes
TX Payload lengths________32 bytes
Auto retry delay__________3500 microseconds
Auto retry attempts_______5 maximum
Re-use TX FIFO____________False
Packets lost on current channel_____________________15
Retry attempts made for last transmission___________3
IRQ on Data Ready___Enabled    Data Ready___________False
IRQ on Data Fail____Enabled    Data Failed__________False
IRQ on Data Sent____Enabled    Data Sent____________True
TX FIFO full__________False    TX FIFO empty________True
RX FIFO full__________False    RX FIFO empty________True
Multicast___________Allowed    Custom ACK Payload___Disabled
Dynamic Payloads____Enabled    Auto Acknowledgment__Enabled
Primary Mode_____________TX    Power Mode___________Standby-I
TX address____________ 0x65646f4e31
Pipe 0 ( open ) bound: 0x65646f4e31
                expecting 32 byte static payloads
Pipe 1 ( open ) bound: 0x65646f4e32
                expecting 32 byte static payloads
Pipe 2 (closed) bound: 0x65646f4ec3
Pipe 3 (closed) bound: 0x65646f4ec4
Pipe 4 (closed) bound: 0x65646f4ec5
Pipe 5 (closed) bound: 0x65646f4ec6
```
Notice the addresses for pipes 2-5 automatically include the first 4 bytes from pipe 1. This is for people that don't read the docs/datasheet. For AutoACK settings that use per pipe control, I use a binary representation instead of "Enabled/Disabled" (I would do the same for Dynamic Payloads but that's not a feature for this library).

This proposal literally interprets all registers' info (except carrier wave stuff). Is this too much info? Would it increase the compile size too much?

I've been trying to hold off opening this issue until we incorporate RF24Log (nudge to @wmarkow 😃 )

TMRh20, 2020-11-18:
I would maybe suggest reducing the amount of info to that specifically needed for troubleshooting if it works out to be much larger than the current iteration, and even shortening the naming convention for itemization if needed. I agree with it being a bit painful referencing the datasheet for everything and converting hex to binary.

2bndy5, 2020-11-18:
thanks for the feedback, that's exactly what I was looking for.
Currrently, the "TX FIFO Full" & "RX FIFO Empty" info is fetched using the STATUS byte, but if we capture the FIFO_STATUS register, we can output a little more precision on that matter. I have found the FIFO info very useful when debugging the library but not as much for debugging applications. Perhaps a `bool show_fifo` parameter can be used for dev purposes.

To reduce the output to reflect only what is output from current implementation, it would end up looking like
```
Is a plus variant_________True
Channel___________________76 (~ 2.476 GHz)
RF Data Rate______________1 Mbps
RF Power Amplifier________RF24_PA_LOW
CRC bytes_________________2
Packets lost on current channel_____________________15
Retry attempts made for last transmission___________3
IRQ on Data Ready___Enabled    Data Ready___________False
IRQ on Data Fail____Enabled    Data Failed__________False
IRQ on Data Sent____Enabled    Data Sent____________True
TX FIFO full__________False    RX FIFO empty________True
Multicast___________Allowed    Custom ACK Payload___Disabled
Dynamic Payloads____Enabled    Auto Acknowledgment__Enabled
Primary Mode_____________TX    Power Mode___________Standby-I
TX Payload lengths________32 bytes
TX address____________ 0x65646f4e31
Pipe 0 ( open ) bound: 0x65646f4e31
                expecting 32 byte static payloads
Pipe 1 ( open ) bound: 0x65646f4e32
                expecting 32 byte static payloads
Pipe 2 (closed) bound: 0x65646f4ec3
Pipe 3 (closed) bound: 0x65646f4ec4
Pipe 4 (closed) bound: 0x65646f4ec5
Pipe 5 (closed) bound: 0x65646f4ec6
```

TMRh20, 2020-11-18:
I think ideally, I would like to see something like the following for the main output:
```
Channel___________________76 (~ 2.476 GHz)
RF Data Rate______________1 Mbps
RF Power Amplifier________RF24_PA_LOW
RF Low Noise Amplifier____Enabled
CRC bytes_________________2
Address length____________5 bytes
TX Payload lengths________32 bytes
Auto retry delay__________3500 microseconds
Auto retry attempts_______5 maximum
Packets lost on current channel_____________________15
Retry attempts made for last transmission___________3
Multicast___________Allowed    Custom ACK Payload___Disabled
Dynamic Payloads____Enabled    Auto Acknowledgment__Enabled
Primary Mode_____________TX    Power Mode___________Standby-I
TX address____________ 0x65646f4e31
Pipe 0 ( open ) bound: 0x65646f4e31
                expecting 32 byte static payloads
Pipe 1 ( open ) bound: 0x65646f4e32
                expecting 32 byte static payloads
Pipe 2 (closed) bound: 0x65646f4ec3
Pipe 3 (closed) bound: 0x65646f4ec4
Pipe 4 (closed) bound: 0x65646f4ec5
Pipe 5 (closed) bound: 0x65646f4ec6
```
Just a quick assessment, I think this would show the most problematic settings. Not sure how usefull the IRQ info is.

2bndy5, 2020-11-18:
great notes!

>Not sure how usefull the IRQ info is

That was only there because the current implementation includes the CONFIG & STATUS register. I have no problem getting rid of it.

2bndy5, 2020-11-18:
>Power Mode___________Standby-I

I forgot to mention that this is evaluated using circuitpython's equivalent to `digitalRead()` about the CE pin. I'm not sure about calling `digitalRead()` on an OUTPUT pin (especially in Linux). Alternative ideas?

2bndy5, 2020-11-18:
>RF Low Noise Amplifier____Enabled

This seem very particular info because it pertains to Si24R1 or nRF24L01 (non-plus) chips. Do you really want this output?

Avamander, 2020-11-18:
> > RF Low Noise Amplifier____Enabled
> 
> This seem very particular info because it pertains to Si24R1 or nRF24L01 (non-plus) chips. Do you really want this output?

It's useful when you do need it.

> This proposal literally interprets all registers' info (except carrier wave stuff). Is this too much info? Would it increase the compile size too much?

I personally think we could just have a separate function that uses more storage but prettifies the output, but also keep the current one in case someone has grown accustomed to it or doesn't want to spend more bytes on their MCUs.

2bndy5, 2020-11-18:
>It's useful when you do need it.

Thanks. I was just trying to avoid capturing the whole register for 1 bit, but you're right (I was just getting lazy).

>I personally think we could just have a separate function

Ok, I can call it `printDetailsDeciphered()` (or `printDetailsForHumans()` 😆 ). I'm currently only going with [TMRh20's suggested output](https://github.com/nRF24/RF24/issues/687#issuecomment-729359899), not what's in the OP. I probably won't do a `bool dev_only` parameter either.

A little off topic, but maybe we should alter the CSN pin's output label to show `/dev/spi<bus_numb>.<bus_device>`. This might satisfy confusion from #684

2bndy5, 2020-11-18:
the following results are from compiling the new ManualAcknowledgments.ino example against the Arduino Nano:

### stable release v1.3.9 without printing debug info
>Sketch uses 6162 bytes (20%) of program storage space. Maximum is 30720 bytes.
Global variables use 285 bytes (13%) of dynamic memory, leaving 1763 bytes for local variables. Maximum is 2048 bytes.

### my fork (master branch) without printing debug info
>Sketch uses 6162 bytes (20%) of program storage space. Maximum is 30720 bytes.
Global variables use 287 bytes (14%) of dynamic memory, leaving 1761 bytes for local variables. Maximum is 2048 bytes.

Extra 2 bytes of global variables is likely from adding private members `status` & `_is_p_variant`.

using `printDetails()`
>Sketch uses 9062 bytes (29%) of program storage space. Maximum is 30720 bytes.
Global variables use 287 bytes (14%) of dynamic memory, leaving 1761 bytes for local variables. Maximum is 2048 bytes.

using `printPrettyDetails()`
>Sketch uses 9676 bytes (31%) of program storage space. Maximum is 30720 bytes.
Global variables use 287 bytes (14%) of dynamic memory, leaving 1761 bytes for local variables. Maximum is 2048 bytes.

### `printPrettyDetails()` output (so far)
```
SPI Frequency_____________10 Mhz
Channel___________________76 (~ 2476 MHz)
RF Data Rate______________1MBPS
RF Power Amplifier________PA_LOW
RF Low Noise Amplifier____Enabled
CRC Length________________16 bits
Address Length____________5 bytes
Static Payload Length_____8 bytes
Auto Retry Delay__________1500 microseconds
Auto Retry Attempts_______15 maximum
Packets lost on current channel_____________________0
Retry attempts made for last transmission___________15
Multicast__________Disabled    Custom ACK Payload___Enabled
Dynamic Payloads___Disabled    Auto Acknowledgment__Enabled
Primary Mode_____________TX    
TX address____________ 0x65646f4e32
pipe 0 (closed) bound: 0x65646f4e32
pipe 1 ( open ) bound: 0x65646f4e31
pipe 2 (closed) bound: 0xc3c3c3c3c3
pipe 3 (closed) bound: 0xc4c4c4c4c4
pipe 4 (closed) bound: 0xc5c5c5c5c5
pipe 5 (closed) bound: 0xc6c6c6c6c6
```
Clearly I'm not done with addresses on pipes 2-5 (think I'll print the single byte instead of tweaking `print_address_register()`). I've abandoned creating a binary representation for "AutoAck" feature status (about per pipe control), instead the output focuses strictly on pipe 0. Notice the "Power Mode" description is missing because I would need to save a private member that holds the last mode set to the CE pin in `ce()`, and it doesn't seem worth it.

I haven't committed this to my master branch yet (still need to test the SPI access changes from #686, but so far so good).
I also changed CSN pin's output label to show `/dev/spi<bus_numb>.<bus_device>`, but I haven't tested it on Linux yet.

closed 2020-12-14 by 2bndy5

combs, 2020-12-26:
love this--many thanks.

I wonder if something went awry with the Primary Mode / Power Mode bits. 

https://nrf24.github.io/RF24/RF24_8cpp_source.html#l00631

```
     config_reg = read_register(NRF_CONFIG);
     printf_P(PSTR("Primary Mode\t\t= %cX\r\n"), config_reg & _BV(PWR_UP) ? 'T' : 'R');
```

should "Primary Mode" be looking at bit `PRIM_RX`? it's currently outputting power-up status (`PWR_UP`)
and "Power Mode" is not currently output.

combs, 2020-12-26:
one other thing I'm noticing--this is trickier to fix--the "Auto Ack" line is only looking at Pipe 0; the `EN_AA` register has settings per pipe, so if you should happen to have AA disabled for Pipe 0, it will show up as disabled across-the-board in `printPrettyDetails()`. 

In my (complex) use case I am using AA settings per pipe and noticed this behavior. (mixed Si24R1 and NRF24L01+; can't use dynamic acks; need multicast channel and also acked point-to-point comms)

One idea for how it could work: show which pipes have AA enabled, perhaps like this:

```
Auto Ack Pipes	= 0 1 2
```

2bndy5, 2020-12-26:
Yes, when putting that function together, my attempt to display it as a binary representation was too dynamic. Currently, it works by using `PROGMEM` array of c-strings (which is why I used static strings as output). I didn't my look to much into "fixing" it after I got it working. You can have a go at it (it doesn't have to be binary representation), and submit a PR. Otherwise, it would be better/easier to use `printDetails()`'s hex value for per pipe control of auto-ack.

2bndy5, 2020-12-26:
Oh good catch on the "primary mode" output. That must've been a copy-n-paste or trigger happy auto-complete mistake. That's an easy fix.