nRF24/RF24 #688: is the python c'tor comment out of date?

2bndy5, 2020-11-20:
https://github.com/nRF24/RF24/blob/a0cd707137607842f0ef782f12f2e40409a12d38/examples_linux/pingpair_dyn.py#L26

I've been proofreading... I came across this line and couldn't find the file **RF24/arch/BBB/spi.cpp**. I noticed the linux examples use a more generic comment that references links to the documentation for non-RPi SoCs.
```c++
/****************** Linux (BBB,x86,etc) ***********************/
// See http://tmrh20.github.io/RF24/pages.html for more information on usage
// See http://iotdk.intel.com/docs/master/mraa/ for more information on MRAA
// See https://www.kernel.org/doc/Documentation/spi/spidev for more information on SPIDEV
```

I had blindly copied these comments into the new examples, so should I convert the later to a python comment and use it the new python examples?

TMRh20, 2020-11-21:
I think RF24/arch/BBB/spi.cpp is the original SPIDEV driver which was for BeagleBoneBlack. It probably still applies, but I don't think we need to mention BBB specifically.

2bndy5, 2020-11-21:
ok I'll replace the 
```python
# RPi Alternate, with SPIDEV - Note: Edit RF24/arch/BBB/spi.cpp and
# set 'this->device = "/dev/spidev0.0";;' or as listed in /dev 
```
with
```python
################## Linux (BBB,x86,etc) #########################
# See http://tmrh20.github.io/RF24/pages.html for more information on usage
# See http://iotdk.intel.com/docs/master/mraa/ for more information on MRAA
# See https://www.kernel.org/doc/Documentation/spi/spidev for more information on SPIDEV
```
in the new python examples (old pyRF24 example will be replaced/removed due to new python examples -- making maintenance easier).

closed 2020-12-14 by 2bndy5