nRF24/RF24 #693: RaspberryPi 4: No Communication with NRF24L01

Luddi96, 2020-12-08:
Hi All,

I am currently having a problem getting the NRF24L01 running on Raspberry PI 4.
I have crated a fresh RaspberryOS on the PI4, enabled SPI, done everything like stated here: 
https://tmrh20.github.io/RF24/Linux.html
Manual install with SPIDEV
When I run the "getting started", I get all 0xff (image 1)
Then I take the same SD card, plug it into a PI2, power on, everything works and I get a reasonable output (image 2)
Connections are 100% the same, I swapped all cables for fresh ones, swapped the modules nothing helped. Also a colleague of mine did this test with the same result.
Is there maybe a problem with BCM2711 vs BCM2835?
Thanks for some ideas and help!
Images: https://www.dropbox.com/sh/uzt4msmqeh0ap48/AABbjTSYK2igxJ2mO6jTaE1ba?dl=0

Luddi96, 2020-12-08:
Ok, my bad.. Somehow I had 3 "defective" modules and only one good... what are the odds... The modules work all on a Arduino but only one of them on the PI4.
But seems to be a HW problem, none of the Lib, so I close the issue.

closed 2020-12-08 by Luddi96

2bndy5, 2020-12-08:
Check you wiring from RPi4 to nRF24L01 module. Usually when all SPI transactions fetch `0xFF` for every byte, it's a wiring problem, or no power getting to the radio.

I've been testing the library with an RPi4 (also using a fresh image on the SD card with all updates), and I haven't been getting any problems.