nRF24/RF24 #698: [Question] About writeFast definition

Hendermike, 2020-12-17:
Hi! Been using the library the past few days and looking at writeFast definition a question came to my mind regarding its internal startFastWrite call. ¿How is it that such function is able to work properly without specifying bool startTx?

I'm pretty much a newcomer to RF comms and C programming in general, and by looking at some stackoverflow threads about this I formed the impression that such function use may lead to "undefined behavior".

Also wanted to say thank you for your dedication on this project, it has been an awesome learning experience to use and study it.

2bndy5, 2020-12-17:
>How is it that such function is able to work properly without specifying bool startTx?

what your asking seems like 2 thing:
1. `startTx` involves manipulating the CE pin while in TX mode.
2. understanding a C++ tactic for using/setting default argument values in function prototypes (located in RF24.h)

About 1: stick to getting your information from the nRF24L01 datasheet. It can be easily misinterpretted, so answers on stackoverflow (or elsewhere) become subjective because they're second-hand information sources.

About 2: This is a rather basic feature in C++. Instead of explaining in detail, I would defer to online sources about learning C++ about using default argument values.

If I've misunderstood your question or you need specific sources, I can elaborate, but we try to stick details about using the library on these issues.

Hendermike, 2020-12-17:
Many thanks for the explanation and advice! Made me realize that my concerns where basically originated by my C++ illiteracy. Will dig a bit more on that.

closed 2020-12-17 by Hendermike