nRF24/RF24 #700: [Request] add intuition to setters of auto-ack & dynamic payload features

2bndy5, 2020-12-20:
For this thread, I'll be using the terms
1. **dyn_pl** to abbreviate **Dynamic Payloads** 
2. **auto-ack** to abbreviate **Automatic Acknowledgements**

### Current lack of intuition
Per the datasheet, feature prequisites are as follows
| feature | prerequisite |
|:-------:|:------------:|
| auto-ack | None |
| dyn_pl | auto-ack |
| EN_DPL bit of<br>the FEATURE regiater | any pipe that<br>has dyn_pl enabled |

Currently the [`setAutoAck()`](https://github.com/nRF24/RF24/blob/045a6898d666a820cd8b995c7fff94d5af268d72/RF24.cpp#L1407) does not disable dyn_pl for any pipe(s) that have auto-ack feature disabled. Furthermore, [`enableDynamicPayloads()`](https://github.com/nRF24/RF24/blob/045a6898d666a820cd8b995c7fff94d5af268d72/RF24.cpp#L1295) does not enable auto-ack for any pipe(s) that have dyn_pl enabled.

### Solution
Ultimately, the following changes would prevent the user from mal-configuring the dyn_pl & auto-ack feature:
1. `setAutoAck()` should force disable the dyn_pl feature when disabling auto-ack for any/all pipes.
2. `enabledDynamicPayloads()` should force enable the auto-ack feature for any/all pipes 

### Additional context
Unfortunately, I can't think of a quicker way to execute this without using a new private member to track dyn_pl feature. Without adding this new private member, we would have to read the DYNPD register to track the dyn_pl feature for every time the feature is relevant. Additionally, we can use the existing private bool `auto_ack_enabled` as a uint8_t to track the auto-ack feature for all pipes.

A case could be made here to also implement per pipe control of the dyn_pl feature, but `enableDynamicPayloads()`/`disableDynamicPayloads()` should be consolidated into `setDynamicPayloads()`. Otherwise, `enableAckPayload()`/`disableAckPayload()` could just enable/disable dyn-pl for all pipes instead of just pipe 0.

TMRh20, 2020-12-22:
I'm curious where in the datasheet it indicates that Auto-Ack is a prerequisite for Dynamic Payloads?

RF24Mesh uses a combination of Dynamic Payloads and No-Ack, and a quick test shows that dynamic payloads work fine with auto ack disabled.

2bndy5, 2020-12-22:
In the description of the DNYPD register, it says for each bit:
>Enable dyn. payload length  data pipe 5. (Requires EN_DPL and ENAA_P5)

You're right I should've tested this, but I remember (about a year ago when developing my circuitpython library to support dyn_pl) this pre-req seemed important. I will also investigate further.

2bndy5, 2020-12-22:
so, dyn_pl enabled does seem to work with auto_ack disabled. 😕 But now I have even more questions:
1. Do you think that the description of the bits in the DYNPD register is meant to be exclusively for ACK payloads?
2. Is RF24Mesh/RF24Network using this combination (dyn_pl=enabled, auto_ack=disabled) to accommodate for the weird behavior in the Si24R1 (concerning the reverse implementation of the NO_ACK bit)?
    1. does adding this intuition hurt RF24Mesh/Rf24Network?
 
I am leaning toward closing this because it is *much easier* to support the features' states independently (as opposed to what the datasheet says).

TMRh20, 2020-12-22:
1. I've no idea why it says that.
2. RF24Network can use 1 pipe for multicast (RF24Mesh) so AA is disabled on a single pipe unless transmitting a 'normal' payload.
 i. Yes, it would break the multicasting because dyn. payloads would be disabled for the one pipe

Doesn't this go against your 'document it not support' it theme? :p

2bndy5, 2020-12-22:
>Doesn't this go against your 'document it not support' it theme? :p

😄 I was only trying to make the lib mimic what the datasheet was saying. But, this is yet another example of Nordic's lazy/poor explanation in the datasheet (just like how the SPI cmds W_ACK_PAYLOAD, FLUSH_TX, FLUSH_RX work despite the CONFIG register's PRIM_RX bit state or when EN_CRC bit isn't read as 1 when enabling auto_ack is forcing it high).

Nonetheless, I'm still glad to be working out these inconsistencies in the datasheet with your help. Doing it alone sucks ("2 minds are better than 1").

BTW, I've hardcoded this intuition in my CirPy lib, so I'll have to undo that also...

closed 2020-12-22 by 2bndy5