nRF24/RF24 #705: [Question] Not conexion between ESP32 and RPi

TaixMiguel, 2020-12-24:
I’m trying to launch the sample files by putting an ESP32 as TX and a RPi 4B as RX and I can’t get them to communicate with each other.

The sample files I am following are:
- https://raw.githubusercontent.com/nRF24/RF24/master/examples_linux/getting_started.py
- https://raw.githubusercontent.com/nRF24/RF24/master/examples/GettingStarted/GettingStarted.ino

For the arduino example I have changed the pins 12 as CE and 5 as CSN, for the rest it is the same source.
My question is, does ESP32 have to be node 0 and RPi node 1, or does it have to be the same node?

2bndy5, 2020-12-24:
They cannot be the same node. It doesn't matter which one is node 1 or 0 as long as they aren't the same node.

TaixMiguel, 2020-12-24:
> They cannot be the same node. It doesn't matter which one is node 1 or 0 as long as they aren't the same node.

Okay, when I run the example, I assign each one a different node, and yet they don’t communicate with each other. In the RPi I get the message "Nothing received in 6 seconds. Leaving RX role".
I have decompressed the lines that take out the configuration details and this is the result:

**ESP32 (TX):**
SPI Speedz	= 10 Mhz
STATUS		= 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	= 0x65646f4e31 0x65646f4e32
RX_ADDR_P2-5	= 0xc3 0xc4 0xc5 0xc6
TX_ADDR		= 0x65646f4e31
RX_PW_P0-6	= 0x04 0x04 0x04 0x04 0x04 0x04
EN_AA		= 0x3f
EN_RXADDR	= 0x03
RF_CH		= 0x4c
RF_SETUP	= 0x03
CONFIG		= 0x0e
DYNPD/FEATURE	= 0x00 0x00
Data Rate	= 1 MBPS
Model		= nRF24L01+
CRC Length	= 16 bits
PA Power	= PA_LOW
ARC		= 15
SPI Frequency		= 10 Mhz
Channel			= 76 (~ 2476 MHz)
RF Data Rate		= 1 MBPS
RF Power Amplifier	= PA_LOW
RF Low Noise Amplifier	= Enabled
CRC Length		= 16 bits
Address Length		= 5 bytes
Static Payload Length	= 4 bytes
Auto Retry Delay	= 1500 microseconds
Auto Retry Attempts	= 15 maximum
Packets lost on
    current channel	= 0
Retry attempts made for
    last transmission	= 15
Multicast		= Disabled
Custom ACK Payload	= Disabled
Dynamic Payloads	= Disabled
Auto Acknowledgment	= Enabled
Primary Mode		= TX
TX address		= 0x65646f4e31
pipe 0 ( open ) bound	= 0x65646f4e31
pipe 1 ( open ) bound	= 0x65646f4e32
pipe 2 (closed) bound	= 0xc3
pipe 3 (closed) bound	= 0xc4
pipe 4 (closed) bound	= 0xc5
pipe 5 (closed) bound	= 0xc6

**RPi 4B (RX):**
================ SPI Configuration ================
CSN Pin		= /dev/spidev0.0
CE Pin		= Custom GPIO22
SPI Speedz	= 10 Mhz
================ NRF Configuration ================
STATUS		= 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	= 0x65646f4e32 0x65646f4e31
RX_ADDR_P2-5	= 0xc3 0xc4 0xc5 0xc6
TX_ADDR		= 0x65646f4e32
RX_PW_P0-6	= 0x04 0x04 0x04 0x04 0x04 0x04
EN_AA		= 0x3f
EN_RXADDR	= 0x02
RF_CH		= 0x4c
RF_SETUP	= 0x03
CONFIG		= 0x0c
DYNPD/FEATURE	= 0x00 0x00
Data Rate	= 1 MBPS
Model		= nRF24L01+
CRC Length	= 16 bits
PA Power	= PA_LOW
ARC		= 0
================ SPI Configuration ================
CSN Pin			= /dev/spidev0.0
CE Pin			= Custom GPIO22
SPI Frequency		= 10 Mhz
================ NRF Configuration ================
Channel			= 76 (~ 2476 MHz)
RF Data Rate		= 1 MBPS
RF Power Amplifier	= PA_LOW
RF Low Noise Amplifier	= Enabled
CRC Length		= 16 bits
Address Length		= 5 bytes
Static Payload Length	= 4 bytes
Auto Retry Delay	= 1500 microseconds
Auto Retry Attempts	= 15 maximum
Packets lost on
    current channel	= 0
Retry attempts made for
    last transmission	= 0
Multicast		= Disabled
Custom ACK Payload	= Disabled
Dynamic Payloads	= Disabled
Auto Acknowledgment	= Enabled
Primary Mode		= RX
TX address		= 0x65646f4e32
pipe 0 (closed) bound	= 0x65646f4e32
pipe 1 ( open ) bound	= 0x65646f4e31
pipe 2 (closed) bound	= 0xc3
pipe 3 (closed) bound	= 0xc4
pipe 4 (closed) bound	= 0xc5
pipe 5 (closed) bound	= 0xc6

Does this help to know what my problem is and why I am not able to run the base example?
Thank you in advance

PS: my sensor nRF24L01 is this: 
https://5.imimg.com/data5/LN/GP/YG/SELLER-27298097/nrf24l01-module-with-antenna-500x500.jpg

2bndy5, 2020-12-24:
>PS: my sensor nRF24L01 is this:
https://5.imimg.com/data5/LN/GP/YG/SELLER-27298097/nrf24l01-module-with-antenna-500x500.jpg

That module is a nRF24L01+PA+LNA. The PA/LNA  part is important because **it requires more power to TX** (also during RX mode because the ACK packets are automatically sent in response to the ESP32). Those PA/LNA modules' store pages rarely say that they require a separate power source and electromagnetic shielding to work correctly -- in a way they're a bit of a scam unless you're an expert. 

The higher power requirement is also probably affecting your ESP32 board (RPi definetly doesn't supply enough power on the 3v pin). What ESP32 board are you using (store link please)?

TaixMiguel, 2020-12-24:
> That module is a nRF24L01+PA+LNA. The PA/LNA part is important because **it requires more power to TX** (also during RX mode because the ACK packets are automatically sent in response to the ESP32). [...]

The ESP32 is purchased here https://es.aliexpress.com/item/4000343057472.html model ESP32-WROOM-32D.

So, instead of connecting him to the 3.3V pin should I connect him to the 5V pin? On the ESP32 I would have no problem, but on the RPi I get a little off. I picked it up with antenna because they’re going to be away (less than 1 kilometer).

About the electromagnetic shielding, you could give me more information or reference me to some manual/guide, I don’t know what that is

2bndy5, 2020-12-24:
>So, instead of connecting him to the 3.3V pin should I connect him to the 5V pin?

Thank you for asking first because that would fry the module. By "power" I mean the current coming from the 3v pin. The module has to be powered by 3v (maximum), but to TX it needs at least 112 mA (possibly more). The RPi taps out at ~40mA and the ESP32 boards often consume most the current from their onboard 3V regulator. This is why a separate 3v power supply is needed to make the PA/LNA modules work.

>About the electromagnetic shielding, you could give me more information or reference me to some manual/guide, I don’t know what that is

Read [my comment on another issue (from my Circuitpython repo)](https://github.com/2bndy5/CircuitPython_nRF24L01/issues/19#issuecomment-711428924) starting with the second paragraph.

2bndy5, 2020-12-24:
[image]
This is the onboard 3v regulator for that ESP32 board. Looks like you'll need an external 3v power supply for the ESP32 as well.

Alternatively, you could replace the nRF24L01+PA+LNA modules with a typical nRF24L01+ module (no extra PA/LNA -- meaning it has an onboard antenna) if your project isn't going to be spanning more than 10 meters (32 feet)

TaixMiguel, 2020-12-24:
Thank you @2bndy5 for the answers

> By "power" I mean the current coming from the 3v pin. The module has to be powered by 3v (maximum), but to TX it needs at least 112 mA (possibly more). The RPi taps out at ~40mA, and the ESP32 [...]

Understood, as I put before, for the ESP32 I have no problem because I had planned to feed it with a battery of 3.7V, I could connect the nRF24L01 to the same battery, adding a voltage reducer to 3.3V (800mA) not to burn it, or would I use a direct connection? About the RPi, I didn’t think I would need an external source (beyond a power bank). Would I do with the same gearbox from the power bank? (I am thinking of a solution that better suits me not to have to put a new source)

> Read [my comment on another issue (from my Circuitpython repo)](https://github.com/2bndy5/CircuitPython_nRF24L01/issues/19#issuecomment-711428924) starting with the second paragraph.

Perfectly explained ;)

> This is a long reach. You must use 250Kbps Data Rate and max PA Level to get the most distance, but I'm not sure how valid the store pages' claims of 1km is.

The distance I need will cover between 100 and 200 meters with walls in between, with the simple module does not suit me, that’s why I went for PA/LNA

2bndy5, 2020-12-24:
>I could connect the nRF24L01 to the same battery, adding a voltage reducer to 3.3V (800mA) not to burn it, or would I use a direct connection?

Don't directly connect it to the batteries! Use a voltage regulator (800 mA should be plenty if there is nothing else consuming that current).

>About the RPi, I didn’t think I would need an external source (beyond a power bank). Would I do with the same gearbox from the power bank? (I am thinking of a solution that better suits me not to have to put a new source)

If you could tap into the power bank's 5v lines (from the power bank's extra USB port), you might be ok. I say *might* because those power banks usually have a current limiter built in, the the total output current is shared between multiple USB ports. BTW, the RPi4 needs more current than older versions of RPi. And typical power banks usually aren't strong enough for even an older RPi (the CPU will lag when the power dips too low).

2bndy5, 2020-12-24:
Remember the GND is **always** shared between multiple power supplies. 

Assuming google translate didn't hit a language barrier with the word "gearbox": Powering motors introduces electric noise in the power lines. Electrical noise is why we have to put capacitors on the nRF24 module's VCC and GND lines. Typically, you don't want to use the same power source for your motors and the MCU/RPi because of the electric noise from the motors.

TaixMiguel, 2020-12-25:
> I say _might_ because those power banks usually have a current limiter built in; the total output current is shared between multiple USB ports.

I didn’t know this, so an alternative source is better. Yesterday I came up with something about this point but seeing your last comment about noise I don’t know how serious it would be: connect the nRF24 module to the RPi by the 5V pin by applying a regulatory voltage to 3.3V/800mA.

> Assuming google translate didn't hit a language barrier with the word "gearbox":

Translator’s failure, I wanted to refer to the regulatory voltage

> Typically, you don't want to use the same power source for your motors and the MCU/RPi because of the electric noise from the motors.

In my case I will not use motors (failure of the translator when introducing that concept), unless the regulatory voltage is considered as an engine, or that the regulatory voltage also generates electrical noise (in which case my error for understanding that not)

What system would you use to power the nRF24 module that is connected to the RPi?

2bndy5, 2020-12-25:
>unless <sup>1</sup>the voltage regulator is considered as an engine, or that the <sup>2</sup>voltage regulator also generates electrical noise 

1. voltage regulators are not considered an engine.
2. Just to be clear, when I say "voltage regulator", I'm talking about [something like this](https://www.mouser.com/Semiconductors/Power-Management-ICs/Voltage-Regulators-Voltage-Controllers/Linear-Voltage-Regulators/_/N-5cg9g?P=1z0wa2eZ1z0z63x&Rl=5cg9gZerbbZ1z0wd8oZ1z0wbyvSGT). If you look at the datasheets for any voltage regulator, it will likely show you what capacitors to use and where to stick them in your circuit. The capacitors are what filters out electric noise (mostly). The regulators only ensure the voltage level stays consistent.

If you feel that wiring a 3V regulator is beyond your skill level, you can use a [specially designed breakout board for the nRF24L01+PA+LNA modules](https://www.amazon.co.uk/WayinTop-NRF24L01-Transceiver-Regulator-Compatible/dp/B07P95X6HM/ref=sr_1_2?crid=31AJ7J1NZEK3E&dchild=1&keywords=nrf24l01+breakout+board&qid=1602761187&s=electronics&sprefix=nRF24%2Celectronics%2C280&sr=1-2). Those breakout boards use a [AMS1117 3V regulator](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=&cad=rja&uact=8&ved=2ahUKEwiJ88OVwLbsAhXDGDQIHc2zB-wQFjAAegQIAxAC&url=http%3A%2F%2Fwww.advanced-monolithic.com%2Fpdf%2Fds1117.pdf&usg=AOvVaw32b7yUSaNtlsyuBUGG7NQr) (you may recognize this one because its the same one your ESP32 board is using). Be careful to not purchase a [breakout board designed for the ESP8266-01 modules](https://www.amazon.com/gp/slredirect/picassoRedirect.html/ref=pa_sp_mtf_electronics_sr_pg1_1?ie=UTF8&adId=A01682831I2HW14ZGLFU2&url=%2FStemedu-Breakout-Breadboard-Pre-Solder-Transceiver%2Fdp%2FB07KBZRZ3Y%2Fref%3Dsr_1_15_sspa%3Fcrid%3D2RJTHED1BVWFG%26dchild%3D1%26keywords%3Dnrf24l01%252B%2Bbreakout%2Badapter%26qid%3D1602761813%26s%3Delectronics%26sprefix%3Dnrf24%252Celectronics%252C203%26sr%3D1-15-spons%26psc%3D1&qualifier=1602761813&id=1389341831199863&widgetName=sp_mtf) as it doesn't look like they have a 3V regulator soldiered in (and the pinout concerning VCC & GND pins is incompatible with the nRF24L01 anyway).

>What system would you use to power the nRF24 module that is connected to the RPi?

Sorry, but I'm going to leave this to your research. There are many acceptable solutions out there, and your project's needs (concerning [voltage and current](https://en.wikipedia.org/wiki/Ohm%27s_law)) should be the deciding factor. Remember the GND is **always** shared between multiple power supplies.

TaixMiguel, 2020-12-28:
> If you feel that wiring a 3V regulator is beyond your skill level, you can use a [specially designed breakout board for the nRF24L01+PA+LNA modules](https://www.amazon.co.uk/WayinTop-NRF24L01-Transceiver-Regulator-Compatible/dp/B07P95X6HM/).

I did not know this module and it makes the work much easier, thank you for introducing me
> Sorry, but I'm going to leave this to your research. There are many acceptable solutions out there, and your project's needs (concerning [voltage and [...]

I understand, thanks to the "breakout board" I feel more comfortable and I can think of several ways to address the problem.

Thanks @2bndy5 for all the help offered, I leave the thread open if you want to add something and in a couple of days I close it.

If I have any further questions (when I receive the "breakout board") I will put it in a new thread as I guess it will be a different doubt than the initial one.

Again, thank you very much

closed 2020-12-29 by TaixMiguel