nRF24/RF24 #708: [Request] properly clamp arg values in setRetries()

2bndy5, 2020-12-27:
Currently the `setRetries()` uses the logic `arg & 0xf` to prevent erroneous configuration of the auto-retry feature. However, this doesn't clamp the value, rather it just truncates the bits that don't "fit" in the SETUP_RETRY register.

Current logic:
- passing 25 (an erroneous value) to either the `delay` or `count`  results in using a value of 9 
`25 & 0xf = 9`

## Solution
use the macro `rf24_min` to ensure that the value is clamped to expected range.
- passing 25 (an erroneous value) to either the `delay` or `count`  results in using a value of 15
`rf24_min(25, 15) = 15`

closed 2021-01-08 by 2bndy5

microtronics, 2021-03-17:
Actually I don't really think that this is a good idea. This fix makes the code slightly larger and erronous inputs are only replaced by a different type of unexpected behaviour. @2bndy5,@TMRh20 and @Avamander I would vote for reverting this back to the slightly more performant solution from before, because it doesn't really "fix" anything. What do you think?

Your improvement for the ARC-shift (you showed ARD in your example, but that value is actually 4!) on the other hand is not really useful as any reasonable compiler leaves this shift out anyways...

2bndy5, 2021-03-17:
>This fix makes the code slightly larger

I wasn't aware this change had an affect on the compile size. 

>erroneous inputs are only replaced by a different type of unexpected behavior

I'm not sure what you mean here. The invalid inputs should be replaced with the max value supported (thus the title of the issue). I wasn't concerned with negative numbers since the parameters are unsigned datatypes.

>Your improvement for the ARC-shift (you showed ARD in your example, but that value is actually 4!)

Yep, you caught a typo in the OP. I meant to use ARC (because that value is 0).

>not really useful as any reasonable compiler leaves this shift out anyways

Despite what the compiler does, my objection about `count << 0` was more geared towards readability (following the ironic euphemism "keep it simple stupid").

>I would vote for reverting this back to the slightly more performant solution from before, because it doesn't really "fix" anything

A PR is always welcome. To be clear though, this request was never meant to fix anything (nothing was broken to begin with), so reverting this wouldn't harm anything (I still prefer the code doesn't shift the `count` value 0 times).

PR #709 (2021-01-08): clamp setRetries() args & doc main page fixes
fixes #708

Tested on RPi4 using python wrapper.

#### Inconsequential doc changes (because I didn't want to open another PR)
- I accidently copied the wrong file names to the datasheets when migrating them from tmrh20.github.io repo. They were backwards.
- closes #712 about littlewire lib links.
- closes #715 about outdated list of changes