nRF24/RF24 #712: [Question] where is the LittleWire reference in the docs?

2bndy5, 2020-12-29:
I noticed there's a broken link to "LittleWire" on the main page of the docs (exists in both nRF24.github.io & tmrh20.github.io variants of the docs. The link is supposed to lead to "littlewire.html"
[image]
The only reference to the littlewire library is in [Linux.html (tmrh20.github.io link)](https://tmrh20.github.io/RF24/Linux.html). 

Furthermore, I can't find online where the littlewire library exists.  I found the [littlewire github org](https://github.com/littlewire) (doesn't seem to be active for a couple years), but it seems to relate to specially designed hardware that uses a ATTiny85 to forward communication from USB to... something GPIO capable (maybe an AVR chip?). But why not just use a [nRF52840 dongle dev kit](https://www.mouser.com/ProductDetail/Nordic-Semiconductor/nRF52840-Dongle?qs=gTYE2QTfZfTbdrOaMHWEZg%3D%3D) with [the Nordic SDK's proprietary ESB stack](https://infocenter.nordicsemi.com/topic/sdk_nrf5_v17.0.2/group__nrf__esb.html?cp=7_1_6_9_0) for something like that (an idea I've had on the backburner for a long time).

Avamander, 2020-12-29:
> But why not just use a nRF52840 dongle dev kit with the Nordic SDK's proprietary ESB stack for something like that

I suspect the idea to use a separate MCU for the radio with LittleWire predates the nRF5x series.

> (an idea I've had on the backburner for a long time).

Hm, yes, same. I wanted to try and get Adafruit Feather working with MySensors just to see how intercompatible nRF24 and nRF5x are, but I set it back on the backburner because of various issues that rose.

2bndy5, 2020-12-29:
@Avamander You and I should have a separate discussion about that idea (great minds think alike). I was going to try and add ESB protocol to the CircuitPython firmware for nRF52 based MCUs, but stopped when I realized that the firmware isn't using the Nordic sofdevice's *timeslots API* (they'd rather use the radio peripheral for only BLE). The Arduino core for the nRF52 has abandon it because they seemed unsure about supporting concurrent usage of both protocols. Ultimately, I think there would need to be a separate arduino core that disregards the Nordic Sofdevice's BLE stack (or maybe a macro that enables ESB xor BLE).

back to the issue at hand... It would probably be better to just removed the hyperlink on the main page, yes?

Avamander, 2020-12-29:
> back to the issue at hand... It would probably be better to just removed the hyperlink on the main page, yes?

Maybe the link, but keep the mention unless LittleWire has been totally stripped out.

2bndy5, 2021-01-01:
I found a link in one of the footnotes for the pinout table that leads to http://littlewire.cc/ but the domain seems to be up for sale...

PR #709 (2021-01-08): clamp setRetries() args & doc main page fixes
fixes #708

Tested on RPi4 using python wrapper.

#### Inconsequential doc changes (because I didn't want to open another PR)
- I accidently copied the wrong file names to the datasheets when migrating them from tmrh20.github.io repo. They were backwards.
- closes #712 about littlewire lib links.
- closes #715 about outdated list of changes
closed 2021-01-08 by 2bndy5